Ultimaker/Cura #10237: Cura 4.10 support pattern no changing. Keep sticking to zigzag

SkylinerXwiz, 2021-08-06:
### Application Version

4.10

### Platform

Windows 10

### Printer

Prusa i3

### Reproduction steps

Change the support pattern from zigzag to line, or any other type. (I tried with line, grid, Gyroid)
Click on slice.
then preview

### Actual results

preview keep seeing zigzag pattern

### Expected results

should be line support or any other type of support pattern that i select
[Desktop.zip](https://github.com/Ultimaker/Cura/files/6942805/Desktop.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Nil

fvrmr, 2021-08-06:
Hi @SkylinerXwiz on your screenshot I can't really see what support pattern is used in the support it self. 
The project file you shared had support pattern triangles and I could see that was used. 
[image]
The reason why you can't see it that well is probably of the support density and the gradual support infill steps. 

If I change it to lines and set gradual support infill steps to 0, this is what I get: 
[image]
The lines may look like zig zag because of the connect support lines.

cliffordvandyk, 2021-08-12:
[image]
I am seeing the same issue - Support structures seem to remain at the zig-zag default, and default 20% density, and the support interface flag appears to be ignored.

fvrmr, 2021-08-13:
@cliffordvandyk can you share your project file (file - save project)?

cliffordvandyk, 2021-08-13:
Hi fvrmr

Changing the support pattern, support density and "Enable support interface" seems to have no influence over the support pattern/density/interface - it remains per my screen grab above.
[CCR10SPRO_display_holder_scaled.zip](https://github.com/Ultimaker/Cura/files/6982252/CCR10SPRO_display_holder_scaled.zip)
 To confirm, using CURA 4.10 on Windows 10.

fvrmr, 2021-08-13:
Thank you, it seems like you want to change the `support interface pattern`.  To see these settings you can select all in the setting visibility. And then you will also some other settings related to the support interface.

cliffordvandyk, 2021-08-13:
Thanks for your reply. I indeed tried to activate support interface as the
default support grid was not dense enough, and Cura did not seem to respond
to me increasing the density.

I activated the support interface in settings. After selecting it and
ticking the checkbox, I again saw no change in slicer output. I also tried
changing the support pattern (and support density mentioned above), but
none of my changes showed up in the sliced view.

Kind regards,
Clifford

cliffordvandyk, 2021-08-16:
Further follow-up:
After "fiddling" a bit with enabling extra settings on the menu, Cura does now seem to be responding to support pattern, density and request for a support interface. I am not sure what caused the change in behaviour, but it is now working.

Ghostkeeper, 2021-08-17:
@SkylinerXwiz probably wants to disable the Connect Support Lines setting. Using that in combination with the lines pattern looks a lot like zigzag. But other than that, it seems to change fine in layer view, starting from their project file.

closed 2021-08-20 by no-response[bot]