Ultimaker/Cura #10242: Pause at height bug #2

EGOiST1991, 2021-08-06:
### Application Version

4.10

### Platform

Windows10

### Printer

Anycubic i3

### Reproduction steps

Bug:
The Post Processing Plugin "Pause at height" has a bug with the Repetier Method where the Z-height is 1mm to high after continue.
Discuss: #8575

Pause at height script uses two settings:
mm and layer.
In my PR #8578 i corrected the mm issue (the on i use)
To fix it in the layer Option: (Its the same error differnt line for method "griffin")

The wrong line starts here:
494: prepend_gcode += self.putValue(G = 1, Z = current_z + 1, F = 300) + "\n"
The correct line:
458: prepend_gcode += self.putValue(G = 1, Z = current_z , F = 300) + "\n"

### Actual results

494: prepend_gcode += self.putValue(G = 1, Z = current_z + 1, F = 300) + "\n"

### Expected results

458: prepend_gcode += self.putValue(G = 1, Z = current_z , F = 300) + "\n"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none. ive proved that before :)

GregValiant, 2021-08-06:
Did you happen to be using "By Height" and were there Z-Hops in the file?
Coincidentally - did the ZHop height happen to be 1mm?
"By Layer" works fine while "By Height" can be fooled by Z-Hops.

fvrmr, 2021-08-09:
Hi @EGOiST1991 I will close this issue, and discuss [this comment ](https://github.com/Ultimaker/Cura/issues/8575#issuecomment-894027534)with the team. 
Keep you posted in the other thread.

closed 2021-08-09 by fvrmr