Ultimaker/Cura #10243: Build plates temperature is 20 degrees higher in sliced gcode than setting in Cura 4.10

chasrobertson, 2021-08-06:
### Application Version

4.10

### Platform

WIndows 10

### Printer

Weedo X40

### Reproduction steps

Tried numerous settings and profiles, but with the build plate temperature set to ( in this case 50 degrees), the actual gcode prodused sets the build plate temperature to 70 degrees. Thia offset is always plus 20 degrees  with all settings.

### Actual results

Always get gcode 20 degress higher for the bed than cura settings,

### Expected results

Should in this case have been set to 50 degrees

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I have tried all the different profiles with the same result.
Here is the gcode output when the cura setting for the bed is 50 degrees

[WX40_flowRateCalibrationCube_0.8mmWalls.gcode.zip](https://github.com/Ultimaker/Cura/files/6945532/WX40_flowRateCalibrationCube_0.8mmWalls.gcode.zip)

fieldOfView, 2021-08-06:
Please also add a *project* file (File -> Save Project...). With a project file we can see what settings you used to create the gcode file you attached.

fieldOfView, 2021-08-06:
Check the setting named `Build Plate Temperature Initial Layer`. The definition for your Weedo may define this as the normal bed temperature + 20.

closed 2021-08-20 by no-response[bot]