Ultimaker/Cura #10249: Multiplying Objects Not Working

gitatmax, 2021-08-07:
### Application Version

4.10.0

### Platform

Monterey 21A5294g

### Printer

Creality Ender 3v2

### Reproduction steps

1. Select an object on the plate
2. Right-click and choose to Multiply Selected

### Actual results

3. A dialog appears but nothing is multiplied

### Expected results

3. Selected items should multiply

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6949100/cura.log)

fvrmr, 2021-08-09:
Hi @gitatmax does it happen on every model? If not could you share your project file (file - save project)?
I am also wondering if this also happened before you updated to Monterey beta?

charlespick, 2021-08-09:
I have also been having this issue for a while now. I am also running Monterey. Not sure if it is related but arrange models is also broken. 

I just tested both features on High Sierra and they are working properly there so it does appear to be a macOS Monterey related issue. 

[image]

fvrmr, 2021-08-12:
Hi @gitatmax and @charlespick thank you for checking! We don't support Monterey yet since it is a beta. But I will bring this up with the team.

charlespick, 2021-08-12:
It won't be beta much longer!

fvrmr, 2021-08-13:
We've decided to keep this issue open for now and when Monterey is not in beta we will take a look at it again.

ylandot, 2021-08-18:
Same problem here, no problem with Cura 4.8, so while it may be related to Monterey, note that it seems to happen only with last release of Cura 4.10. 
thanks

43moon, 2021-08-25:
Same problem with Cura 4.10 and 4.11 Beta on Monterey

sghoweri, 2021-09-09:
+1 running into this exact same issue. Problem seems to happen in at least v4.11, v4.10, and v4.9.1 (just tested). Super frustrating.

Akira-Maia, 2021-09-19:
Yeah same problem here. But just thought I'd mention the workaround is to import the stl multiple times from Finder.

Ghostkeeper, 2021-09-27:
What happens if you import multiple models one by one and then right click in the scene and select "Arrange All Models"? Does it have the same effect?

sghoweri, 2021-09-27:
> What happens if you import multiple models one by one and then right click in the scene and select "Arrange All Models"? Does it have the same effect?

Importing the same file multiple times (as a workaround) duplicates the object as expected -- the problem however is that none of the other actions (ie. arrange all models) work when these objects are selected. That functionality is broken.

Ghostkeeper, 2021-10-04:
Interesting. That should help a bit to narrow down the cause of this issue. Thanks!

We use the same method internally to place a single object as when multiplying multiple objects or arranging all models. The difference is that when loading a single object, all other objects are "fixed" and not allowed to move. Maybe something in libnest2d crashes when needing to decide which model to place first.

gitatmax, 2021-11-01:
@fvrmr Much appreciated.

shamannrunner, 2021-11-03:
I'm having the same issue here. Supposed to ship out product.

ragecub3, 2021-11-06:
I have the same issue.

YassineGherbi, 2021-11-07:
Same issue, with Multiplying and Arranging.

Servus7, 2021-11-07:
Im having the same issue after updating to Mac OS Monterey. And it’s not a beta anymore.

neofuture, 2021-11-08:
Same issue here, Creality slicer which is a port does work ok !

mrrstrat, 2021-11-08:
The issue emerged right after I upgraded to Monterey from Big Sur. I am using the latest current Cura and right after upgrade to Monterey I could no longer multiply objects and when I try to load multiple objects they all are put in the same spot making the workflow painful to make many of a single part as they have to be arranged and positioned manually after loading them one at a time.

Multiply was so much easier and even arranged the items. This is not likely a permissions issue - I have not dug into the code to se why this could be happening.

nallath, 2021-11-08:
> Same issue here, Creality slicer which is a port does work ok !

Creality slicer is a pretty old port. We have updated the arrange algorithms since then.

mrrstrat, 2021-11-08:
Ok - issue exists, has been known since Monterey beta, and is front and center for all Mac users. Is this issue even being worked on?

gitatmax, 2021-11-08:
@mrrstrat It doesn't feel like it, but then again I wouldn't know how they work in there. I'm sure there's a system and that it's in the pipeline! But, yeah, it's not feeling like it's been a priority—which is a shame.

nallath, 2021-11-08:
> Ok - issue exists, has been known since Monterey beta, and is front and center for all Mac users. Is this issue even being worked on?

It's in the pipeline. No need to fret. But it should not come as a surprise that we also work on other things, some of them which have an even higher priority than this. 

Also; feel free to propose a solution to this bug. We would be more than happy to have a look at any solutions that people could find.

GJSchaller, 2021-11-08:
> Also; feel free to propose a solution to this bug. We would be more than happy to have a look at any solutions that people could find.

Between this, and other macOS based issues (lack of multi-core support, lack of M1 chip support), the best solution may be a larger one of updating Cura's macOS development & compiling process & environment.  It might be worthwhile for the Cura team to reach out to Apple about getting better support / resources, and help bolster their own internal macOS support.

fargiolas, 2021-11-08:
not sure if this helps, here's another log file produced while seeing the issue, there's a traceback of the error at the end of the file

[cura.log](https://github.com/Ultimaker/Cura/files/7499648/cura.log)

nallath, 2021-11-09:
> Between this, and other macOS based issues (lack of multi-core support, lack of M1 chip support), the best solution may be a larger one of updating Cura's macOS development & compiling process & [...]

Apple has made the choice over and over again to pull the rug away from developers every time an update is done. We aren't large enough to influence a company like apple; Even the frameworks that we use (that should have a lot more influence than us since more people use it) don't have that. If someone has those contacts with Apple, I'd love to get into a conversation with them. 

As for the threading support, we are working on that.

Gelob, 2021-11-09:
I found a temporary solution to this and maybe this will lead to finding a permanent fix in the future.

1. Close Cura & Open Finder and navigate to Applications folder
2. Right click on Ultimaker Cura and select Show Package Contents
3. Go to the folder Content > Resources > lib > shapely > speedups
4. Rename _speedups.cpython-38-darwin.so to _speedups.cpython-38-darwin.so.bak
5. Open Cura and test multiplication

This disables speedups from the shapely python plugin which is enabled by default. I'm not sure if any adverse affects from this besides maybe performance so do it at your own risk. But it does fix the multiplication and arrangements of model issues. https://shapely.readthedocs.io/en/latest/manual.html#performance

Another project had a similar issue with OSX 12. Maybe these pre-compiled enhancements from Shapely are not valid on Monterey. 
https://github.com/inkstitch/inkstitch/issues/704
https://inkstitch.org/docs/install-macos/#valueerror-null-geometry-supports-no-operations

UnchartedBull, 2021-11-09:
Can confirm that the temporary fix worked fine for me as well with Cura 4.12.0 and macOS 12.0.1. Thanks!

GJSchaller, 2021-11-09:
A note if you upgrade to 4.12 - you have to launch it and approve it BEFORE you apply the fix, or the app / package will be deemed damaged and unable to launch.

neofuture, 2021-11-10:
Considering that all new Macs are now M1(1+ 1Pro)M2 and moving forward its imperative that the code base is built with this in mind, and if its not multicore then this will cause real issues with the arm environment going forward, Rosetta is only a temporary solution to run none arm code and then will remove this soon.

I have 5 x86 Macs, and all my new Macs are ARM 

I think its more important that you fix multicore and ARM support before even fixing old bugs or new releases because its running like a complete dog right now, cant preview slices, cant slice complex models on a Mac, I have to slide and load the slices in IdeaMaker to see the are ok....

neofuture, 2021-11-10:
maybe the team can release a 4.12 (soft fix) release that has this quick fix that will enable as an install to allow others none technical to be able to have this fix, I mean its broken, so enabling the quick fix, is fixed for a while for consumers until its not broken anymore ?

4.12.1 anyone ?

neofuture, 2021-11-10:
slow is better than broke !

fargiolas, 2021-11-10:
Maybe this helps too, renaming bundled shapely to shapely.bk and replacing with latest shapely installed with `pip install shapely --no-binary shapely` fixes the bug. I guess that's either a bug with bundled shapely or GEOS and this forces it to use system installed ones (here shapely 1.8 and geos 3.10.1 installed respectively with pip and with brew)

mrrstrat, 2021-11-10:
I might advise against modification of dependency components as these might be broken when the Dev team releases additional updates Cura (and it might be unclear how large the expanse of the break and the scope of the fix is without details of the deployment packaging). It sounds like they only need a re-aggregation of the dependancies to the Cura application and all is working. This might be a good 4.12.1 candidate release and easy to implement/test/deploy.

mrrstrat, 2021-11-10:
I would gladly be a beta tester on this if it helps drive a high opportunity and easy release :-)

evtrados, 2021-11-10:
If I install `shapely v1.8.0` on our build environment and create a Cura build based on that, then indeed the object multiplies normally on that Cura build. Note that I checked this on our build server VM and not on a physical device, but that theoretically should not behave differently.
I will discuss with the rest of the team tomorrow to see whether we can give you guys one of our nightly builds to test yourselves on your physical macbooks and make sure it works as it used to.

mrrstrat, 2021-11-10:
I have a MBP 13 2019 running Monterey I have use. Its my lab computer and use it for the Cura slicing for the printers in the lab. I can test this if you want - also you can give me some use cases or whatever you use to validate test before it goes to QA stage. I am dialing in ABS on an Ender 3 V2 currently so the testing would be part of my ABS tuning (I am assuming the changes will not convolute my effort and only are addressing the load/placement and multiply/placement functionality.

I can do this as soon as I get a beta as I am sure there are many others that can do this as well.

evtrados, 2021-11-12:
Hi everyone.
We decided with the team to provide you with a [Cura build]([url](https://software.ultimaker.com/cura_nightlies/Ultimaker_Cura-4.12.99-Monterey_fix-Darwin.dmg)) that seems to fix the Monterey issues you've been facing, so that you can test if - indeed - everything is resolved after using the 1.8.0 version of `shapely`.

flipace, 2021-11-12:
> Hi everyone. We decided with the team to provide you with a Cura build that seems to fix the Monterey issues you've been facing, so that you can test if - indeed - everything is resolved after using the 1.8.0 version of `shapely`. Please find it [here](https://software.ultimaker.com/cura_nightlies/Ultimaker_Cura-4.12.99-Monterey_fix-Darwin.dmg).

Awesome, thanks - multiply works perfectly fine for me now!

fargiolas, 2021-11-12:
> Hi everyone. We decided with the team to provide you with a Cura build that seems to fix the Monterey issues you've been facing, so that you can test if - indeed - everything is resolved after using [...]

thank you! the nightly build with shapely 1.8.0 works fine here! (macbook air m1 with monterey)

JWRu, 2021-11-12:
Thanks a lot. The fix works fine.

neofuture, 2021-11-12:
Awesome and a quick and efficient fix, works like a charm, thank you

mrrstrat, 2021-11-13:
I tested it as well - WORKS!  This x.x.99 version replaces my working Cura now :-)

When the update rolls around I'll update it again. Great job Cura Dev team!!!

gitatmax, 2021-11-14:
Thanks for the fix. It's lovely to have it working again.

fredlex3dprints, 2021-11-16:
That's for the fix. Tested it and works fine now !

fvrmr, 2021-11-18:
We have released an official version with contains this fix. You can download it here: https://ultimaker.com/software/ultimaker-cura

I will close this issue, since it is fixed!

closed 2021-11-18 by fvrmr