Ultimaker/Cura #10250: The default scale factor for Scale Extremely Small Models

GregValiant, 2021-08-07:
### Is your feature request related to a problem?

The current default scale factor is 10000%.  I would suggest that the default be made 2540% or that there be a toggle that the user could set for 10000 or 2540.
I think that most people who come across this have models that were designed in English units so auto-scaling to metric would be handy.  The dialog that informs the user that autoscaling was used may need an update.

### Describe the solution you'd like

A toggle or child checkbox to switch between 2540 and 10000.

### Describe alternatives you've considered

Select the model.  Select the Scale tool.  Set the scale to 2540.

### Affected users and/or printers

Anyone importing a model into Cura that was designed in Inches.

### Additional information & file uploads

N/A

fieldOfView, 2021-08-08:
The MeshTools plugin has a neat function to scale models on load.

GregValiant, 2021-08-08:
I like it.  It was hidden in the weeds but I finally found it.  Thank you.
I'll close this.

closed 2021-08-08 by GregValiant