Ultimaker/Cura #10251: Adjust the "doubling" of infill for gradual infill steps

bigntallmike, 2021-08-07:
### Is your feature request related to a problem?

When using gradual infill with say five steps, maxing out at 35% infill, the jump between the first two steps is just massive, and the poor nozzle is basically printing 90% in thin air.

### Describe the solution you'd like

I'd like to be able to change the "double" every step to an option like 1.5x or 1.3x denser at each step.  We could do more steps, closer together to achieve a better outcome.

### Describe alternatives you've considered

N/A

### Affected users and/or printers

All users of gradual infill

fvrmr, 2021-08-09:
Hi @bigntallmike the reason why we double every step is to make sure the infill lines are overlapping each other. 
But I will discuss this with the team. Keep you posted.

konskarm, 2021-08-16:
This feature is now in our backlog. Devs see internal ticket CURA-8485.

MakManIfest, 2021-09-10:
This is an issue I am also struggling with in Cura. I've been using S3d and achieve nice tops while using low density strong infill patterns by splitting the process and changing to an infill pattern and density that will support the top layer formation.
For example I printed a fang part cooler that I wanted to keep light as possible. I used 20% honeycomb infill and split the process 3 layers below the 1st top skin layer. Switched to 30% rectilinear infill for the second process, then split again to return to 20% honeycomb for the rest of the part after the top had formed.
Implementing in Cura the ability to control infill densities and patterns within groups of layers would be extremely useful.

Ghostkeeper, 2021-09-20:
Note that you can get very precise control over infill densities wherever you like using infill meshes. That has so far been the intended way to work with this. This feature is merely to make that control easier to use.