Ultimaker/Cura #10252: Unable to sign in

guillaumechauvat, 2021-08-08:
### Application Version

4.10.0

### Platform

Arch linux

### Printer

Ultimaker 2

### Reproduction steps

1. Launch Cura with no data saved
2. Click on "sign in" when the menu appears
3. Get redirected to Firefox with a login page and sign in there.

### Actual results

The login appears successful in Firefox, but Cura doesn't see it.

[image]

[image]

Clicking on "sign in" again in Cura has no effect.

If I skip the login there, I can click on "Check for account updates" but it fails and looks like this:

[image]

### Expected results

I can sign in and synchronise my account.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Logs: 
[cura.log](https://github.com/Ultimaker/Cura/files/6950476/cura.log)

fvrmr, 2021-08-09:
Hi @guillaumechauvat did this happen with the AppImage of Cura?

guillaumechauvat, 2021-08-09:
No, it's the [Arch Linux package](https://archlinux.org/packages/community/any/cura/). I can try installing another way if you want.

guillaumechauvat, 2021-08-10:
Okay, so I tried using the AppImage and it gave some clues about what is going on.

I started launching but crashed with
```
ModuleNotFoundError: No module named 'keyring.backends.null'
```

It turns out that I had a file `$HOME/.config/python_keyring/keyringrc.cfg` that contained:
```
[backend]
default-keyring=keyring.backends.null.Keyring
```

After removing that file, the AppImage launched. It failed in a different way when I clicked on "sign in":
```
/usr/bin/xdg-open: line 613: 595893 Segmentation fault      (core dumped) kde-open${KDE_SESSION_VERSION} "$1"
```

However now the Cura version installed from the Arch Linux package manager works fine!

fvrmr, 2021-08-13:
Thank you for taking a look at it. I will bring this up for discussion with the team. Keep you posted!

konskarm, 2021-08-18:
That was a really helpful and detailed report @guillaumechauvat! We do not support the Keyring on Linux (since we'd have to ship a specific keyring package with it) and it wasn't explicitly ignoring local Keyring configs. I've fixed it now to make sure that it won't load any keyring backends on Linux https://github.com/Ultimaker/Cura/pull/10302. 
For our internal tracking, devs see ticket CURA-8490.

Ghostkeeper, 2021-08-19:
This is fixed by #10302 for Cura 4.11.

closed 2021-08-19 by Ghostkeeper