Ultimaker/Cura #10253: Wall protrudes to the inside at certain layer change

Triangulix, 2021-08-08:
### Application Version

4.8 + 4.10

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

I designed a model in Fusion 360 and sliced it with Cura 4.8.0. Later I confirmed the issue with Cura 4.10.0.

### Actual results

In the layer view I noticed that there is a sudden expansion of all walls at a certain layer and upwards. There is no such change in the model and Meshmixer also doesn't show anything suspiciuos.

### Expected results

The walls should go straight up without any protrusion.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[Festplattengehause 2.1.3mf.zip](https://github.com/Ultimaker/Cura/files/6950921/Festplattengehause.2.1.3mf.zip)

GregValiant, 2021-08-08:
So when it a hole not a hole?  All inside features below layer 41 describe a hole.  From layer 41 to layer 125 there is not a hole.   From 126 to 820 constitutes a "Hole".  From layer 821 up is not a hole.  You have "Hole Horizontal Expansion" at 0.27 and so the features in the areas where the inside and outside are separate Hole Features are enlarged which pushes out the walls by .27.  The areas where the inside line and the outside line are connected do not get HHE because those areas aren't defined as holes.  

The same thing happens with a spoked wheel.  The gaps between the spokes are "holes" and so Hole Horizontal Expansion acts on them as well as on the axle hole which is typically the only thing that we want it on.  You've come across the same thing.

Turn off HHE for the model.  Create a support blocker and scale it to X200 Y100 Z2.0 then place it at 0,0,0.  With the support blocker selected - choose "per model settings" and "modify settings for overlaps".  Choose "select settings" and scroll to "HHE" and select it.  Set it to .27.  The Hole Horizontal Expansion will only effect the holes on the bottom of the model and the upper area will be unaffected.

Triangulix, 2021-08-08:
Thanks GregValiant, this seems to be the cause of this issue.

closed 2021-08-08 by Triangulix