Ultimaker/Cura #10254: 4.9-4.10 does not open stl files

rv0arm, 2021-08-08:
### Application Version

4.9

### Platform

Windows 10

### Printer

Flying bear ghost 5

### Reproduction steps

running the program

### Actual results

I give the command to download the STL
[Il-2 aileron L1.zip](https://github.com/Ultimaker/Cura/files/6951061/Il-2.aileron.L1.zip)
 file

### Expected results

the program thinks and closes
[cura.zip](https://github.com/Ultimaker/Cura/files/6951063/cura.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.zip](https://github.com/Ultimaker/Cura/files/6951064/cura.zip)
There was no problem in 4.8 versions.
I did not install plugins at all

rv0arm, 2021-08-08:
I have no Chitubox.
There is also nothing that caused such a failure, which was written about earlier.
Additional file DxDiag 
[DxDiag2.zip](https://github.com/Ultimaker/Cura/files/6951111/DxDiag2.zip)

fvrmr, 2021-08-09:
Hi @rv0arm this is a duplicate of https://github.com/Ultimaker/Cura/issues/9644
For the next release we have a fix that Cura doesn't crash anymore.

Here is what I found in your log:
`2021-08-08 22:31:39,293 - ERROR - [JobQueueWorker [3]] UM.Logger.logException [116]: OSError: Failed to open file b'C:\\Users\\\xd0\xba11234\\AppData\\Local\\Temp\\scipy-i5sd0_iz'`

The temp folder doesn't have access because of non-Latin characters in your Windows username. A workaround could be changing the path of the %temp% folder in the Windows environment variables configuration screen.

I will close this issue as a duplicate.

closed 2021-08-09 by fvrmr

rv0arm, 2021-08-09:
This is a Windows bug.
When buying a laptop, a salesperson in a store entered a test name at the entrance.
I changed the name of the laptop to the one that is registered with microsoft after that.
But the folder remained under the test name even after changing the name and account on the laptop.
But I did not feel problemm putting bought licensed software.
You cannot change the folder name manually and through the registry
The purchased programs will fly off and I will not restore the licenses again without additional payment.
The price exceeds the cost of a laptop to restore licenses due to such a folder name change.
--------------------------
This is a known bug windose that they have not changed, and many publishers avoid this problem.
I only ran into this with CURA 4.9-4.10...
Honestly, there have never been problems.
I can't change the folder name, I don't want to enter a new user on my laptop either, and it's not a fact that this will be a solution to the problem.
I really hope that you will find a way to solve this problem, you are the first who began to conflict with me.

Ghostkeeper, 2021-08-17:
This is a bug in Scipy, a library that we're using in Cura. This library is fairly commonplace so you might encounter it in other Python-based applications too, especially if they deal with geometry like Cura.
We've added a workaround for Cura 4.11 that prevents the crash, and hope to be able to contribute a real fix to the library upstream if they want to have it.

rv0arm, 2021-08-17:
Thank you for not staying indifferent even to such sometimes non-standard situations.
I am a beginner user, but I love your products that allow you to make very neat details.
I am interested in exploring the possibilities of your program.
I'm looking forward to the new version to start working with it.
Dest Regards/
Evgeny.

Ghostkeeper, 2021-08-24:
@rv0arm The 4.11 beta is now released if you want to give that a spin, see if you can load files now.

fvrmr, 2021-08-31:
Thank you for reaching out. It seems like we missed an other place where this crash happens. 
We have changed this for the stable release: https://github.com/Ultimaker/Uranium/commit/0be417ac2f690e09acd8d396ce5b50ee90ae91bc

The reason why we missed this is because we couldn't reproduce/test this bug ourselves.

rv0arm, 2021-08-31:
Thanks.
Tell me where you can download the corrected version for the test?

Ghostkeeper, 2021-09-07:
Unless you can build Cura yourself, you'd have to wait until we release the 4.11 stable version.
The release is planned shortly, so we don't need to wait for long.

fvrmr, 2021-09-07:
Hi @rv0arm 4.11 is live now: https://ultimaker.com/software/ultimaker-cura
Let me know if it works for you!

rv0arm, 2021-09-07:
Good afternoon..
The program is closed again.
And again, it cannot find the folder when opening the file.
[cura.zip](https://github.com/Ultimaker/Cura/files/7122941/cura.zip)

fvrmr, 2021-09-09:
Thank you, I will bring this up with the team.

fvrmr, 2021-09-10:
I have discussed it with the team and I made a ticket on our backlog for this. Further updates will be in the other thread (https://github.com/Ultimaker/Cura/issues/9644) so we keep everything in one thread.

Ghostkeeper, 2021-09-15:
Developers, see issue CURA-8416.