Ultimaker/Cura #10255: USB connection to printer: Multiple connections in consecutive prints / all stay open + message when closing Cura to close each USB connection

RipperDrone, 2021-08-08:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. print over USB from Cura (PC, Win10x64)
2. after printer finishes, print next job over USB (+ maybe more)
3. close cura

### Actual results

--> Cura will throw a dialog message asking for each of the print jobs to close USB connection. If you print 3 times same file over USB, Cura will ask you 3 times to close USB:

[image]

### Expected results

After terminating a print job, the next print job should NOT trigger a new, separate USB connection, Cura should close without throwing errors / dialog box asking to close previous print jobs which have completed long before.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Untitled.zip](https://github.com/Ultimaker/Cura/files/6951577/Untitled.zip)
[cura.zip](https://github.com/Ultimaker/Cura/files/6951580/cura.zip)

fvrmr, 2021-08-09:
Hi @RipperDrone I will discuss this with the team. Keep you posted!

fvrmr, 2021-08-12:
I have discussed it with the team and we decided to defer this bug, since it is not a priority for us to fix.
I hope you understand.

RipperDrone, 2021-08-12:
@fvrmr what a pity, this makes USB connection unstable, after printing 3-4 jobs, USB suddenly auto- disconnects. Maybe too many USB ports open simultaneously. We cannot trust the connection like this, it needs fixing... 😟

closed 2023-06-10 by RipperDrone