Ultimaker/Cura #10256: Blockers are even to the plate when opening project and the object not any more

DaLimboster, 2021-08-09:
### Application Version

4.10

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

1. Open an STL
2. Create Blockers which are not even to the plate (negative Z)
3. Save project
4. (may close Cura)
5. Open project 

### Actual results

The Object has Offset to the plate, the Blockers are now even to the plate

### Expected results

Object even on the plate, blockers with negative offset as original

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-08-09:
Hi @DaLimboster can you share a project file (file - save project) where this is happening.

DaLimboster, 2021-08-09:
I added the 3mf-project-file als a png to upload it here :-)

[image]

fvrmr, 2021-08-09:
Could you zip your 3mf file? Png doesn't work

DaLimboster, 2021-08-10:
[3D4U_Coffee_Clip_Large_V2.zip](https://github.com/Ultimaker/Cura/files/6959409/3D4U_Coffee_Clip_Large_V2.zip)
Sorry, didn´t realize, that it takes zips. Now as zip. Thanks

fvrmr, 2021-08-13:
Hi @DaLimboster thank you, I can now reproduce your issue. And this seems to happen because automatically drop models to the build plate is enabled. You can disable this as a work around for now. (Preferences - configure Cura)
See also this ticket: https://github.com/Ultimaker/Cura/issues/10184 and there we linked a ticket on our backlog to solve this. 
I will close this one as a duplicate. Hope you understand.

Ghostkeeper, 2021-08-17:
This is a different issue than #10184, because the issue here is not that the model is below the build plate (to cut off the bottom side from the model). The issue is that some objects are support blockers and they need to stay aligned to the model. For #10184 the intended solution is to make sure that dropping models to the build plate doesn't move models up, but still drops them down. That solution won't solve this issue, because these support blockers are still above the build plate.

konskarm, 2021-08-23:
Hi @DaLimboster . I've been trying to reproduce your issue on 4.11-beta and I am unable to. Could you give it a try and let me know if it works?

konskarm, 2021-08-23:
This issue is now on our backlog (Devs see CURA-8511).

konskarm, 2021-08-23:
Note: I think that https://github.com/Ultimaker/Cura/issues/9973 has the same root cause as this issue.

DaLimboster, 2021-08-24:
Thank you very much

closed 2021-11-05 by nallath