Ultimaker/Cura #10260: "Open Folder" after "Save to Disk" does not work anymore

Duke-Jones, 2021-08-09:
### Application Version

4.10.0

### Platform

Manjaro (current) with KDE Plasma (current) 

### Printer

Ender 3 Pro

### Reproduction steps

Slice something and press "Save to Disk.
Then Press button "Open Folder" on the shown dialogbox.

### Actual results

Nothing happens. If I open Cura from the command line I can see following message:

`/usr/bin/xdg-open: Zeile 613: 20165 Speicherzugriffsfehler  (Speicherabzug geschrieben) kde-open${KDE_SESSION_VERSION} "$1"
` 

### Expected results

Current folder in the file-tool (in this case Dolphin) should be opened.

[cura.log](https://github.com/Ultimaker/Cura/files/6954991/cura.log)

I

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

At the beginning of the year, this already worked. 
However, the error has been occurring for a few months.

fvrmr, 2021-08-10:
Hi @Duke-Jones I will bring this up with the team. Keep you posted!

fvrmr, 2021-08-12:
Hi @Duke-Jones do you use the AppImage of Cura?

Duke-Jones, 2021-08-12:
Hi.
Yes, I do.

fvrmr, 2021-08-12:
Ah, we don't maintain those. Here you can find the appimage from 4.10: https://github.com/Ultimaker/Cura/releases/tag/4.10.0
Could you try if this one doesn't have your issue?

Duke-Jones, 2021-08-12:
Ok. 
I uninstalled Cura, downloaded and installed the appimage from Github.
But nothing changes.

fvrmr, 2021-08-13:
Hmmm, can you share your log from the appimage?

Duke-Jones, 2021-08-13:
Sure. Where can I find it ?

fvrmr, 2021-08-13:
Here: `$HOME/.local/share/cura/<Cura version>/cura.log`

Duke-Jones, 2021-08-14:
[cura.log](https://github.com/Ultimaker/Cura/files/6986227/cura.log)

fvrmr, 2021-08-17:
I can't see anything in your log. 
Could you try to use GNOME instead of KDE?

Duke-Jones, 2021-08-17:
I am sorry. I am glad that KDE runs reasonably.
Switching to GNOME is too much for me with my level of knowledge.

Ghostkeeper, 2021-08-18:
Yeah switching windows managers is very intrusive. It should run on KDE just as well as on GNOME. Some of our developers are on KDE and some on GNOME. No crashes with these reproduction steps for us.

The error itself indicates that a segfault occurred in a script in your computer, in `/usr/bin/xdg-open` on line 613. This script is part of your system, not part of Cura. It is a script that is supposed to open paths and determine the correct application to do that with. Since it's part of your computer, I don't think there's a lot that Cura can do to prevent this crash. It seems like a bug in Arch to me.

Of course, Cura might call the xdg-open script in a different way to workaround the problem. However Cura's own code doesn't call upon this script either. It's called by the Qt framework. The code we have under our control is simply this:

    QDesktopServices.openUrl(QUrl.fromLocalFile(message._folder))

(Where `message._folder` is the path to the folder where the file was just saved, e.g. `/home/ghostkeeper/Projects/Slices`.)

There's very little we can control here. Couldn't call this function differently. We can only control the URL we provide to that function. But to debug that, we need to test what URLs is giving your system script problems.

According to [this question on StackOverflow](https://stackoverflow.com/questions/27061701/qdesktopservicesopenurl-fails-if-path-contains-spaces-after-apt-get-upgrad), some xdg-open scripts don't deal well with spaces in the path. However in your log it appears like you saved the g-code file to `/home/duke/Dokumente/3D/g_code/Lagerring.gcode`, which contains no spaces so that couldn't be it.

You could try to save the g-code to a folder without any underscores, maybe? But I wouldn't think that would give your xdg-open script any problems.