Ultimaker/Cura #10264: Plugin USB PRINTING bug.

devidblejn, 2021-08-10:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

all

### Reproduction steps

When you connect an arduino device, the USB printing plug-in connects it to the program, even if the printer is already connected

### Actual results

it is not possible to work with other devices on the arduino while printing

### Expected results

it is necessary to add several buttons/ drop-down menus/ checkboxes to the plugin in the monitoring menu. 1) disable the port. 2) select a port manually from the list of available ones 3) select a permanent port and remember 4) disable auto-search for ports.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

pleaze

GregValiant, 2021-08-10:
No Ultimaker printers use USB Printing so it is unmaintained.  You can disable it in the MarketPlace or you can attempt to change the python files to suit your situation.

devidblejn, 2021-08-10:
if I or other users had this knowledge, they would not have asked for help from the developers. for you, it's a matter of one hour for me, it's at least a couple of days to figure out the syntax. If your program positions itself as a program for ultimaker printers, for which a list of supported printers is constantly being added. If your printers do not support printing via USB, maybe you will add this feature to your printers and modify this plugin?)

GregValiant, 2021-08-11:
I don't have anything to do with Ultimaker or Cura.  I do a lot of work with serial communication though.  I've read enough posts to know that Ultimaker doesn't want to put any more time and money into USB printing because their printers don't use it.

fvrmr, 2021-08-12:
Hi @devidblejn it is indeed correct what @GregValiant mentioned. Disabling the USB printing plugin in would work, the reason why this is not disabled by default is because it is working for most people. And indeed since Ultimaker printers doesn't use USB printing we don't have any plans to update or change this.

See also https://github.com/Ultimaker/Cura/issues/5207

I will close this issue, I hope you understand.

closed 2021-08-12 by fvrmr

devidblejn, 2021-08-12:
ленивые жопы

nallath, 2021-08-12:
> ленивые жопы

We have google translate and I hope you enjoy your ban ;)