Ultimaker/Cura #10268: extrusion fails part way through print

jewalt, 2021-08-11:
### Application Version

4.10.0 and 4.9.x as well

### Platform

Windows 10

### Printer

Maker Select V2

### Reproduction steps

1. Slice the attached files (both source and gcode files are attached) and right around layer 40, the extrusions stops. 
[extrusion_failuer.zip](https://github.com/Ultimaker/Cura/files/6970137/extrusion_failuer.zip)
[Uploading extrusion_failuer.zip…]()

### Actual results

Around layer 40 of the print, the extruder jams and will not extrude from that point forward.

I graphed the extruder position from the gcode and I see a strange flat spot in the extrusion position right around this time.  I have seen the same problem in 4.9.x and I have never seen the problem in 4.8.0 (which I have reverted to in the meantime).  When I graph the extrusion length in the 4.8 sliced gcode file I see a very consistent slope until the extruder position is reset.  What I am assuming is going on is there may be too many retractions close together and the filament jams in the heat break.  When I notice the failure, I have to disassemble the printer or at least remove the heatsink to get the filament to flow again.  As I said, I do not see this problem in the 4.8.0 version.  I have an older printer, but it is usually rock solid and I have only hit this problem when slicing with 4.9.0 and 4.10.0 using the same settings.

### Expected results

The extruder should not jam and prints should succeed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Also, I have just noticed that the 4.10 gcode estimated print time is 1 hour longer than the 4.8 version.  I have not verified the estimate against real time or the octoprint estimate (which is quite good) so it may be the estimating is more realistic in 4.10.

I have attached project files for v4.10 and 4.8. (the 4.10 project file does not quite match the gcode file above as I had to regenerate the project by placing the objects in a similar location on the print plate)
[project files.zip](https://github.com/Ultimaker/Cura/files/6970135/project.files.zip)

GregValiant, 2021-08-12:
I looked at the gcode file.  There are about 106,004 extrusions in the file.  There are 355 retractions.  Because you have Max Combing Distance without retract set to "0" there are no retractions for any combing moves.
That doesn't seem like an inordinate amount of retractions.  The retract distance of .3 seems very reasonable for a DD machine.
_"I see a strange flat spot in the extrusion position right around this time."_  I don't understand that.  The extruder "position" is a function of the length of any extrusion.  Short line lengths mean short extrusions and consequently shorter E distances to add to the cumulative total.  There are about 10,000 lines of gcode in layer 39 and only 18 retractions.  Layer 40 isn't much different.
There are settings like "Retraction minimum travel" that will limit retractions.  You have it set to .8 but there just aren't very many retractions in the file.
Maybe someone from the Cura team will have a better take on it.  One retraction every 300 extrusions doesn't seem like it's overly frequent and it is responding to your settings.  I really don't see any basis in the gcode for a clogged hot end.
Within the 3mf file you have your material as PETG.  The print temp is 215 and the bed temp is 60.  Those are really close to my PLA temps.  I print PETG at 240 with the bed at 80.

jewalt, 2021-08-12:
Thank you for taking a look at this @GregValiant.  I've included 2 pictures that show what I was trying to say with the "strange flat spot".  To create the two graphs I extracted all of the extrusion commands from the gcode file and plotted them in a graph with the total extrusion distance on the y-axis and the "line" number on the x-axis.  I also started the graphs at the G92 E0 right after layer 33 as the layers before this were fine.   Some movements are longer so those will obviously have a larger step from the previous value and retractions will result in a negative slope, but there are so many extrusions that the small differences between each extrusion length and the retractions really are not visible.  What I would expect is a consistent upward slope to the line with slight variations as the extrusion type and x and y speed changes across the layer (this the extrusion amount per gcode line).  You can see this slope with variation in both pictures, but there is an elongated flat spot in the middle of 4.10 graph that shows the extrusion rate has significantly changed (assuming the x and y movement is constant for the layers at this point).  I'm not sure this graphing is any help, but I think it does point out that there is something going on with the extrusion rate starting in layer 39 as the entire flat spot on the graph is in layer 39.  There are 8225 extrusions for a total of 385.24765mm in layer 39 and 1819 extrusions for 303.10774mm in layer 38 which is a very similar layer.  I don't know if I have the tools to analyze the differences between the layers but I will help in any way I can with providing debug or testing new code.  
[image]
[image]
 the movements and extrusion between layer 38 and 39, but it seems something is very off here.

jewalt, 2021-08-12:
One more comment. I was printing PLA even though the configuration says PETG.

fvrmr, 2021-08-13:
Hi @jewalt the files in project files.zip are not the same. So that could be the cause that the print is not identical and the time estimation is different.
For example in 4.10 is ironing enabled.

GregValiant, 2021-08-13:
I have an Ender 3 Pro that I keep finely tuned.  It is running well.  I use the latest version of Cura (currently 4.10) exclusively for my own projects.  I have not noticed any issues either slicing or printing the files.

My plan was to print layers 35 thru 45.  Layers 35 through 38 were fine.  During gcode layer:39 (Cura layer 40 @  Z=8.0) when the "ironing" started there appeared to be massive over-extrusion to the point that "puddles" showed up in the print and my extruder started skipping steps.  On your DD machine this may have produced a back-flow of molten plastic upwards due to the high pressure in the nozzle which in turn may have caused your blockage.  I don't know that is true but it makes a certain kind of sense.
I used old wet PLA for the first test.  I printed the layers again using fresh PLA with the same results but I have to revise my conclusions regarding the extrusion amount.

I think the problem is in your Ironing settings (or possibly the way Cura is applying those settings?).  The print head is bulldozing a lot of plastic and there is just no place for the new plastic to go.  This "bulldozing" also leaves some mountains on the part.  (As an aside, my nozzle did plug up on the second test.)

BTW - prints I have done since my little tests have been fine.
The beige print was the first one.  The ridges that the Ironing caused are clearly visible on both pieces.  

[image]

jewalt, 2021-08-13:
@GregValiant, I think you may have found my issue.  It must be the ironing setting.  I had thought I was using the same configuration in v4.8 as in v4.10 but now I see I have several differences.  I'm not sure how that happened as I rarely change settings but I must have turned ironing on at some point to try it.  That would explain all of the moves with very little extrusion (10% is probably too high).  Thank you for your help and I will close this issue as user error in settings and not a Cura problem.

closed 2021-08-13 by jewalt

GregValiant, 2021-08-13:
If I hadn't printed it I would have never seen it.  There were so many data points on my graph in Excel that it was just a blob so it was useless.  This is the kind of thing I keep scrap rolls of filament for and I think this time it worked out.