Ultimaker/Cura #10270: Ability to assign profiles to materials

Tech0verlord, 2021-08-12:
### Is your feature request related to a problem?

So I've noticed that in Cura, your custom profiles are available no mater what material you have currently selected. As more custom profiles are made, whether it be brand specific, or print specific, they all show up in the list regardless of material currently selected or what material was active when the profile was made.  So now, if I have a profile I have made specific to a filament from a given brand, I have to sort through all my custom profiles to find it.

### Describe the solution you'd like

I would like the ability to assign profiles to specific materials. It would help no only sort through profiles easier, but also help with clutter in the profile selection dropdown.

### Describe alternatives you've considered

I've checked through the settings have had no luck finding the ability to do so. Right now the only option is to dig through your profiles to find the one you want.

### Affected users and/or printers

Anyone who has specific prints with specific needs and needs to be able to repeat the print when needed, especially if the print has a lot of changes different from the defaults, and already has a bunch of other profiles they use that have been tuned to specific filament brands.

### Additional information & file uploads

[image]

fvrmr, 2021-08-13:
I have discussed it with the team and we decided to defer this. Since it would be very complex to implement, since this would be easy for printers with one extruder, but for two extruders it would be difficult. I hope you understand.

Ghostkeeper, 2021-08-19:
> So I've noticed that in Cura, your custom profiles are available no mater what material you have currently selected.

This is not entirely true. Your custom profiles are available as long as the quality level they were built on (usually identified by layer height) is available for the material you have currently selected. If you built the profile on top of a 0.2mm quality profile, then the custom profile will be available for all materials that also have a 0.2mm profile available.

This availability of quality levels is global; the quality level must be available for all extruders. If the left extruder has a 0.2mm profile available but the right extruder has a different material and thus doesn't have that, then the 0.2mm quality level is not available and the custom profile will not be selectable.

Custom profiles are also global; You can't select one custom profile in the left extruder but another in the right extruder. So this matches.

Depending custom profiles on individual materials would break that logic. Since individual materials can differ per extruder, this is problematic for determining which custom profiles are available.

hugoalves6, 2021-08-19:
Hello @Tech0verlord,
I understand you problem, but i think you are making the profiles wrongly

Your using the profiles for the material, but the base of the profile is based on the layer height.

I've opened a suggestion that can help you too
  #[https://github.com/Ultimaker/Cura/issues/10275](url)

With this option and the "Material Settings" plugin, you specify on the material Options Tab, what and which settings to show and you can change them at your will, if my suggestion is implemented, at each profile, no matter the nozzle number, because each nozzle can have a separate profile that use specified material settings if the user wants to, with a click of a button like a material icon button, that would make that field use the material settings the user defined.

closed 2023-07-13 by github-actions[bot]