Ultimaker/Cura #10275: Ability/Option to use material setting instead of default or "changed by user"

hugoalves6, 2021-08-12:
### Is your feature request related to a problem?

No it's not a problem is a sugestion.

### Describe the solution you'd like

I'm using several types of materials and every time I change material, settings like temperature, flow, fan speed, retraction, etc, are always in need to change, since I don't know that much about codding to make a plugin to do it.
I'm sure that are more people like me, and it would be a fantastic feature to have like a icon like a spool, to use the material setting and that would grey out that field (or even just saying in the field the option like in the config file "material_speed_print", "material_flow", etc), so that the user would know that it is being used the material setting.

Like me more users use the plugin to add more material settings, that helps a lot for this kind of things if this plugin/option was added to the software, it would improve many things.

### Describe alternatives you've considered

I haven't found any alternative.

### Affected users and/or printers

It will affect everything and everyone, for better and easier use of the program.

fvrmr, 2021-08-13:
Hi @hugoalves6 thank you for your feature request.
I have discussed this with an colleague and we've decided to defer this request. The reason why is adding the solution would make Cura slower and that is something we don't want.
You could create profiles for materials, but those are not linked to each other. 
I hope you understand.

hugoalves6, 2021-08-18:
Hello @fvrmr, i get your point, but i don't think that a simple button next to the "TextField" like we have the rollback and link icon would make the program that much slow, and it would help everyone for sure.

Hope to hear news back, thank you.

Ghostkeeper, 2021-08-19:
Note that we already have this in our backlog, as CURA-6722. Back then it was determined (by all the developers) that it would make Cura faster since we could then remove the "reset to inheritance formula" button.

hugoalves6, 2021-08-21:
And isn't there a possibility to make a plugin, so that only those who want that option to activate it, even if it makes the cura "slower"

Ghostkeeper, 2021-08-25:
Kind of. The buttons next to each settings are not plug-in modifiable. There is no API function to add buttons there. You could replace the entire prepare screen with a customised version (like what the Settings Sidebar plug-in does) but that is prone to mistakes and will quickly run behind the main interface if we make modifications to the GUI.

TheModernWizard, 2021-10-04:
Ok, this would be amazing. Please add this to Cura. Couldn't it just be a toggleable setting if it slowed the program down? I don't understand how this is going to slow Cura down significantly alongside all the other new features being added each update.