Ultimaker/Cura #10279: Rotation ring arrows have only one direction

3Duke3, 2021-08-13:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

All

### Reproduction steps

I'm not sure if this is a bug, or perhaps how it's just intended to be. In which case then its a feature request!..

But since the addition of the arrows included in the 3 coloured rotation rings to rotate the model in 90deg increments, this has bugged me for some time...

### Actual results

Load any model into Cura, Select Rotate from left menu and the Green, Red, Blue rotation rings appear around the model. Each ring has two arrows, but regardless of which arrow is clicked, the model only rotates in one direction.

### Expected results

Click one arrow to turn clockwise, click other arrow to turn anti clockwise.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-08-13:
Hi @3Duke3 thank you, I will bring this up with the team. Keep you posted!

fieldOfView, 2021-08-13:
> I'm not sure if this is a bug, or perhaps how it's just intended to be.

Yes, this is a bug. I'm the one who implemented them, it is not how I intended this functionality. I'll have a look (too).

3Duke3, 2021-08-13:
@fieldOfView I was so glad when I saw it got implemented, I use it all the time so a big thanks to you. But I was never sure if it was only meant to rotate one way not. Its not a biggie, but every time I click it, there is a 100% chance it rotates the opposite of what I want lol.

fieldOfView, 2021-08-13:
> every time I click it, there is a 100% chance it rotates the opposite of what I want 

I guess I managed to implement it in the same way as USB-A plugs.

fieldOfView, 2021-08-13:
Fixed in this PR: https://github.com/Ultimaker/Uranium/pull/724

nallath, 2021-08-13:
This has been fixed for the 4.11 release! The beta will be released in about a week.

closed 2021-08-13 by nallath

Ghostkeeper, 2021-08-20:
> every time I click it, there is a 100% chance it rotates the opposite of what I want lol.

I had this problem too, so I resorted to clicking 3 times blindly. From that moment on, it turns out that the initial rotation direction was actually correct and I would have to click 5 times.

3Duke3, 2021-08-20:
Thanks all for fixing, superstars. 

I have a feeling it’s been that way since is was first implemented in a Cura update, so was never sure if maybe it was an issue, or how it was intended :)

anyway thanks

nallath, 2021-08-20:
> Thanks all for fixing, superstars.
> 
> I have a feeling it’s been that way since is was first implemented in a Cura update, so was never sure if maybe it was an issue, or how it was intended :)
> 
> anyway thanks

The beta is out, so you can test it for yourself now :)