Ultimaker/Cura #10280: Import/Export Issue

dboverheim, 2021-08-13:
### Application Version

4.10.0

### Platform

Mac

### Printer

any

### Reproduction steps

- Attempt to import or export more than 1 filament profile
- Program will not allow second import/export
- Close management window, re-open, try again with no luck
- Reopen app allows import/export of a second profile

### Actual results

Program does allow import/export of a second filament profile. Only closing and restarting of program allows another profile to be imported/exported. Must be done for each subsequent import/export

### Expected results

Should allow for as many import/exports of filament profiles as wanted

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

When trying to import filament profiles in 4.10.0, only able to import one profile at a time. Even closing the printer management window and re-opening it does not allow for a second profile to be imported. The app has to be quit and restarted to import another profile.

fvrmr, 2021-08-16:
Hi @dboverheim can you share your materials files and your log file? You can find your log file here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

dboverheim, 2021-08-16:
I can’t seem to find the log files… but here are the filament profiles

fvrmr, 2021-08-17:
Can you add the files on Github? I can't see them since you've replied via email.

Ghostkeeper, 2021-08-23:
The "import" button is enabled if the printer supports having materials at all. This shouldn't change over time.

The "export" button is enabled if any material is selected. To make sure this button is enabled, be sure to select a material in the list of materials that you want to export.

But yeah, we really do need that log file to see what's going on.

fvrmr, 2021-12-02:
I can reproduce this issue on MacOS Monterey thanks to issue #10965

fvrmr, 2021-12-07:
I have discussed this with the team and we think this is related to Qt. We are planning to upgrade Qt and have a ticket on our backlog to check if the upgrade of Qt will fix this issue. 
Devs see CURA-8739

MariskaMaas, 2022-06-10:
Hey @dboverheim did you have a chance to try Cura 5.0? 
Are you still seeing the same behavior?

bjoernbusch, 2022-06-11:
Issue seams to be resolved in Cura 5.0.0. I can export multiple materials one after the other without having to restart Cura.

MariskaMaas, 2022-06-13:
Thank you for reporting back to us, @bjoernbusch 😄 . I'll be closing this ticket.

closed 2022-06-13 by MariskaMaas