Ultimaker/Cura #10281: no G  Code

2ndAmend, 2021-08-13:
### Application Version

4.10

### Platform

iOS

### Printer

ender 3 v2 pro

### Reproduction steps

sliced image no errors recorded or shown .
saved to sd card . 
no file found except . printable that came on sd card .
removed factory given code . re inserted Gcode from factory " test " print and printer reads it.
done this before stopped then started working. now doing it again. can't seem to get G code to appear on printer again. 
plz help 

### Actual results

info on settings updating ?

### Expected results

an idea of what to look for on why this happens 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I tried different profiles or images and can't seem to slice and save as G code

GregValiant, 2021-08-13:
If the file name is too long it may not appear on the LCD screen.  Keep file names to 16 characters before the .gcode.
There is an option under "Preferences | configure Cura | Open and Saving files" called "Add machine prefix to job name".  Turn that off.

fieldOfView, 2021-08-13:
> Platform
> iOS

Cura does not run on iOS. Did you mean Mac OS? What version?

fvrmr, 2021-08-16:
Hi @2ndAmend could you give some more information about your reproductions steps? Does the gcode not appear on your sd card? Did you try to save to disk in Cura? Did saving to your sd card give you a notification that it is saved correctly?

2ndAmend, 2021-08-17:
Ok . Sorry . I haven’t responded quicker . It was one of the plug ins . I removed all and now the G coding is working again either I need to update my ender or my computer don’t agree with the plug in . Running Apple computer. And I noticed some of the plug ins wasn’t for Apple and had to remove.

2ndAmend, 2021-08-17:
Again ty everyone sorry if I’m not talking the proper lingo and at the moment I’ve written this I was frustrated and confused . Then I deleted the software and started over and I noticed it worked without the plug ins added from the market place . Again thanks for all help

fieldOfView, 2021-08-18:
It would be really helpful to know *which* plugin, or at least a list of plugins you had installed.

fieldOfView, 2021-08-18:
and a log file will help us get the plugin fixed.

closed 2021-09-01 by no-response[bot]