Ultimaker/Cura #10282: Cura 4.10.0 Pause at Height/Layer corrupts line width and layer height on resuming print with large models.

Jallen813, 2021-08-14:
### Application Version

4.10.0

### Platform

Windows 10 64bit

### Printer

Custom HyperCube running Marlon or RepRap 

### Reproduction steps

1. Added a Pause at Height/Layer # so I can change the filament color.
2. After Resuming Print the extruded lines are not the correct line width or layer height. Print fails due to inconsistent extrusion.
3. If I open the GCODE in Cura you can see the corrupted section at the next layer after the pause. Several layers are affected.
4. This seems to occur on larger volume prints (example x-294mm  y-110mm  z-22mm)

### Actual results

[image]
[image]

### Expected results

You can see the blue sections that are the areas where I have a Pause at Layer Number 130 and another Pause at Layer 160.
The printer basically starts skipping areas that appear to have been corrupted by this action. This model is uniform and has no defects. If I remove the Pause options, the print looks normal and complete.
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project File,
[CC Bottom Name Plate v2.zip](https://github.com/Ultimaker/Cura/files/6985782/CC.Bottom.Name.Plate.v2.zip)

GregValiant, 2021-08-14:
Pause At Height is a "post-processor" and Cura has completed the gcode before Pause at Height goes through it and adds a few lines.  There is no change to the gcode that follows the new code inserted by Pause At Height.

When you sliced the model you had "Relative Extrusion" turned on.  When the Pause at Height routine runs, about the last thing it adds is M82 which tells the printer it is in "Absolute" mode.  The consequence is that all the E values in the rest of the file are interpreted as absolute positions and so the extruder just rocks back and forth.

This has been noted before.  I don't know how the Cura team is handling it, but there are three workarounds.
1. Turn off "Relative Extrusion" in Cura.
2. Open your gcode file in a text editor and search for M226.  About 8 lines below will be _"M82 ; switch back to absolute E values"_.  You can change the M82 to M83 and it will work.  Do it for each instance after a pause.
3. If you always use Relative Extrusion you can alter PauseAtHeight.py to always leave in relative extrusion mode.  Near the end of the file is "_prepend_gcode += self.putValue(M = 82) + " ; switch back to absolute E values_" and you just have to change the "82" to "83".  That means if you later go to Absolute Extrusion in Cura that after a pause you will have the problem you have now (all the following extrusion commands will be interpreted incorrectly).

Jallen813, 2021-08-14:
Hello Greg,

Thank you for your quick response and the solution to work around this
issue.

Sincerely,
Jim Allen

Jallen813, 2021-08-16:
Some additional info. 
1. Using a layer height of 0.2 mm, I can use the “Pause At Height” post process option up to layer 66, or a height of 13.2 mm and not experience the issue. Above this height and the bug will happen.
2. After examining the GCODE per GregValiant’s instructions, the referenced M82 line was already set to M83 for Relative mode.
3. Doing a GCODE comparison of previous instances of “Pause At Height” that work correctly against those instances above layer 66 or higher than 13.2 mm which introduce extrusion errors, there does not appear to be any difference in the code.

GregValiant, 2021-08-16:
To post a Cura Project - file load your model and use "File | Save Project".  Zip the 3mf file and a gcode you feel has an error and post the zip file here.

Jallen813, 2021-08-16:
[Bottom Plate with Pause at Height errors.zip](https://github.com/Ultimaker/Cura/files/6989413/Bottom.Plate.with.Pause.at.Height.errors.zip)

Jallen813, 2021-08-16:
[Bottom Plate with Pause at Height errors.zip](https://github.com/Ultimaker/Cura/files/6989427/Bottom.Plate.with.Pause.at.Height.errors.zip)

GregValiant, 2021-08-16:
I sliced the 3mf without making any changes and got this:
M226 ; Do the actual pause
M109 S208 ; resume temperature
G1 F1500 E30.0
G1 F1500 E-30.0
G1 F9000 X98.172 Y125.978
G1 F300 Z26.2 ; move back down to resume height
G1 F1500 E30.0
G1 F3600 ; restore extrusion feedrate
**M82 ; switch back to absolute E values**
G92 E0.00455
;LAYER:130
;TYPE:FILL
;MESH:CC Bottom Name Plate v2.stl
It is the 8th line after every occurrence of M226 in the file.  That line must be an M83 if you are going to use relative extrusion when you slice a model.

Jallen813, 2021-08-16:
Thanks Greg for having a look. What’s odd is When I opened the file in Note
Pad ++ , it shows that line as M83.
I’ll have another look at it tomorrow. Thanks again for your quick response
and help.

GregValiant, 2021-08-16:
It occurs to me that although a printer, settings, and model come within a 3mf file - Cura is probably using my installation of PauseAtHeight.py which exits with M82.  Whenever I use it I must use Absolute Extrusion or the gcode after the pause doesn't work.
It sounds like your installation of PauseAtHeight exits with M83 and so all the following gcode-after-pause should run as well as the code before the pause did.

@Jallen813 post one of those bad gcode files here.  Maybe it's something else other than the M82/M83 thing and maybe it will jump out.  As I say, since PauseAtHeight is a post-processor it doesn't have an effect on the gcode that Cura developed...it just inserts some lines so the print pauses and resumes seamlessly.  (At least that is the plan.)

Jallen813, 2021-08-16:
[Bottom Plate with Pause at Height errors.zip](https://github.com/Ultimaker/Cura/files/6995351/Bottom.Plate.with.Pause.at.Height.errors.zip)
Here is the GCODE file if you want to browse through and see if you spot anything unusual.
I did experiment with the M83 entry by editing it to M82 to see the results. It caused a lot of missing areas to appear on the model. 
The threshold for introducing the error starts at 13.4mm or 67 layers using 0.2 layer height or above. Very strange to say the least.

GregValiant, 2021-08-17:
Cura looks at the model, and slices it according to your settings.  The gcode is complete at that point.  When you Save or Print to USB Cura goes through and executes any post-processors that you have active and it does them in the order that they are listed.  As a post-processor PauseAtHeight goes through the gcode, finds the correct layer number, and inserts about 19 lines of code into the existing gcode "stack" in Cura.  That happens just before the gcode is saved to a file or printed.

Within the gcode file you posted - lines 107333 through 107352 (inclusive) and lines 142168 through 142187 (inclusive) were added by the two active PauseAtHeight routines.  There were no changes to any movement or extrusion commands that existed in the file.  The purpose of those added lines is to:  get the current location, send the print head to the park position, retract, pause, prime the hot end, and restart the print from the correct XYZE locations.  The  second to last line added is indeed M83 so that is good.
If you are having trouble at a Z of 13.2 then that is in a portion of gcode before the first Pause that occurs at Z=26.2.  The Pause is not involved at all.
Additionally, I don't see anything odd within the file.  No strange movement, the extrusion distances are reasonable, the pauses occur at the correct layers, I don't see anything that would cause the printer to stop extruding.
The physical size of the model means I can't do a test print, but after looking at the gcode I am confident that the PauseAtHeight plugin is not the problem.
How have your other prints been coming out?  Have you printed this model without pauses?

Jallen813, 2021-08-17:
Thanks @GregValiant for examining the GCODE file.
If I do not include any “Pause at Height” parameters, the model prints without issue. I know because I had to get this job finished for the client, and I did all pauses manually for the filament changes and encountered no issues.
I have another part of this job that also needed pauses above 13.4 mm (layer 67) in the model and yes it too encountered the same issues with”Pause at Height” (layers 44, 88, 102) which the first pause at layer 44 was good, but layer 88 failed.
For now I’ll continue to doing the pauses manually on any models that need a pause above layer 66. 
Thank you again for looking into this and your in depth explanation of how Cura preforms these slices.

GregValiant, 2021-08-17:
I hate when the problem isn't found.
I sliced the 3mf with both your pauses, and with no pauses.
I removed the 2 blocks of 19 lines of gcode that PauseAtHeight added.  I removed the line from the front of the file that says "post-processed".
I compared that file with the No Pause file.
The result was two identical files.  Not just nearly identical, exactly identical.
Have you tried using M0 or M24 to pause?

Jallen813, 2021-08-18:
Hey @GregValiant I think I have finally resolved the problem. I re-exported the file from Fusion 360 and sliced it again and recreated the "Pause at Height" post processing scripts and resliced it. 
As you can see from this preview photo, there are no more line width and layer thickness errors showing.
I have no idea what changed, even the other models I was having similar issues with also seem to be resolved. 
[image]
I will go ahead and close the ticket. Thanks again for your help and explanations about the post processing scripts.

GregValiant, 2021-08-19:
It's good that you are back up and running.
In the end...we still don't really know.  I hate that.

Ghostkeeper, 2021-08-23:
Well, the original project file still reproduces the issue, at least for me with unmodified Pause At Height script, looking at the line width in layer view, so we can figure this out. I'm putting this up for the team to discuss.

I hope we can figure out though what changes you guys made that fixed the issue. The Pause At Height script should already deal with relative extrusion.

Jallen813, 2021-08-23:
Update to team on our workaround.
I found that if I comment out the lines referring to Extrude Amount/Speed found in the "Pause at Height" settings, then the Layer and Line Width errors are resolved. (see attached image)
[image]
Since I manually reload and prime the nozzle, I do not need these commands. 
Also there is an extrude command (G1 E5 F3600) found in REPRAP resume.g file which runs when the printer resumes from pause. Possibly this introduces a conflict with the Cura routine?

GregValiant, 2021-08-23:
You caught me on that one @Ghostkeeper.  I had altered the defaults for "Disarm Timeout" and "Gcode Before Pause" and copied that 2019 version of Pause at Height into succeeding versions of Cura.  A fresh install of 4.11 shows that the 2021 version of Pause At Height does account for Relative Extrusion.

I'm feeling sheepish here.  Sorry for any confusion I caused.

Ghostkeeper, 2021-08-24:
:D @GregValiant No worries, we always appreciate all the help you're giving to get to the bottom of these issues. Even if you don't eventually succeed in finding the root cause.

The relative extrusion fix was a really recent change: https://github.com/Ultimaker/Cura/pull/9518

It's really weird that those 3 lines would cause this issue. It unretracts, extrudes 1mm slowly, then retracts again. I'm starting to think this is only a visualisation issue. Maybe it's somehow ending up on negative E values somewhere.
Adding some debug prints in the g-code visualisation reveals no actual negative widths, just a whole lot of 0's. But I can't debug it too much without actually planning in some time for it.

This morning it was discussed (without me as I was a bit ill) and they concluded that we'd try reproducing the issue again (though I already succeeded in reproducing the issue). It's on the radar in any case, but no real decision was made yet.

Ghostkeeper, 2021-11-05:
So the g-code itself looks fine. There is a small issue with that it's creating an unretraction and retraction in a row which is unnecessary if there's no extra prime amount to prime for. But other than that it's fine. I fixed that up quite quickly.

However it turns out that there is a g-code visualisation problem. The reduced line width effect is always for 6 layers in your print, regardless of how much it retracts. It turns out that this is because of the head lift movement it does, which moves the nozzle up by 1mm. This by itself wouldn't be a problem for the g-code visualisation. However because of the unretract extruding 30mm of filament, the g-code visualiser thinks it's actually extruding material at that height. As such it thinks that the following layers have a negative height, but catches that and caps the height to 0. The line width for the visualisation is calculated by dividing the volume of the line by its length and height. With the height at 0, this is a division by 0. It guards against that and uses a layer height of 0 instead.

Detecting the layer height from g-code is sometimes pretty hard. I can make the detection a bit more strict though. It currently considers it a layer if there is any E movement and the difference in Z is less than 1.5mm. I'm adding another requirement to that: The E movement needs to be accompanied with an X or Y movement, i.e. an extruded line rather than just a retraction/unretraction.

This does not fix any print quality issues. Jallen813 reported that it's also actually not extruding properly. However the g-code looks correct. Pausing often causes a temporary stop of extrusion after resuming. That's why there's the Redo Layer option and why you can have this extra prime movement after the pause. I suggest you use those to fix the actual issue. If they are not sufficient to fix the issue, we'll have to know what we can change to the g-code to make it work.

closed 2021-11-08 by nallath

Jallen813, 2021-11-09:
Thank you for your attention to this bug and creating a fix in the upcoming release.

PR #10751 (2021-11-08): Fix visualisation of layer view with pause at height enabled
We do retracts/unretracts on different heights sometimes, for instance after a pause. This would get seen as a new layer. It's quite safe to say that purely retracts and unretracts on a different height do not constitute a layer in a normal g-code file. There would be nothing to draw anyway.

There was also an unnecessary unretract/retract introduced by the Pause At Height script sometimes. The unretract/retract is only necessary if it's priming some extra material. Fixing that up should reduce ooze for those who don't need to prime. [... cut]
commit fd5b2482 (2021-11-08): Don't register a new layer if not extruding a line

We do retracts/unretracts on different heights sometimes, for instance after a pause. This would get seen as a new layer. It's quite safe to say that purely retracts and unretracts on a different height do not constitute a layer in a normal g-code file. There would be nothing to draw anyway.

Contributes to issue CURA-8522. Fixes #10282.