Ultimaker/Cura #10284: multiplying objects in cura 4.10.0

johanmeijberg, 2021-08-15:
### Application Version

4.10.0

### Platform

MacOS Monterey

### Printer

Creality ender 3 pro

### Reproduction steps

1. uploaded a file
2. selected the object
3. right clicked to multiply the selected object

### Actual results

The placing objects popup appears, then nothing

### Expected results

Cura should multiply the objects.
[cura.log](https://github.com/Ultimaker/Cura/files/6987883/cura.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6987884/cura.log)

fieldOfView, 2021-08-15:
Does this happen with all models, or a specific model only? If a specific model, could you please post a link to that model?

johanmeijberg, 2021-08-15:
It happens with all models, and cura 4.10.0 won’t even start up anymore

fvrmr, 2021-08-16:
Hi @johanmeijberg this is a duplicate of https://github.com/Ultimaker/Cura/issues/10249
I will close this issue as a duplicate.

closed 2021-08-16 by fvrmr