Ultimaker/Cura #10285: Multiply object not working

ylandot, 2021-08-15:
### Application Version

4.10

### Platform

MacOS

### Printer

Ultimaker 3

### Reproduction steps

1. put an object (any object)
2. select the object
3. try to multiply the object (1,2 ...)

### Actual results

Object doesn't multiply (only one remaining)

### Expected results

Object doesn't multiply (only one remaining)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

ylandot, 2021-08-15:
In this example i had to put manually 10 version of the object myself, the multiply function doesn't work and leave a message like it can finish the process ...

jellespijker, 2021-08-15:
Possible duplicate of #10284 
@ylandot which version of MacOS are you running?

nallath, 2021-08-16:
Could you also include the log files? They should give is more information as to what is going on.

ylandot, 2021-08-18:
Hi

I’m using a beta of MacOS (Monterey, beta 4 will update this week end to beta5) but have no problem with the same OS & functionnality with cura 4.8
it's stuck only with cura 4.10

Here is a log with 2 attemps at 10h25 and 10h29 today (2021-8-18 ) 

Sincerely

Yann

nallath, 2021-08-18:
Unfortunately, attachments are stripped by GitHub if you send them in by email. Could you attach them via the web interface?

jellespijker, 2021-08-18:
Also definitely a duplicate of #10249 so I'm closing this bug report. Please continue the discussion there.

ylandot, 2021-08-18:
OK, thanks

closed 2021-08-18 by fvrmr