Ultimaker/Cura #10286: Conveyer (belt) slicer

kirk-loeten, 2021-08-15:
### Is your feature request related to a problem?

For belt printer is no actually version of Cura available. Blackbelt3d has an old version of Cura 3.6 to slice the programs it doesn't work on Mac OS Big Sur.

### Describe the solution you'd like

Integration of the functions form Blackbelt Cura 3.6 in 4.x

### Describe alternatives you've considered

I work with the version 3.6 and don't update my OS

### Affected users and/or printers

All users with belt printer find this useful.

### Additional information & file uploads

https://blackbelt-3d.com
https://ifactory3d.com

fieldOfView, 2021-08-15:
I'm the creator of BlackBelt Cura. Blackbelt has stopped the development of BlackBelt Cura.

Ultimaker does not make a belt-style printer, so Ultimaker will not spend resources on developing this. I am a third party developer. I have been mulling over the idea of picking up where I left off and creating a plugin for Ultimaker Cura out of the work I did for BlackBelt Cura. However, I too need bread on my table, so (many) other projects get priority at the moment. Having said that, you can sponsor my work here: https://github.com/sponsors/fieldOfView/

fvrmr, 2021-08-16:
Hi @kirk-loeten thank you for your feature request. 
@fieldOfView is correct since we don't have a belt-style printer we will not develop this and are depend on community members to develop this.

I will close this issue, hope you understand.

closed 2021-08-16 by fvrmr