Ultimaker/Cura #10287: Allow printing the bottom skin before walls for better holes in PETG

skandalfo, 2021-08-16:
### Is your feature request related to a problem?

When printing PETG with small holes on the bottom layer, it's very difficult to get the "hole walls" to stick at all. Even with "inside brims" enabled, these short runs of PETG usually end up dragged by / stuck onto the nozzle, and the first layer quality becomes horrible.

"Optimize wall printing order" can help to make a longer line by keeping adjacent walls printed together, but usually that's not enough for small holes. Sometimes slowing the print helps a little (maybe using "Small Feature Speed" for holes), but it's by no means a safe bet.

This bug is related to requests in #7324, #7724 and #9110, but I'm not sure I've seen this solution proposed elsewhere.

### Describe the solution you'd like

PETG sticks more easily on already-printed big surfaces. While #9110 centers on supports and brim order, but for my usual use case (bottom layer is a biggish surface of my model with holes in it) I think a simpler solution is possible.

I'd request to be able to request at least the bottom skin to be printed *before* the walls, so that the hole walls have a bigger surface to stick to already adhered to the bed.

Then I could get my use case served with these settings:

 * NEW! (Bottom Layer) Print Skin Before Walls: TRUE
 * Outer Before Inner Walls: FALSE
 * Optimize Wall Printing Order: TRUE ?
 * Brim Only on Outside: TRUE (already have the outer brim for priming, and the hole walls would have the skin to stick to).

Assuming there's a good reason to normally print walls first, I think it's fine to have this setting exclusively for the first layer. Once the first layer has stuck, the next ones are OK.

### Describe alternatives you've considered

As said before, I've played with reducing speed, small feature speed adjustment, wall print order optimization and first layer extra flow to no avail... small brims or walls end up as a ball rolled around the bed by the nozzle.

Another alternative would be to have all inner walls (outside of the model and holes) printed as multiple long continuous lines, connected together via skin-following lines, so that instead of short lines we have a whole "contiguous" segment. Example where two cuts along a skin diagonal are added for stability:

```
 ___________ ____
/         _//    \
\       /   \    /
 \      \_ _/   /   
  \      //    /
   \____//____/
```

But in general every hole (unless many can be connected via skin-aligned diagonals) would require partitioning the inner wall one more time. Also, two lines of skin on each side are less likely to provide reliable adhesion for the inner wall than the actual full surface of the skin having been printed beforehand.

### Affected users and/or printers

Anybody wanting to print glassy-flat bottom surfaces with small (screw) holes in them.

yet-another-average-joe, 2021-08-16:
2 tricks I've been using successfully. (I mainly print PETG)

- underextrude the 1st layer (95% works for me) and underextrude the rest of the model (98%). I never get any blob anymore since I decided to do this !

- if you have access to the CAD model (or design your models yourself, or are familiar with Meshmixer) : close all holes with a skin the same height that the first layer. The 1st layer will be continuous, giving a much better adhesion and aesthetics. It will require some post processing, but it is well worth the effort !

fvrmr, 2021-08-17:
Hi @skandalfo thank you for your feature request. 
I will bring this up with the team. Keep you posted.

Ghostkeeper, 2021-08-18:
See also https://github.com/Ultimaker/Cura/issues/2988, which was closed as a duplicate of https://github.com/Ultimaker/Cura/issues/6734. And https://github.com/Ultimaker/CuraEngine/issues/1171. Perhaps we should close this as a duplicate, to contain the discussion in a single thread?

fvrmr, 2021-08-18:
I will close this ticket as a duplicate of https://github.com/Ultimaker/Cura/issues/6734 we have a ticket on our backlog for this feature.

closed 2021-08-18 by fvrmr