Ultimaker/Cura #10290: Overhanging wall speed to be applied to inner walls attached to overhanging outer walls

brunoosti, 2021-08-16:
### Is your feature request related to a problem?

Only if you want to print faster and don't have the cooling capabilities to deal with cornering overhangs. 

### Describe the solution you'd like

Overhanging walls speed works very well, but it is only implemented on outer walls. If you want to print inner walls faster, they won't change velocity at overhangs, causing bad cooling an pulling the outer walls when it slowly cools, ruining the effect of printing outer walls overhangs slowly.

### Describe alternatives you've considered

The overhang wall speed should be applied to outer and inner walls to avoid cooling issues. Consequently, the setting unit shouldn't be %, but mm/s, as you already might be printing outer walls at the desired speed (lets say 30mm/s to avoid ghosting and have better surface quality) and want to print the faster inner ones (100mm/s),  but, at overhangs , the same outer slow speed to avoid cooling issues. 

### Affected users and/or printers

Most cura users. 

### Additional information & file uploads

Thank you very much! : )

fvrmr, 2021-08-17:
Hi @brunoosti thank you for your feature request.
I will bring this up with the team. Keep you posted.

brunoosti, 2021-08-17:
Thank you very much! : )
I understand that there might be a flow issue in reducing speeds too much in a single segment, but the advantage might suppress this collateral effect (more present some Bowden extruders and firmware without linear advance, lets say) as it would be possible o overcome cooling issues for overhanging corners in most machines. The way it works right now you have to slow down outer and inner walls anyways to compensate for cooling, defying the purpose of this setting.  
Thank you again!

konskarm, 2021-08-20:
Hi @brunoosti. This sounds more like a bug than a feature request, since the Overhanging Wall Speed should be applied to both outer and inner walls.

Still, I am having a bit of trouble reproducing the issue. I tried the following:

1. Outer Wall Speed = 10mm/s
2. Inner Wall Speed = 30mm/s
3. Overhanging Wall Speed = 200%

The result in the following picture shows clearly that both the outer wall and the inner wall speeds have been doubled in the overhang (Outer Wall Speed = Green = 20mm/s, Inner Wall Speed = Orange = 60mm/s).
[image]

My project file that reproduces the issue can be found in [overhang_#10290.zip](https://github.com/Ultimaker/Cura/files/7021725/overhang_.10290.zip)

What is going on differently on your side and the overhanging Wall Speed is not applied? Could you share a project file?

brunoosti, 2021-08-20:
I think the issue is when the inner wall is not really overhanging above air  (with absolutely no support). Did the same experiment below with a benchy. All 30mm/s and over 40º 200%. Printing that inner wall at slower speeds could prevent curling of the outer wall below it. But, in your example, the complete segment of the inner wall is affected, even if it is supported by the layer below. I'm confused. 
[image]

Ghostkeeper, 2021-08-23:
This project file uses an unknown machine definition `artillery_b_sidewinder_x1`. I think you downloaded that from somewhere, and modded Cura to put that in. Without that same machine definition we can't reproduce the slice.

> But, in your example, the complete segment of the inner wall is affected, even if it is supported by the layer below. I'm confused.

The Overhanging Wall Speed is indeed applied to both the inner and outer walls. However it doesn't _break a wall into multiple segments_ to apply the overhanging wall speed to only a part of a wall segment. That is what you're seeing with Konskarm's model. It's printing too much too slow because there's just one long line segment of which about half is overhanging. (Also you're seeing an old bug that renders the colour wrongly with a gradient, which was fixed in 4.10 iirc.) The code itself seems to mark it as overhang if one of the endpoints is overhanging, so in theory it could also happen that both endpoints are supported but the middle is not, and then the middle will not be printed at the adjusted speed.

As I understand now, your request is that if the outer wall is overhanging, any inner walls _next to it_ should also be seen as overhanging and have the overhanging wall speed applied to them too, even though the inner walls themselves are not overhanging.

brunoosti, 2021-08-23:
Oh sorry! that is only a Sidewinder X1 definitions I created for myself. Attached a new project with the actual X1 profile from cura.
[overhang.speed.3mf.zip](https://github.com/Ultimaker/Cura/files/7032035/overhang.speed.3mf.zip)

"As I understand now, your request is that if the outer wall is overhanging, any inner walls next to it should also be seen as overhanging and have the overhanging wall speed applied to them too, even though the inner walls themselves are not overhanging." EXACTLY! : ).  And as a defined speed (mm/s) to not slow down walls that are already being printed slow).
To prevent overheating of the previews layer. In my case, with the benchy, if I don't print the inner wall at 30 mm/s it curls up the previews layer, ruining the overhang at the front of be boat's hull, though everything else prints just fine at 100mm/s.  
Konskarm example is what I'm aiming for, but his object does not represent a real world situation. There, supports would be used anyway, because, no matter what speed, those walls corners would be printed midair, with high chance of failing. If you notice on the overhang test beside the benchy, the setting only applies the slower speed to the walls that have no support at all (not very practical, as you'd need supports anyway), but, the way I see it, that could be used to prevent cooling issues making the overhang speed a more versatile setting, improving printing quality and overall speed. 
 
[image]

konskarm, 2021-08-25:
@brunoosti while we can see the value of this feature, we decided to defer it since we have other, more pressing issues in our ever-growing backlog. I hope you understand.

brunoosti, 2021-08-25:
Sure! I understand. Add it to the future to do list : ) and get Arachne pully up and running : D. 
I think this can really have a great impact in printing quality as an easy adjustable setting can compensate for most hardware. (In my case I was able to update my printer's fan)
Hope someday Cura gets this update. Thank you!

brunoosti, 2021-09-03:
Adding to the mix "small feature speed", also a % that could be set as mm/s. Using a benchy again, if you do not print walls thick enough, Cura tends to compensate some portions of overhangs by adding top/bottom layers (of course, it wants to keep with whatever thickness had been configured). But small portions like the "arrow head" in front of the boat tend to overheat the already printed walls, curling them up. Maybe, if we could set a defined speed compatible with our cooling fans (instead of a % of a faster top/bottom speed), we could avoid overheating the area. Even more useful for those who have "top surface skin layers" to print the intermediate ones faster. Food for (future) thought.
[image]

closed 2023-07-13 by github-actions[bot]