Ultimaker/Cura #10291: Additional nozzle size

Pukwudji, 2021-08-16:
### Is your feature request related to a problem?

Increasingly nozzles are sold with a 0.25mm option, but this is not an option within Cura 4.10.0 that I can see.  My knowledge of editing the definition files is very basic so I do not know how to do this myself.
I am using a Tronxy XY-2 Pro (but using a modified Ender 3 Pro profile),  and generic PLA.

### Describe the solution you'd like

Either an additional value in the drop-down list or the ability to configure nozzle sizes in either the printer or extruder settings

### Describe alternatives you've considered

Selecting .3mm nozzle and reducing line width to .25, but this seems messy!

### Affected users and/or printers

All

yet-another-average-joe, 2021-08-16:
Workaround (what I've been doing for years) :

Just create as many printers as you have nozzles. Of course, you will have to copy/paste printer definition, start and end gcode. Then assign them various nozzle diameters.
Works really fine, and super easy.

It seems the feature does exist (one printer and various nozzles that can be selected in a dropdown list box. But edtiting configuration files is not easy, and I never found any tutorial, nor any information about the object model used in the definition files...

GregValiant, 2021-08-17:
Some of this stuff you have to jump through hoops for.  I'm a Windows user so make adjustments for Linux or Mac.
Your printer definition file "tronxy_xy2pro.def.json" inherits the base Tronxy definition file "tronxy_x.def.json".  Both have their own associated nozzles and they have to match.
By adding an additional nozzle file for each printer definition you can get the .25 as an option for your printer.  The nozzles are in "...Ultimaker Cura 4.10.0\resources\variants".
Make a copy of the nozzle file "tronxy_x_0.2.inst.cfg" in the Variants folder.  Rename it "tronxy_x_0.**25**.inst.cfg".
Open the file in a text editor and alter the two lines containing .2 to read .25 (the new nozzle size).  Save the file.
Make a copy of the nozzle file "tronxy_xy2pro_0.2.inst.cfg" in the Variants folder.  Rename it "tronxy_xy2pro_0.**25**.inst.cfg".
Open the file in a text editor and alter the two lines regarding the nozzle size.  Save the file.
When you re-start Cura the .25 nozzle should be available.  I make no guarantees.
When you update Cura you would need to manually copy the new files over as they won't be included in the install.

Pukwudji, 2021-08-17:
> Some of this stuff you have to jump through hoops for. I'm a Windows user so make adjustments for Linux or Mac.
> Your printer definition file "tronxy_xy2pro.def.json" inherits the base Tronxy [...]

Thanks this has worked perfectly, I didn't realise you had to do both the base profile and the model profile. 

Cheers, Steve

closed 2021-08-17 by Pukwudji

Ghostkeeper, 2021-08-23:
Cura provides a text box to change the nozzle size if there are no nozzle profiles available for the printer. If there are nozzle profiles, it'll show the list of nozzle profiles instead in the material/nozzle dropdown.

This does mean that if there is a set of nozzle profiles, those are the only nozzle profiles the user can select from.

A workaround that doesn't involve file hacking would be to download the Printer Settings plug-in from the Marketplace which allows you to change the nozzle diameter as a normal setting.