Ultimaker/Cura #10292: Crashes on start up.

TelliMantelli, 2021-08-16:
### Application Version

4.10

### Platform

Windows 10 64bit

### Printer

Cant even get there

### Reproduction steps

Start program.

### Actual results

When Cura Loads GUI crashes

### Expected results

Program should start

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Log file - I have uninstalled all version, have reinstalled gfx driver..issue still remains.
2021-08-16 09:48:31,475 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-08-16 09:48:31,486 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-08-16 09:48:31,488 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-08-16 09:48:31,489 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2021-08-16 09:48:31,491 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.10.0
2021-08-16 09:48:31,491 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49676
2021-08-16 09:48:31,493 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2021 Ultimaker
2021-08-16 09:48:31,494 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-08-16 09:48:31,496 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-08-16 09:48:31,497 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-08-16 09:48:31,499 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-08-16 09:48:31,501 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-08-16 09:48:31,502 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-08-16 09:48:31,504 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-08-16 09:48:31,506 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-08-16 09:48:31,509 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2021-08-16 09:48:31,510 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2021-08-16 09:48:31,512 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-08-16 09:48:31,513 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2021-08-16 09:48:31,516 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2021-08-16 09:48:31,516 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdater
2021-08-16 09:48:31,522 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzReader
2021-08-16 09:48:31,528 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeGzWriter
2021-08-16 09:48:31,534 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeProfileReader
2021-08-16 09:48:31,544 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeReader
2021-08-16 09:48:31,550 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin GCodeWriter
2021-08-16 09:48:31,557 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ImageReader
2021-08-16 09:48:31,563 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-08-16 09:48:31,568 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalContainerProvider
2021-08-16 09:48:31,575 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LocalFileOutputDevice
2021-08-16 09:48:31,581 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MachineSettingsAction
2021-08-16 09:48:31,587 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ModelChecker
2021-08-16 09:48:31,594 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin MonitorStage
2021-08-16 09:48:31,601 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PerObjectSettingsTool
2021-08-16 09:48:31,607 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PostProcessingPlugin
2021-08-16 09:48:31,613 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-08-16 09:48:31,619 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PreviewStage
2021-08-16 09:48:31,627 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin RemovableDriveOutputDevice
[... 103 lines cut ...]
2021-08-16 09:48:34,248 - DEBUG - [MainThread] cura.CuraApplication.run [851]: Booting Cura took 4.467816591262817 seconds
2021-08-16 09:48:34,249 - DEBUG - [JobQueueWorker [1]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed
2021-08-16 09:48:34,366 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2021-08-16 09:48:34,368 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [74]: No active GlobalStack, set quality profile model as empty.
2021-08-16 09:48:34,369 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2021-08-16 09:48:34,371 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [38]: No active GlobalStack, set CustomQualityProfilesDropDownMenuModel as empty.
2021-08-16 09:48:34,521 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [210]: Support for Vertex Array Objects: True
2021-08-16 09:48:34,528 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [112]: Initialized OpenGL subsystems.
2021-08-16 09:48:34,530 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [113]: OpenGL Version:  4.1.0 NVIDIA 471.68
2021-08-16 09:48:34,533 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [114]: OpenGL Vendor:   NVIDIA Corporation
2021-08-16 09:48:34,534 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [115]: OpenGL Renderer: NVIDIA GeForce GTX 1650 Ti/PCIe/SSE2
2021-08-16 09:48:34,536 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [116]: GLSL Version:    4.0.0
2021-08-16 09:48:34,538 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\default.shader]...
2021-08-16 09:48:34,542 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\selection.shader]...
2021-08-16 09:48:34,547 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\select_face.shader]...
2021-08-16 09:48:34,550 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\default.shader]...
2021-08-16 09:48:34,554 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\composite.shader]...
2021-08-16 09:48:34,559 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\overhang.shader]...
2021-08-16 09:48:34,564 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\striped.shader]...
2021-08-16 09:48:34,566 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\transparent_object.shader]...
2021-08-16 09:48:34,570 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\striped.shader]...
2021-08-16 09:48:34,574 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\xray.shader]...
[... cut ...] d] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\xray_composite.shader]...
2021-08-16 09:48:34,581 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\platform.shader]...
2021-08-16 09:48:34,590 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [59]: Loading shader file [C:\Program Files\Ultimaker Cura 4.10.0\resources\shaders\xray.shader]...

fvrmr, 2021-08-17:
Hi @TelliMantelli thank you for your bug report. 
This is a duplicate of https://github.com/Ultimaker/Cura/issues/7744

As a workaround you can edit `%appdata%/cura/4.10/cura.cfg` and add the line `opengl_version_detect = force_legacy` under the `[view]` header.

I will close this issue, but let me know if this works for you!

closed 2021-08-17 by fvrmr

TelliMantelli, 2021-09-08:
This did work thank you!