Ultimaker/Cura #10293: Arachne beta: Support layers missing

mcer12, 2021-08-16:
### Application Version

arachne beta

### Platform

Windows 10

### Printer

Creality CR10

### Reproduction steps

There is a similar issue regarding gaps in supports (but it focuses on tree support):
https://github.com/Ultimaker/Cura/issues/8218
To me this happends with normal zig-zag support. It seemingly happends randomly. Moving the model and re-slicing doesn't help.

### Actual results

Gaps in supports in arachne

### Expected results

Cura 4.9.1 doesn't have this issue using the same profile.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Arachne:
[image]

4.9.1:
[image]

Project file (saved in arachne):
[rails.zip](https://github.com/Ultimaker/Cura/files/6993617/rails.zip)

fvrmr, 2021-08-17:
Hi @mcer12 thank you. I can reproduce your issue. 
I will bring this up with the team. Keep you posted!

fvrmr, 2021-08-18:
I have discussed this with the team. And I have created a ticket on our backlog. 
Devs see CURA-8492

mcer12, 2021-08-18:
@fvrmr thank you, I only use arachne beta unless I need supports. It's much better than the normal version.

nallath, 2022-03-04:
This has been fixed for the 5.0 release!

closed 2022-03-04 by nallath