Ultimaker/Cura #10295: Cura 4.10.0 Crashes when opening file prompt

Meagleson88, 2021-08-17:
### Application Version

4.10.0

### Platform

Windows 10 64

### Printer

Creality CR-10S Pro V2

### Reproduction steps

Click Open File. Crashes.

### Actual results

Cura locks up and crashes immediately.

### Expected results

Should open prompt window and let me select file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Meagleson88, 2021-08-17:
Well I did some more reading and found out that Chitubox was infecting it. Uninstalled Chitubox and now it works as should. But now I don't have a software for my Elegoo....

nallath, 2021-08-17:
You don't have to uninstall it. Check: https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

closed 2021-08-17 by nallath