Ultimaker/Cura #10299: Overlap setting does nothing with Support Blockers [Arachne]

Theguffy1990, 2021-08-17:
### Application Version

Arachne

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Brought in a model, printed, had issues with top layer not connecting so went to increase the overlap percentage beyond 10% with a testing process involving sequential printing, and support blocker modification. When this didn't work, I verified that it also did not work with just a single model in "all at once" printing.
[image]

### Actual results

Nothing. It's almost as though the default of 10% is not even being followed and it's stuck at 0%. Attempted infill mesh and cutting mesh.
[image]
[image]

### Expected results

Expected to increase overlap, tested to extremes from 50, 100 and even 300 percent.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Program crash on save project, Log file included (I never save projects, this is a non-issue for me).
[cura.log](https://github.com/Ultimaker/Cura/files/7003394/cura.log)

nallath, 2021-08-18:
I've just tried this with arachne. I can't seem to reproduce your issue.

Can you check if this project reproduces what you mean? 

[image]

[OverlapTestProject.zip](https://github.com/Ultimaker/Cura/files/7006061/OverlapTestProject.zip)

Theguffy1990, 2021-08-18:
> I've just tried this with arachne. I can't seem to reproduce your issue.
> 
> -snip-

Ahh, apologies for incorrect terminology on the title there @nallath. 

It would seem that this is definitely a "me" issue, and I am really unsure what the problem is resulting from. Your Project file did not work (it did not crash at any point either). Would there be any changes if I were to open the project fresh and save it again providing it doesn't crash? It almost feels as though something is conflicting with it, but this was tried with my version that has a moderate amount of usability plugins, but also with a clean install.

Again, sorry for the incorrect title, and the new knowledge of what to refer to them as.

nallath, 2021-08-19:
The project should be the same if you open & save it again. One main difference between our setups is that I was using the nightly build of Arachne. We just completed a full sprint of fixing things in Arachne, so it could be that we already fixed the issue that you're encountering.

Ghostkeeper, 2021-08-25:
One way of disambiguating that would be if Theguffy1990 could also save his project, to allow us to reproduce his case. It could be that Nallath's test case was missing some override that Theguffy doesn't know could influence this result. If Theguffy would go to File -> Save project... and share that file here, we should be able to reproduce their case precisely.

And if it doesn't reproduce then, we can (more) confidently say that the issue has been fixed in the past half year.

Theguffy1990, 2021-08-25:
Apologies, I hadn't noticed an update to this. 

I stated in the Additional Information that when I try to save that specific kind of project, it crashes outright. 32GB of ram (24GB free usually when trying to replicate), I7-8700K, trying to save to an NVME, SSD, SSHD and HDD, all fail to save and crash. That's why I included my most recent log file with the original post. There was not much information I could extrapolate from it, likely because I did not know what I should be looking for, so left it to the more experienced to either tell me I had screwed up somewhere, or that there was a genuine bug there.

I doubt I have any right to ask, but curiosity can't be helped; when would the latest version of Arachne be expected? I use it exclusively now, and would be interested in knowing if/when my workflow will be altered.

Thank you again for keeping me in the loop,
Guffy.

Ghostkeeper, 2021-09-03:
Thanks for those reproduction steps. They don't cause Cura to crash for me though. I've been loading [this cube](https://www.thingiverse.com/thing:1278865) which might not be identical though, and with "modify mesh cubes" I'm assuming you meant support blockers changed to be a modifier mesh.

closed 2024-12-09 by github-actions[bot]