Ultimaker/Cura #10300: Preview after Slicing and Preview of saved GCode do not match!

eGit, 2021-08-18:
### Application Version

4.7 and 4.10.0

### Platform

Linux

### Printer

Ender3

### Reproduction steps

I sliced/printed for days now without problems. Suddenly I start a new print and the extruded lines of the first layer are super thin and don't stick. I abort the print and assume a printer problem. After calibrating everything and doing a calibration print (some old GCode I have on my SD Card), I am satisfied. This is perfect.

I start the print again: first layer lines are super thin and don't stick. I run the calibration GCode again. Everything perfect as can be. Lines have the perfect thickness. No over or underextrusion. So I print again. Thin lines. I think it is a problem with the SDCard. Damaged GCode or something. New SDCard. New slicing in Cura. Same problem.

Now I load the generated GCode in Cura. Strange! First layer lines are super thin (see screenshot) but ONLY the first layer. The second layer looks like completely overextruded. The rest of the layers looks fine.

I slice the STL again and preview the GCode. Everything looks perfect. No thin lines on the first layer. When I save the GCode, the generated GCode is broken again. I test an old version (4.7) of Cura now. Same problem! Cure 4.10.0 and 4.7 should be completely isolated from each other. It's two different Appimages and I think they store their settings at different places. Also I was slicing and printing all day long without issues!

I will restart my computer now. Maybe this fixes it for some reason. Still, I wanted to document the problem since this cost me two hours and a lot of nerves to track down and I think other people would exchange their extruder or something.

[image]
[image]
[image]

### Actual results

Lines of the first layer were printed too thin.

### Expected results

Normal printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]

eGit, 2021-08-18:
Maybe this old problem from 2019 is related: https://community.ultimaker.com/topic/28462-first-layer-line-gap

In the screenshot you can see, that the lines are thinner in the GCode for the first layer. Is this a long standing bug in Cura people always blamed on bad printer calibration or extruder problems?!

eGit, 2021-08-18:
Strange times. I nuked all the settings, cache files and configs of Cura from my harddisk.

/home/egit/.config/cura
/home/egit/.local/share/cura
/home/egit/.cache/cura
/home/egit/.cache/Ultimaker B.V./cura

It should ALL be gone. In theory. I restart Cura. I have to setup everything from scratch. Still the problem persists. No I am out of ideas. Please help. Don't make me reinstall the OS ;)

eGit, 2021-08-18:
Now tested it on a different account on Linux. Copied over the Cura appimage and an STL. Setup Cura/Printer. Same problem. I am going nuts! Linux accounts are separated. This makes zero sense! Is it some underlying Python update or something automatically installed during a package update so it is the same for all users?

nallath, 2021-08-18:
> Strange times. I nuked all the settings, cache files and configs of Cura from my harddisk.
>
> /home/egit/.config/cura
> /home/egit/.local/share/cura
> /home/egit/.cache/cura
> /home/egit/.cache/Ultimaker [...]

Correct. Nuking the OS won't add anything to removing those files

> Now tested it on a different account on Linux. Copied over the Cura appimage and an STL. Setup Cura/Printer. Same problem. I am going nuts! Linux accounts are separated. This makes zero sense! Is it [...]

The appimage should run in complete isolation. It should use the python version included in the appimage. System updates (apart from things like GPU driver updates) should not influence it. 

I think that this shows that it's not something that _suddenly_ changed, but has been happening for quite some time. It could even be that the preview has always been broken and that it accidentally shows the same issue as that you have on your printer. 

Anyhow; what happens if you load the g-code in another gcode viewer?

eGit, 2021-08-18:
The GCode looks similarily wrong in another gcode viewer (PrusaSlicer). First layer lines too thin. Second layer lines too thick. Layer3-to-Last-Layer looks good. Same as in Cura.

BUT I think you are right and the GCode generated by Cura prints correctly. It just displays wrong and I was wrongly assuming that my print problem was a problem with Cura, since the preview looked to match my problem. But to be absolutely sure I will have to do further tests later today.

Still, even if the GCode is printed correctly, it should also display correctly in Cura and other viewers like Prusaslicer. The question is, why doesn't it display this way when it then later prints correctly?

nallath, 2021-08-18:
It could well be that it's the gcode viewer that is incorrect. It's hard to say :(

eGit, 2021-08-18:
Can somebody please check quickly if they have some same result? Just slice a random STL, save the GCode and the drag&drop the saved GCode on the Cura window to load it. Then observe/screenshot the first and second layer and post the results here.

fvrmr, 2021-08-18:
Hi @eGit I tried and I could reproduce it on an Ender 3, but it didn't happen on an Ultimaker 3 both on Windows.

eGit, 2021-08-18:
@fvrmr thanks for the effort. This really helps narrowing down the problem. I tested on Linux on the latest and a few older versions of Cura with the Ender3 and Ender3 Pro. I always found the problem.

I think we should also note again, that this is not only a problem with Cura's preview. The same GCode is displayed like this in Puraslicer identically. So it is most likely a GCode problem - unless Cura and Puraslicer share some "preview" or "GCode parsing" code?

Ghostkeeper, 2021-08-25:
Your reproduction steps are not steps to reproduce the issue, but more of a story of what you observed. The actual reproduction steps are in [a comment](https://github.com/Ultimaker/Cura/issues/10300#issuecomment-900984072), so I'll move them to your initial post.

Ghostkeeper, 2021-08-25:
This is a problem we've known about for a long time in Cura's g-code parser. I'd expect that we already have an issue open about it, but it turns out we don't, at least not as far as I could find.

The g-code doesn't show the line width directly. We have to calculate what the line width is from the amount of material extruded, the length of the line and the height of the line. The last one is also not shown in the g-code though. Cura guesses the layer height from the first height at which material is extruded.

The Ender 3 has this bit in its start g-code, as part of its priming routine:

    G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line

Cura sees that and from that will guess that the first layer is 0.3mm high. However in actuality, the first layer is 0.2mm high, with the sole exception of this prime line. So the prime line will have the correct line width, but the rest of the first layer (the actual print) will be shown at 66.7% width because it calculates that the extrusion is not enough to fill a 0.3mm layer.

The second layer has the opposite problem. Cura sees that the layer is printed at Z=0.4mm, and since it thought that the previous layer is printed at Z=0.3mm, it'll think that the layer is only 0.1mm thick instead of the actual 0.2mm. So the line width will be calculated as 200% of the actual line width.

This is all merely a visualisation problem of the layer height -- and as result a visualisation problem of the line width. It will not actually print that way.

To see a problem in the print process caused by g-code, we'd need to know where in the g-code it's actually going wrong. As far as I can see with [this project file](https://github.com/Ultimaker/Cura/files/7046852/CE3_cube.zip), the g-code is fine and it's just a visualisation problem.

The decision to guess the layer height from the first extruded line is quite logical so it's possible that PrusaSlicer has made the same heuristic.

eGit, 2021-08-25:
Yes. I agree. It seems just to be a visualization problem. Still, many people encountering problems will first visually check the G-Code using Cura's parser and will see that layer1 and layer2 are not what they expected.

closed 2023-06-24 by github-actions[bot]