Ultimaker/Cura #10301: Cura is just eating memory

the-cobbler, 2021-08-18:
### Application Version

4.10

### Platform

Windows 10

### Printer

Cr-10

### Reproduction steps

Nothing special, tho i suspect its one of the plugins that make it bloat

It happens little by little over a period of a few days when printing (I suspect the octoprint plugin)

[image]

After closing Cura, it get really crazy and just want up fast even more
Even Though its probably a plugin, the plugin sdk should prevent it

### Actual results

High mem use

### Expected results

flat memory use

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-08-18:
As the creator of the OctoPrint connection plugin, is there any reason in particular you suspect my plugin? Personally, I rather suspect a plugin you may not be using, but is enabled (and active) by default: the UM3 Network Printing plugin. It uses an older version of the Python Zeroconf module, which is known to have a memory leak.

No plugin SDK can 100% prevent plugins from leaking memory.

the-cobbler, 2021-08-18:
It started happening after I started using the octoprint integration 
Just a speculation, nothing substantial

nallath, 2021-08-18:
The UM3 network plugin uses the latest & greatest zeroconf. We recently updated (but it still seems to have a mem leak)

the-cobbler, 2021-08-18:
Here is a list of all the plugins

Of course its also a speculation that its even a plugin

`14/07/2021  11:25    <DIR>          3MFReader
14/07/2021  11:25    <DIR>          3MFWriter
14/07/2021  11:25    <DIR>          AMFReader
14/07/2021  11:25    <DIR>          ConsoleLogger
14/07/2021  11:25    <DIR>          CuraDrive
14/07/2021  11:25    <DIR>          CuraEngineBackend
14/07/2021  11:25    <DIR>          CuraProfileReader
14/07/2021  11:25    <DIR>          CuraProfileWriter
14/07/2021  11:25    <DIR>          DigitalLibrary
29/06/2021  22:12    <DIR>          FileHandlers
14/07/2021  11:25    <DIR>          FileLogger
14/07/2021  11:25    <DIR>          FirmwareUpdateChecker
14/07/2021  11:25    <DIR>          FirmwareUpdater
14/07/2021  11:25    <DIR>          GCodeGzReader
14/07/2021  11:25    <DIR>          GCodeGzWriter
14/07/2021  11:25    <DIR>          GCodeProfileReader
14/07/2021  11:25    <DIR>          GCodeReader
14/07/2021  11:25    <DIR>          GCodeWriter
14/07/2021  11:25    <DIR>          ImageReader
14/07/2021  11:25    <DIR>          LegacyProfileReader
14/07/2021  11:25    <DIR>          LocalContainerProvider
14/07/2021  11:25    <DIR>          LocalFileOutputDevice
14/07/2021  11:25    <DIR>          MachineSettingsAction
14/07/2021  11:25    <DIR>          ModelChecker
14/07/2021  11:25    <DIR>          MonitorStage
14/07/2021  11:25    <DIR>          PerObjectSettingsTool
14/07/2021  11:25    <DIR>          PostProcessingPlugin
14/07/2021  11:25    <DIR>          PrepareStage
14/07/2021  11:25    <DIR>          PreviewStage
14/07/2021  11:25    <DIR>          RemovableDriveOutputDevice
14/07/2021  11:25    <DIR>          SentryLogger
14/07/2021  11:25    <DIR>          SimulationView
14/07/2021  11:25    <DIR>          SliceInfoPlugin
14/07/2021  11:25    <DIR>          SolidView
14/07/2021  11:25    <DIR>          SupportEraser
14/07/2021  11:25    <DIR>          Toolbox
29/06/2021  22:12    <DIR>          Tools
14/07/2021  11:25    <DIR>          TrimeshReader
14/07/2021  11:25    <DIR>          UFPReader
14/07/2021  11:25    <DIR>          UFPWriter
14/07/2021  11:25    <DIR>          UltimakerMachineActions
14/07/2021  11:25    <DIR>          UM3NetworkPrinting
14/07/2021  11:25    <DIR>          UpdateChecker
14/07/2021  11:25    <DIR>          USBPrinting
29/06/2021  22:12    <DIR>          VersionUpgrade
29/06/2021  22:12    <DIR>          Views
14/07/2021  11:25    <DIR>          X3DReader
14/07/2021  11:25    <DIR>          XmlMaterialProfile
14/07/2021  11:25    <DIR>          XRayView`

cbaker, 2021-08-24:
I've got a model open and sliced in Cura right now that is using 4GB of memory, I don't think this is abnormal or an issue.

Ghostkeeper, 2021-08-24:
> The UM3 network plugin uses the latest & greatest zeroconf. We recently updated (but it still seems to have a mem leak)

Actually, we just updated from 0.27 to version 0.31, which included a supposed fix for a memory leak but didn't seem to fix our issue with it. The latest is 0.36 as of this writing. We chose 0.31 because 0.32 was a bigger change.

GregValiant, 2024-11-18:
Is this still an issue in current versions of Cura (5.8.0 and up)?  Can this be closed?

JiMcNulty, 2024-11-18:
> Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

Close it :)

closed 2024-11-18 by JiMcNulty