Ultimaker/Cura #10303: Filling up the void instead of break-away

ds2077, 2021-08-18:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

The input file in attachment are sliced and the error is visual by checking the generated early layers and compare with the original input file. An example is shown in the png file.

### Actual results

There is a void in under the object is filled with print material instead support break-away material.
[image]
[gcode.zip](https://github.com/Ultimaker/Cura/files/7007380/gcode.zip)
[object 1 projectfile.zip](https://github.com/Ultimaker/Cura/files/7007381/object.1.projectfile.zip)

### Expected results

the early layers of the printed object has void that should have support material to support the layers higher up. Details visual in the cad file clearly show four pillars that are located in this volume. Instead they are replaced by a big block.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

n/a

nallath, 2021-08-18:
You have the "Make overhang printable" enabled, so it's doing exactly what you're asking it to do. 

If you disable that, it will print support as normal.

closed 2021-08-18 by nallath

ds2077, 2021-08-18:
Yes, you are correct. I disabled the "Make overhang printable" and now the slice looks as expected. Thank you.