Ultimaker/Cura #10310: Larger gap between supports and part

sbk1198, 2021-08-19:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I've had this happen a lot lately with the 4.10.0 version. Often times I'm seeing gaps between supports and the part of 2 or 3 layers, which causes the filament to have very poor adhesion or none at all and you get heavy stringing being dragged around. It's noticeable in the preview model after slicing going up layer by layer. In the attached project file, you can see where the last layer of the support ends (layer 8), but then the material that's supposed to go directly on top of the support only gets printed in layer 11. So at that point the nozzle gap is 3 times higher than it should be and the part doesn't come out right in that location because of that. I never had this problem with older versions.

[UMS5_bug_report_example.zip](https://github.com/Ultimaker/Cura/files/7016459/UMS5_bug_report_example.zip)

### Actual results

See above.

### Expected results

See above.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_bug_report_example.zip](https://github.com/Ultimaker/Cura/files/7016465/UMS5_bug_report_example.zip)

GregValiant, 2021-08-19:
The layer height is .15.  The Z gap is .3 (2 layers).
The support angle is 60.  A support angle of 45 allows for more coverage.  I made a couple of other changes to support settings.  See what you think.
[UMS5_bug_report_GV.zip](https://github.com/Ultimaker/Cura/files/7017176/UMS5_bug_report_GV.zip)

sbk1198, 2021-08-19:
Hello,

Thank you for the reply! I see that works better for the early layers,
however i went to 60 degrees because i wanted to avoid the supports at the
top side of the part because those will be problematic as well. See gap
between the part and the support in this screenshot. I've had this happen
in the past and the filament just oozes down since there's nothing directly
underneath to hold it. I have never had success with CURA placing supports
on steeper angle structures like in this example so i just try to avoid the
supports completely by increasing the angle. I know my printer is capable
of printing with decent quality up to 65 degrees without any support
material.

However support angle shouldn't make a difference when it's a flat
horizontal surface that needs to be supported. Regardless of the angle
selected, there shouldn't be a 2 layer gap between the part and support.

GregValiant, 2021-08-19:
That was my bad.  I was trying to get better support on the upper bosses.
This one has support blockers on the upper bosses and holes.  "Support Z Distance" = .15 so only layer 10 is empty of support.  I changed the support interface as well.  I think the grid at 75% will be easier to break away.
[UMS5_bug_report_GV2.zip](https://github.com/Ultimaker/Cura/files/7018101/UMS5_bug_report_GV2.zip)

sbk1198, 2021-08-20:
Oh I see. I didn't even have that setting on because it's not something
I've ever had to mess with. Not sure why it was set to 0.3 by default on my
computer (2 x layer height). On my home computer with my own personal
printer (different brand), it was set at 0.2 and my layer height is also
0.2, so that's why I've never experienced that issue. But at work on the
Ultimaker S5, I've been getting a lot of ugly surfaces above the supports
because of this. Now I know why! Thank you for the help. I'm very impressed
with the level of support received! I was not expecting to hear anything
back to be honest. I figured it was just a platform to report a bug and
hopefully something would be changed in the next version, but this has been
very helpful. Glad to see it's not an actual bug! :)

Thanks,
Alex

konskarm, 2021-08-20:
@GregValiant thanks for providing assistance to @sbk1198! It's great to see that we have such a helpful community.

Since this is not an issue, I am closing this.

closed 2021-08-20 by konskarm