Ultimaker/Cura #10311: Fails to import backup

hcker2000, 2021-08-19:
### Application Version

4.10.0

### Platform

windows 10

### Printer

Custom

### Reproduction steps

1. Open cura
2. Go into backup and tell it to restore a backup
3. Cura closes 
4. re-launch Cura
5. Cura asks me to add a new printer

### Actual results

Cura asks me to add a new printer

### Expected results

Cura would have loaded my printer and not have me re-create it

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

This has happened with quite a few versions of cura in the past
[cura.log](https://github.com/Ultimaker/Cura/files/7017724/cura.log)

fvrmr, 2021-08-24:
Hi @hcker2000 from which version do you want to restore a back up?
Could you share your configuration folder after you did the back up?
Please remove the `ultimaker_auth_data` in the cura.cfg file before sharing.

konskarm, 2021-08-30:
This issue should no longer happen in the 4.11 stable release since it has been fixed in https://github.com/Ultimaker/Cura/pull/10358.

PR #10358 (2021-08-30): Add the missing preference upgrade for 4.9 to 4.10
This could break backups in certain cases

CURA-8521

Fixes #10311
closed 2021-08-30 by konskarm