Ultimaker/Cura #10320: Max Volumetric Speed

johnkray, 2021-08-20:
### Is your feature request related to a problem?

Evey hotend has a maximum volume throughput it is capable of with each material.  Volume flow rate can be hard to keep track of with changing layer heights and linear print speeds. 

### Describe the solution you'd like

Addition of a Max Volumetric Speed setting to the material or speed settings section. This would be a hard cap such that linear speeds resulting in a volume flow rate higher than the volumetric max would not be allowed. This solution has been implemented in Prusa Slicer for a while. It means that you can be confident that if you have your Max Volumetric Speed set correctly, there will be no print moves that will exceed this max. 

### Describe alternatives you've considered

The alternative is manually calculating the max linear speed based on a volumetric max for every layer height and line width combination and manually adjusting the 10s of linear speed settings in Cura.

### Affected users and/or printers

All users of FDM printers could benefit from this

yet-another-average-joe, 2021-08-21:
Flow value is displayed, and it's max value as well. Trivial math with Print Speed will do the job.
This was a major enhancement that came with 4.9 (IIRC).

[image]

This being sayed, optional max volumetric speed field limiting the print speeds, or making their background color orange or red could be usefull.

johnkray, 2021-08-23:
Thank you, I am aware of this. While this is a useful visualization, it's a pretty inefficient solution to the problem I put forth. Having to slice the part, do math (even if easy), then enter new values for the 10s of speed fields is time-consuming and could easily be handled by the slicer. Not to mention you would have to redo this work each time you change layer height, line width, or material.

fvrmr, 2021-08-24:
Hi @johnkray thank you for your feature request.
I will bring this up with the team. Keep you posted!

johnkray, 2021-08-24:
@fvrmr Great, thank you!

fvrmr, 2021-08-25:
I have discussed this with the team. But we decided to defer this issue, since this is not a priority for us to add on our backlog. Hope you understand.