Ultimaker/Cura #10321: Dense infill before solid infill layers

simpat1zq, 2021-08-20:
### Is your feature request related to a problem?

If you have a dense infill, like Super slicer, before the solid layers, then you can lower your infill by a lot because the regular infill is no longer needed to support the dense top layers.

### Describe the solution you'd like

So the layer( or maybe x lateral below the top layers should use an anchored dense(50% or so) infill that will support those top layers. If this is down, you can even get away with no infill. The main thing is to make sure that the dense layer is anchored to something. 

Do get a better idea, check out the feature in super slicer and how it handles it. I think that's the only slicer that has it. It's open source so you might be able to adapt their alogorithm.  

### Describe alternatives you've considered

I've looked at gradual infill but it just doesn't save as much filament as that dense layer. 

### Affected users and/or printers

Everyone. Doesn't really matter as to the printer model.

fvrmr, 2021-08-24:
Hi @simpat1zq thank you for your feature request. With gradual infill you can set the gradual infill height. Would that help?

simpat1zq, 2021-08-25:
@fvrmr Unfortunately it won't work.  It might work if the top layer is only at a specific height. But if you had a design, like most designs, that has a top layer at one height, and then another top layer at another height, you can only optimize the infill for one of those heights.

fvrmr, 2021-08-25:
Hi @simpat1zq thank you, I will bring this up for discussion with the team. Keep you posted.