Ultimaker/Cura #10323: excessively long slice time for one part

lownbttr, 2021-08-21:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality Ender 3pro

### Reproduction steps

Slice up the part like usual. one particular file takes forever to slice, then says it will take 23+ days to print instead of about 8 hours. there appear to be strings printing along the Y axis into infinity :)
[image]
[test print-bug.zip](https://github.com/Ultimaker/Cura/files/7025755/test.print-bug.zip)

### Actual results

see above

### Expected results

part should slice with no infinity strings, and print time should be about 6-10 hours.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7025766/cura.log)
[CE3_test print-bug.zip](https://github.com/Ultimaker/Cura/files/7025768/CE3_test.print-bug.zip)

GregValiant, 2021-08-21:
23 days to get to infinity (and beyond!!!) a few times doesn't sound unreasonable.
I opened the STL and sliced it with 4.11 beta and 4.10.  I don't see any issues (Ender 3 Pro).  Maybe someone will notice something in your log file.

[image]

yet-another-average-joe, 2021-08-23:
There was such a bug long ago (like 2 years). It has been fixed AFAIR. 

Similar issue with @lownbttr STL file, using the profile I'm currently printing with... Couldn't slice.

Windows 10, Cura 4.10
project file I saved with @lownbttr file :
[CE3_test print-bug.zip](https://github.com/Ultimaker/Cura/files/7028625/CE3_test.print-bug.zip)

Clicking the slice button, the progress bar resets after 2 seconds, then remains stuck at 0 ; I'm unable to slice this file.

fvrmr, 2021-08-24:
Hi @lownbttr I can't slice your model, like @yet-another-average-joe.
I will bring this up with the team. Keep you posted!

lownbttr, 2021-08-24:
Thanks! I was able to slice it it prusa without any issues. First time I ever had an issue with cura before.

yet-another-average-joe, 2021-08-24:
Similar issue with 4.8 (9 month ago, not 2 years, my bad)

#9073
#9072

fvrmr, 2021-08-25:
I have discussed it with the team and created a ticket on our backlog for this.
Devs see CURA-8717

Ghostkeeper, 2021-09-01:
@fvrmr CURA-8717 doesn't exist at the moment. Which ticket is it?

Ghostkeeper, 2021-09-01:
Seems to be CURA-8517.

lownbttr, 2022-01-11:
> We are not able to reproduce this in 4.12.1 and 4.13 (which is released
> today). @lownbttr <https://github.com/lownbttr> can you confirm that this
> doesn't happen in the newest versions of Cura?

I opened it in V 4.12 and it sliced just fine. V4.10 did not slice. I have
not tried in V4.13 yet, but sounds like you fixed it. thanks!!

Mark

fvrmr, 2022-01-13:
I will close this issue then!

closed 2022-01-13 by fvrmr