Ultimaker/Cura #10326: the move in the z direction does not work properly. z move works only at 400 or above.

andy1at, 2021-08-23:
### Application Version

4.10

### Platform

windows 10

### Printer

creality 10s

### Reproduction steps

type 3 in the z move box

### Actual results

z move box resets to 0

### Expected results

expected the part to move 3 mm in the z direction.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

thank you

nallath, 2021-08-23:
Did you enable "automatically drop parts to the buildplate" in the preferences?

andy1at, 2021-08-23:
> Android [4].
> Links:

yes

andy1at, 2021-08-23:
thank you,
I turned it off and it works.
I also have another question. How can I print a single line the width of 
the nozzle?

Andy

nallath, 2021-08-23:
> I also have another question. How can I print a single line the width of the nozzle?

I don't really know what you mean by that.

closed 2021-08-23 by nallath

andy1at, 2021-08-23:
I have a model of a 1mm diameter rod, I am using a 1 mm tip mounted on 
the extruder. how can I just extrude a single line with out walls, top 
and bottom, fill etc.

nallath, 2021-08-23:
Set wall line count to 1 and top bottom thickness to 0 and infill to 0.

andy1at, 2021-08-23:
the part won't slice, the initial preparation part works fine, but it 
just finishes with no extrusion, like there is nothing to print.?

andy

Ghostkeeper, 2021-09-01:
Cura will produce walls, which are closed loops, by default. It'll print your rod with loops as well. There is a workaround by enabling Wall Overlap Compensation and setting the Minimum Wall Flow to erase the compensated wall, but it won't be precise.

The experimental [Arachne](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta) version of Cura lifts this constraint.