Ultimaker/Cura #10329: Error: "Not signed in. Please allow the application to access the requested data. You can do this by signing in again and clicking Allow."

JDitit, 2021-08-23:
### Application Version

4.10.0

### Platform

W10

### Printer

Ultimaker S5

### Reproduction steps

I am unable to login to Cura 4.10.0 - getting "Not signed in. Please allow the application to access the requested data. You can do this by signing in again and clicking Allow."

Logs attached:
[cura.log](https://github.com/Ultimaker/Cura/files/7033806/cura.log)

### Actual results

Error "Not signed in. Please allow the application to access the requested data. You can do this by signing in again and clicking Allow."

### Expected results

I should be able to login so I can upload sliced files to my printer.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7033833/cura.log)

nallath, 2021-08-24:
We've added a lot more logging to the login flow for 4.11, which is now in beta. Could you try again with the beta and share the logs with us again?

JDitit, 2021-08-24:
>  CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I have installed 4.11 Beta and when I when to login for the 1st time I got the same error.  If I change the URL from https://account.ultimaker.com/app/auth-error to https://account.ultimaker.com<https://account.ultimaker.com/>, It opens up my profile page.  It still does not log me in on Cura.

Joey Donley
Additive Manufacturing Team
Keyword: 3Dprinting<http://maintenance.intranet.dow.com/Repair&Maintain/3D_Printing.asp>
Kelly Services on assignment to The Dow Chemical Company
979-238-4623

General Business

JDitit, 2021-08-24:
I have installed 4.11 Beta and when I when to login for the 1st time I got the same error.  If I change the URL from https://account.ultimaker.com/app/auth-error to https://account.ultimaker.com<https://account.ultimaker.com/>, It opens up my profile page.  It still does not log me in on Cura.

Edit***  This is the pic of the error in Cura:
***@***.***

Joey Donley
Additive Manufacturing Team
Keyword: 3Dprinting<http://maintenance.intranet.dow.com/Repair&Maintain/3D_Printing.asp>
Kelly Services on assignment to The Dow Chemical Company
979-238-4623

General Business

nallath, 2021-08-24:
The error didn't change, but we did update the logging that occurs around it. So could you post the logs that you get from running it with 4.11?

nallath, 2021-08-24:
Also note that we don't get attachements via emails. If you want to add attachments, you need to add them via the interface.

JDitit, 2021-08-24:
Attached

[cura.log](https://github.com/Ultimaker/Cura/files/7039394/cura.log)

nallath, 2021-08-24:
Hmm, what i can see in the logs is that there is something wrong with SSL certificates:

` certificate verify failed: self signed certificate in certificate chain`

Is there something different about your setup? Maybe some self installed SSL certificates?

JDitit, 2021-08-24:
I have not personally installed any, but we did have a new program pushed to our work PC's.  The name of it is "Zscaler".  It's an Internet Security program.  If this is the issue, what do I need to say to corporate IT to allow the connection?

nallath, 2021-08-24:
> I have not personally installed any, but we did have a new program pushed to our work PC's. The name of it is "Zscaler". It's an Internet Security program. If this is the issue, what do I need to say to corporate IT to allow the connection?

Well, i'm not entirely sure what is happening. It's just conjecture from my side. What i think is going on is that they intercept stuff and use a self signed certificate to make that work. Unfortunately, that is caught by Cura and then rejected (which makes sense, since it's not a correctly signed certificate, which indicates a man in the middle attack). 

If the issue started after Zscaler was added, it could very well be that that is the culprit. I don't know what it does exactly, but based on the brief overview I got from wikipedia, it does add up.

nallath, 2021-08-24:
We also have a page that provides more info what should be available for Ultimaker products to work over the network: https://support.ultimaker.com/hc/en-us/articles/360012113719

JDitit, 2021-08-24:
Is there something specific I can say to my network IT crew to get this working again?

nallath, 2021-08-24:
I don't know enough about ZScaler, so i don't know what can / needs to be changed. Maybe they can exclude ultimaker.com as a domain from needing to be checked? I assume that something like that is possible.

Ghostkeeper, 2021-09-02:
So basically the issue is that:
* Cura establishes a secure connection to ultimaker.com to transfer the log-in token.
* This secure connection uses SSL to ensure that nobody can read the token.
* ZScaler adds a "trusted" SSL certificate authority to the computer so that they can decode the traffic and look what's in it.
* Cura doesn't verify the SSL certificate authority with the certificates on the computer, but the default authority (as per Python's implementation).
* Cura sees that someone is snooping on the log-in token and then prevents the token from being stolen.

I think a correct solution from Cura's side would be to adhere to the computer's own certificate authorities, which would include ZScaler then. We're currently still using the `requests` module from Python to transfer this token. Using `QNetworkManager` might provide better integration. To verify if this would be a solution, @JDitit can test if other networking operations still work, such as the Marketplace.
We should be using `QNetworkManager` everywhere, but currently there are two places left where we still use `requests`: The OAuth log-in flow (this issue) and the UM3NetworkConnection plug-in.

Also we should probably improve that error message.

Ghostkeeper, 2021-11-26:
Fixed in Cura 4.13, by this change: https://github.com/Ultimaker/Cura/pull/10901

closed 2021-11-26 by Ghostkeeper