Ultimaker/Cura #10333: Support blockers add extra walls without changing walls

DaLimboster, 2021-08-24:
### Application Version

4.10

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

1. Add a Model
2. Add a support blocker
3. Go to settings for overlaps and remove all (or add any other, it doesnt make a difference)
4. Slice it

[image]

I know if changing infill, there has to be an extra wall between the different infills (one or max. 2 walls).
But there ist no change on infill and also on changing infill there shouldnt be new walls all around.

### Actual results

the print get unwanted extra walls...

### Expected results

don´t get unwanted extra walls...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-08-25:
Hi @DaLimboster this is probably because under `select settings` the settings for walls are enabled. Let me know if that is the case for you. 
For 4.11 beta this is changed and when using an infill mesh only, all the settings are set to 0 by default. 

You can download the beta here: https://github.com/Ultimaker/Cura/releases/tag/4.11-beta

Ghostkeeper, 2021-09-03:
> why not override only the picked settings.

Well, that's just the thing. If we don't override this, the per-object setting will be "2" in your case, causing the infill mesh to have 2 walls inside the infill.

Indeed, we can currently only adjust a part of the wall speed by breaking the wall into multiple loops. This is an architectural problem, listed as feature request under https://github.com/Ultimaker/Cura/issues/7119.

Glad to hear that setting the wall count to 0 fixes the issue. This will be the default in 4.11 so that the user doesn't have to do this.

closed 2021-09-03 by Ghostkeeper