Ultimaker/Cura #10335: Seem like Cura locks files and folders on Windows 10 even after they aren't in use.

raould, 2021-08-24:
### Application Version

4.11.0 BETA

### Platform

Windows 10

### Printer

Monoprice Select Mini V2 (E3D)

### Reproduction steps

1. I open an STL file "modelA.stl" in Cura, do things like slice and save.
2. then Select All and hit the Del key, and "modelA" goes away.
3. I load some other STL file "modelB.stl". I have not done New Project... in-between.
4. I go to File Explorer to delete "modelA.stl".

### Actual results

1. Windows won't let me delete it.
2. if i quit Cura then
3. Windows *will* let me delete it.

### Expected results

If I remove "modelA" from the project, I guess I'd like Cura to release the file lock.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

fvrmr, 2021-08-25:
Hi @raould thank you for your report.
I tried your reproduction steps. But I couldn't reproduce your issue. 
Do you mean with save, save the gcode or save the model?

raould, 2021-08-25:
@fvrmr thanks for taking the time to look at this. I am trying to repro and have narrowed it down more - This is apparently only happening to me with models extracted from .zip's from e.g. Thingiverse.

1. download a model from Thingiverse e.g. benchy https://www.thingiverse.com/thing:763622
2. extract zip file. Don't know if the manner of unzipping matters. What I do is right-click the zip folder in Windows File Explorer, then do Extract All...
3. in Cura, load the now extracted file e.g. "C:\Users\xyzpdq\Downloads\#3DBenchy+-+The+jolly+3D+printing+torture-test+by+CreativeTools.se\files\3DBenchy.stl"
4. after the model loads, select it by clicking on it with the mouse, then delete it by pressing the Del(ete) key on the keyboard.
5. go to File Explorer to where the unzipped parent folder is and try to delete the whole thing i.e. open Downloads in File Explorer, right click on "#3DBenchy+-+The+jolly+3D+printing+torture-test+by+CreativeTools.se" and select Delete from the menu.

expected: can delete
actual: Folder Access Denied
solution: quit Cura, then I don't get the Folder Access Denied error.

Ghostkeeper, 2021-09-03:
My suspicion is that it's the [QFileSystemWatcher](https://doc.qt.io/qt-5/qfilesystemwatcher.html) which we're using to check if the file got modified while Cura loaded it. We do close the file handle after reading the mesh data from it, but since it's the folder being held in use rather than the file it narrows it down quite a lot. QFileSystemWatcher is the only one I can think of that works on a folder-level.

closed 2023-06-09 by github-actions[bot]