Ultimaker/Cura #10347: Crash Cura 4.10 Big Sur

JRRN, 2021-08-28:
### Application Version

4.10

### Platform

MacOs

### Printer

Artillery Genius

### Reproduction steps

Using Cura 4.10 crash randomly.

### Actual results

Restart Cura

### Expected results

Do nothing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Process:               cura [2114]
Path:                  /Applications/Ultimaker Cura4.10.app/Contents/MacOS/cura
Identifier:            cura
Version:               4.10 (4.10.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [2114]
User ID:               501

Date/Time:             2021-08-28 11:08:51.166 +0200
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        90580E39-45D0-86D2-5EDB-1FA72CF60838

Sleep/Wake UUID:       B3BFF842-D255-4D10-9B67-7E6FFB60D95F

Time Awake Since Boot: 5000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff2039a92e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff203c95bd pthread_kill + 263
2   libsystem_c.dylib             	0x00007fff202adbd5 raise + 26
3   libsystem_platform.dylib      	0x00007fff2040ed7d _sigtramp + 29
4   ???                           	000000000000000000 0 + 0
5   libQt5Qml.5.dylib             	0x0000000124747cfc QV4::QObjectWrapper::getProperty(QV4::ExecutionEngine*, QObject*, QQmlPropertyData*) + 2924
6   libQt5Qml.5.dylib             	0x0000000124748794 QV4::QObjectWrapper::getQmlProperty(QV4::ExecutionEngine*, QQmlContextData*, QObject*, QV4::String*, QV4::QObjectWrapper::RevisionMode, bool*, QQmlPropertyData**) + 436
7   libQt5Qml.5.dylib             	0x000000012472cc44 QV4::QQmlContextWrapper::lookupInParentContextHierarchy(QV4::Lookup*, QV4::ExecutionEngine*, QV4::Value*) + 436
8   ???                           	0x00000001409440e9 0 + 5378425065
9   libQt5Qml.5.dylib             	0x0000000124766f93 0x12463f000 + 1212307
10  libQt5Qml.5.dylib             	0x000000012470b8c6 QV4::Function::call(QV4::Value const*, QV4::Value const*, int, QV4::ExecutionContext const*) + 342
[... 1315 lines cut ...]
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Jul  7 2021 02:54:21 version 9.30.444.17.32.5.70 FWID 01-6551fb93
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet Slot 0, Ethernet, en6
Network Service: Wi-Fi, AirPort, en0
PCI Card: ethernet, sppci_ethernet, Thunderbolt@170,0,0
PCI Card: pci1b73,1100, sppci_usbxhci, Thunderbolt@169,0,0
PCI Card: pci1b21,1142, sppci_usbxhci, Thunderbolt@175,0,0
USB Device: USB 3.0 Bus
USB Device: USB 3.0 Bus
USB Device: ASM107x
USB Device: TUSB3410 Boot Device
USB Device: ASM107x
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 85.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 85.0
Thunderbolt Device: THUNDERDOCK2, Comdis, 3, 25.1

fvrmr, 2021-08-30:
Hi @JRRN thank you for your report. 
Could you share also your log file? You can find your log file here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

JRRN, 2021-08-30:
@fvrmr 

[cura.log](https://github.com/Ultimaker/Cura/files/7078210/cura.log)

fvrmr, 2021-08-31:
Thank you for your log file. I can't find in your logs why it is crashing. Right now there is nothing we can do to solve this and we need to wait till we got other/better reproduction steps to reproduce this issue.
If you or somebody else runs into it again with reproduction steps please let us know!

closed 2024-12-09 by github-actions[bot]