Ultimaker/Cura #10349: Mesh modify overlap settings are being ignored

BloodyRain2k, 2021-08-28:
### Application Version

4.10.0

### Platform

Win 7 x64

### Printer

CR-6

### Reproduction steps

Set up a model with "Alternate Extra Wall" enabled
Add a 2nd model to use as overlap and turn "Alternate Extra Wall" off for it

### Actual results

"Alternate Extra Wall" is still treated for the overlapping area as if it's still turned on.

[image]

And in case the overlap mode "Infill mesh only" is to be taken literally, this is the result when I use the alternative "Cutting Mesh":
[image]

Why isn't there a 'normal' overlap mode that doesn't alter the result already by just existing?

### Expected results

Actually overriding the setting and changing it.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Die_Case_Adapter.zip](https://github.com/Ultimaker/Cura/files/7071451/Die_Case_Adapter.zip)

fvrmr, 2021-08-30:
Hi @BloodyRain2k thank you for your report. I will bring this up with the team, keep you posted!

BloodyRain2k, 2021-09-01:
I found another case where the override settings aren't exactly behaving as intended:
[image]

I was messing around in a similar way, this time trying to print the bottom of something hollow while the top is filled as much as possible.
How ever, as the setting "Infill Density" defaults to 0% from the primary settings, which is correct, the overrides break when the density is being left at 0% and instead an "Infill Line Distance" is entered as shown on the screen.
This will turn the infill to the proper density, but won't turn on any handling for the needed calculations of the resulting infill.

On top of this is it *then* neccessary to remove the "Infill Line Distance" setting, to reset it and unlock the normal density field again, because they don't have reset buttons.

[Pellet_6mm.zip](https://github.com/Ultimaker/Cura/files/7092944/Pellet_6mm.zip)

Ghostkeeper, 2021-09-02:
If set to Infill Only, an infill mesh will only affect the area of infill; the area that would get filled by orange lines. The infill volume gets made smaller by sometimes adding an extra wall. This alters the area over which the infill mesh gets influence. The infill mesh doesn't adjust the settings of the original walls of the objects it's overlapping with; only the infill.

An infill mesh itself can add walls inside the infill, and those walls get the alternating extra wall too. If Alternate Extra Wall is disabled for the infill mesh, then the infill mesh doesn't get the alternating extra wall, but the outer object still does. This is your misconception.

Here is an annotated screenshot that shows the behaviour more clearly:
[image]

So this seems to be behaving as expected here.

> While it DOES alter the selected setting as intended, it on the other hand messes with the walls, in whatever way...

I'm not entirely sure what you mean here. Cutting meshes will cut up the walls on the borders of the mesh, but your cutting mesh of course overlaps with the entire layer so there should not be any downside of it then.
There is not currently a way to change the settings of the walls without breaking those walls into separate loops. This issue is known with us, but architecturally quite difficult to fix. Especially if those changes involve things like changing the wall line width, such that the walls don't match up any more.

> How ever, as the setting "Infill Density" defaults to 0% from the primary settings, which is correct, the overrides break when the density is being left at 0% and instead an "Infill Line Distance" is entered as shown on the screen.
This will turn the infill to the proper density, but won't turn on any handling for the needed calculations of the resulting infill.

Yes, I can reproduce this and confirm that it is a bug. The Infill Pattern setting was disabled, but if the Infill Line Distance is greater than 0 you should just be able to edit it.

BloodyRain2k, 2021-09-02:
> > While it DOES alter the selected setting as intended, it on the other hand messes with the walls, in whatever way...
> 
> I'm not entirely sure what you mean here.
Look at the gap that appeared between the outer and inner wall.
I don't know why, and this also didn't happen before as far as I can remember, but it altered the line width a little, creating a gap between those two wall types.

I should have marked it, but it was so glaring obvious to me at the time, that I didn't...

rburema, 2021-09-07:
I just made a commit that at least prevents the infill pattern from being disabled when only the infill line distance is > 0 in mesh overlaps. (Whatever we decide do with all other issues and/or non-issues, they're bigger than a 5 minute fix, so that will be resolved some other way.)

See https://github.com/Ultimaker/Cura/commit/7eeae4db87f3f6e860c2bc4faab79af50c59e484

Ghostkeeper, 2021-09-13:
> Look at the gap that appeared between the outer and inner wall.
I don't know why, and this also didn't happen before as far as I can remember, but it altered the line width a little, creating a gap between those two wall types.
> 
> I should have marked it, but it was so glaring obvious to me at the time, that I didn't...

I'm unable to reproduce that gap using your project file. In Cura 4.10 as well as current Master, there is no gap. Maybe this screenshot was taken with different settings?

That is the only remaining issue in this thread. The real issue was not a bug but a misunderstanding of what Infill Mesh does, and the secondary issue is now fixed.

BloodyRain2k, 2021-09-13:
I've created a seperate issue (#10424) for the line width, as I was able to instantly reproduce it on a different part in 4.11 and 4.9.1, but it seems to again be a feature I'm not capable of understanding correctly...

But yeah, this issue here had a different focus and can be closed now.

closed 2021-09-13 by BloodyRain2k