Ultimaker/Cura #10350: Loss of settings on switch of material or profile (Differences dialog didn't show acceleration rate changes/loss of settings)

tyeth, 2021-08-28:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Reprappro Mendel rev2

### Reproduction steps

Set my acceleration maximums and flow percentages, saved them into a new profile and updated the profile a couple of times while a print was going on. Changed material (PLA 200c to PLA 205c 6.5mm retraction) and it asked me to confirm my changes from default profile settings, I did. I did the print again then later goto adjust another setting after cancelling the second print part way through (over extraction and testing new roll of PLA+) and noticing the infill changed to gyroid from grid. Looking further my acceleration settings are lost as are my flow percentages. 

### Actual results

Settings for acceleration are reset to 3000mm/s/s and flow percentages to 100% for all values (infill, top bottom, primary, etc)

### Expected results

Profile Dialog warns you if acceleration or flow percentage or any values will change when switching profile or material or any other thing affecting such properties

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Set my acceleration maximums and flow percentages, saved them into a new profile and updated the profile a couple of times while a print was going on. Changed material (PLA 200c to PLA 205c 6.5mm retraction) and it asked me to confirm my changes from default profile settings, I did. I did the print again then later goto adjust another setting after cancelling the second print part way through (over extraction and testing new roll of PLA+) and noticing the infill changed to gyroid from grid. Looking further my acceleration settings are lost as are my flow percentages. Are they both saved as part of the profile? Does the Change of profile dialog (difference checker) detect differences to advanced/nested settings, and if so please investigate whether the reset occurs depending on changing material to trigger the profile differences checker dialog.

Thanks, I'll go and recheck on beta 4.11 which is great by the way, no crashes so far unlike 4.10, where it seems to be after the first slice the chance of crashing or at least very significant freezing on subsequent slices is high. Not sure if I resolved that with firewall enabling like someone suggested or if I just moved onto 4.11 for the last couple of uses, I'll try to be sure which I'm using.

tyeth, 2021-08-28:
Can't see what happened, retested in 4.10 and the dialog works as expected. Even installed Profile Analyser and seems like I must have just been asleep at the wheel.

tyeth, 2021-08-29:
I've finally worked out where "the problem" is, apart from being in my head...

The settings dialog only mentioned "Acceleration control" being enabled, and not the accerleration rates that also deviate from default values (so called nested settings in my head).

Screenshot **before** settings change showing rates that I've set:
[image]
 
Screenshot **after** changing to a default profile showing the acceleration rates and _only warning that acceleration control enabled is changed_ from the default profile settings.
[image]

I would expect to be warned that my rates are also changed. this should be the case for any nested changed settings like flow rate percentages etc.

fvrmr, 2021-08-30:
Hi @tyeth thank you for your detailed report. 
But if you change the value for example the infill acceleration and want to change the profile it show that you've changed that value. 
[image]

But I understand that it is a bit confusing. I will bring this up with the team, keep you posted!

Ghostkeeper, 2021-09-08:
I unchecked the "project file" and "log file" since you didn't provide any. A project file of the before/after state would help to debug this.

So that pop-up only warns you about settings that were not saved to a profile. It pops up when you change profiles, and asks you whether you want to carry those unsaved changes to the new profiles or want to discard them to start fresh from the profiles.

Since your acceleration and flow changes are saved in your previous profile, they don't show up in that dialogue. They are saved in your profile, and will still be there when you switch back to your custom profile.

So really what's happening is:
* You save acceleration/flow setting changes in a custom profile.
* You switch profiles from your "Fine skirt-015 1000mm/s/s" profile to one of the default profiles, "Extra Fine".
* The acceleration/flow setting changes are not in that default profile you switched to.

That seems quite logical to me.

> Thanks for the update, it might be worth offering a percentage and mm/s/s option so I can set a maximum safe value for my printer and then override as percentages or direct acceleration rates.

There are machine settings for Maximum Acceleration in the X, Y, Z and E directions. They can normally only be set in a printer definition profile (which is not editable from the GUI) but the Printer Settings plug-in from the marketplace allows setting them. Similarly, it is not possible to have formulas in settings from the GUI, only when editing the profile files with text editors, so setting it as a % of another value is not possible until https://github.com/Ultimaker/Cura/issues/3856 is implemented.

fvrmr, 2021-09-09:
We have discussed this issue with the team and decided to defer this, thanks to @Ghostkeeper for explaining how the settings work. I hope this makes sense to you.

tyeth, 2021-09-09:
Thanks for your replies all. Makes sense and I look forward to the
percentage support in the future.

closed 2021-09-10 by tyeth