Ultimaker/Cura #10352: Favorite settings

zaquarius, 2021-08-29:
### Is your feature request related to a problem?

Too many settings, and scrolling through settings/searchings lags like crazy

### Describe the solution you'd like

There are a few settings that I use frequently (such as layer thickness, infill, support, etc) and would save a lot of time and scrolling to have them all in one section (right click > add to favorites) above all the others. 

### Describe alternatives you've considered

Hiding unused settings? Doesn't really address the problem

### Affected users and/or printers

Everyone

fvrmr, 2021-08-30:
Hi @zaquarius thank you for your feature request. You can set the setting visibility to basic, advanced or all. Or you can set an custom selection. Let me know if that works for you. 
[image]

Ghostkeeper, 2021-09-08:
That's not a real solution to what Zaquarius is asking, though.

He considered the possibility of hiding less frequently used settings (in the "alternatives considered"), but indeed it's not really a solution because you do want to change those too, once in a while. So you have to make them visible, at least temporarily, which is cumbersome and defeats the purpose of having all the frequently-used settings listed together.

The counter-argument that I do have to this is that together with the setting visibility profiles, settings showing/hiding depending on other settings, and the complexity of the profile system in general, adding this feature could make it even more complex to understand for the user where their settings are.

But I'll ask what the team thinks about having a "Favourites" header similar to what the material management page has.

lorenzoromagnoli, 2021-09-14:
Hey @zaquarius could you elaborate a bit more on why "hiding unused settings won't really address the problem?

On a side note, as we are considering restructuring a bit the recommended print settings dialogue  it would help to get some insights in which are the settings that you would set as your favourites...

you mentioned:
- layer height 
- infill (I guess you refer to infill %, also present in the recommended settings view )
- support (is this just enable/disable or also support type?)

zaquarius, 2021-09-24:
There are many settings that is use somewhat often, but not frequently. I have about 10 settings visible in most tabs in cura, and I use all of them from tone to time (and cura is so slow, it would be very inconvenient to hide them all and have to search every time I need one. The is only a small handful that I use constantly (layer height, infill %, generate support, support overhang angle, top/bottom layers, build plate adhesion, off the top of my head). It would save a lot of time if I could just have a tab at the top with those, but still have all the others visible. I hope this clarifies

fieldOfView, 2021-09-24:
I think this can be implemented as a plugin, so it would not add to the default complexity of what is currently there. I'm a "bit" swamped for time, so I won't make any promises though.

fieldOfView, 2021-09-24:
I've put it on my plugins todo list here: https://github.com/fieldOfView/Cura-Plugins/issues/10