Ultimaker/Cura #10353: Print pausing a random layers for 2Hr+ jobs

abtrooper82, 2021-08-29:
### Application Version

4.10

### Platform

Windows 10 x 64

### Printer

Artillery Genius and Ender 3

### Reproduction steps

1. Attempted to re-print an STL with a pint time of approx 2Hrs. which I had printed with no issues in V4.9

### Actual results

2. The printer stopped at around 82% and Cura crashed
3. I had to restart my PC as the USB port being used was locked down by Cura
4. Attempted to re-print but stopped at 30%
5. Had to re-boot Cura again

### Expected results

The job to print as required according to the file specification as per version 4.9 of the software

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

fvrmr, 2021-08-30:
Hi @abtrooper82 thank you for your report.
Does this always happen when printing via USB?

fvrmr, 2021-08-31:
Can you share your log file from 4.10? You can find your log here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Ghostkeeper, 2021-09-08:
The log file indicates that Cura still tries to continue sending g-code to the printer, but doesn't get a response. In this case, a mere 25 minutes after the print started it shows this message repeatedly:

> 2021-08-28 15:48:00,336 - WARNING - [USBPrinterUpdate] USBPrinting.USBPrinterOutputDevice._sendCommand [251]: Timeout when sending command to printer via USB.

So from Cura's perspective, it looks like the printer firmware is frozen or the connection severed/unstable.

I would suggest trying a different USB cable. There isn't really anything Cura can do about this. The printer doesn't read any telepathy protocol, so it needs to have a stable USB connection. This is the most important weakness of USB printing, in my opinion.

closed 2021-09-08 by Ghostkeeper

abtrooper82, 2021-09-08:
So while I can understand the comments about printer firmware / usb cable etc. What I don't get is why it only happened on this version. Moving back to the previous version printing the same file worked fine.  I will try 4.11 when its released as a stable version

Thanks anyway

Ghostkeeper, 2021-09-09:
Well, I don't know what sort of sample size you've created, and if you interweaved those tests. We regularly get bug reports from people who had their hardware break around the same time as when they upgraded Cura, and blame Cura on the failing function.

Nothing changed between 4.10 and 4.11, so I wouldn't think that changes anything. The change to the USB printing functionality was [this PR](https://github.com/Ultimaker/Cura/pull/8463) which did get merged in 4.10, but only runs before the print to detect whether there is a printer there.
The last change to the code that runs while printing via USB was [this one](https://github.com/Ultimaker/Cura/commit/1946615fff768bff4795bba5974f07315597c389) in 4.7.

abtrooper82, 2021-09-09:
Ok, well thanks anyway. BTW I'm a big fan of Cura and not like many, blaming the software in the first instance. Just curious when things like this happen.