Ultimaker/Cura #10354: Dual Extrusion - Build Volume / Disallowed Area is Incorrectly Calculated

RFBomb, 2021-08-29:
### Application Version

4.10

### Platform

Windows 10

### Printer

Custom Ender 3 with 2 extruders

### Reproduction steps

Set up a custom printer with 2 nozzles
 - Nozzle 1 X = 0, Y = 0
- Nozzle 2 X = -24, Y = 0

### Actual results

The Disallowed Areas prevent putting an object less than 24 mm into the build volume, even though both nozzles can reach that location.

### Expected results

Nozzle 2 should be able to reach ( 0 , 0 ). 
I would expect Nozzle 2 max X = (Stroke Limit - 24)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Attached is a screenshot of my hot end setup, the Nozzle Offset page for Extruder 2. (Extruder 3 was only mocked in as an additional extruder to test various values). Likewise, the Y Nozzle Offset was greatly exaggerated to showcase the Disallowed Area. 

[image]

As you can see, even though the The left-hand nozzle, which I have set up as Extruder # 2 (T1) is -24mm away from T0, the 'reachable area' is getting defined much further in on the left-hand side. Based off this calculation, CURA seems to believe that the Left nozzle (T1) can reach the right-most side of the bed (which it can not), but it is unable to reach the left-most side of the bed. 

Likewise, I have set the Y offset for E2 to -50. As you can see, CURA thinks that now between 0 and 50 are unusable, but 50-200 are usable. The reverse should be true. If the bed Y-Max is 200, and Y nozzle 2 is -50 offset compared to nozzle 1, then when Nozzle 1 is at 200, nozzle 2 can only reach maximum 150. 

The GCODE calculation itself is perfectly correct. To reach (0,0) with nozzle 2, the print head will move to (24,0). BUT I cannot slice if anything to be printed by nozzle 2 is less than 0 due to this 'Disallowed Area' calculation. 

Attempting to work around the issue and modify the nozzle offsets in Cura so that Nozzle 1 is offset compared to Nozzle 2 (which is now set to (0,0), you get an identical image. 

Now, changing the offset from (-24) to (24) for Nozzle 2 does two things:

- The 'Disallowed Area' is now 'correct'.
- The print will be incorrect because the GCODE will move to X=0 instead of X=24 to print with Extruder 2 at X=0. (in reality, this puts E2 at -24, off the print bed altogether.) Meaning this setting isn't usable (well, it can be if you turn 'Apply extruder offsets to gcode' setting to OFF', but this is ON by default). 

### Code Base 

I found this code inside of **CuraEngine\gcodeExport.cpp**

```
Point GCodeExport::getGcodePos(const coord_t x, const coord_t y, const int extruder_train) const
{
    if (use_extruder_offset_to_offset_coords)
    {
        const Settings& extruder_settings = Application::getInstance().current_slice->scene.extruders[extruder_train].settings;
        return Point(x - extruder_settings.get<coord_t>("machine_nozzle_offset_x"), y - extruder_settings.get<coord_t>("machine_nozzle_offset_y"));
    }
    else
    {
        return Point(x, y);
    }
}
```

and this code in  **BuildVolume.py -- _computeDisallowedAreasStatic function** 

```
{get details for some extruder }
offset_y = -offset_y  # Y direction of g-code is the inverse of Y direction of Cura's scene space.

... { loop through and compare all active extruders to determine unreachable borders } ...

other_offset_y = -other_offset_y
left_unreachable_border = min(left_unreachable_border, other_offset_x - offset_x)
right_unreachable_border = max(right_unreachable_border, other_offset_x - offset_x)
top_unreachable_border = min(top_unreachable_border, other_offset_y - offset_y)
bottom_unreachable_border = max(bottom_unreachable_border, other_offset_y - offset_y)
```

### SUMMARY

- The above code for generating gcode will move the head POSITIVE if the offset is NEGATIVE. This is correct functionality, and this occurs for both X and Y axes. (you wind up subtracting a negative in my examples, meaning you add.).
- The code for determining the "Unreachable_Border" is likely incorrect, and is the reason that the display engine doesn't match up to expectation.
- Here are my proposed changes:

Replace 
```
offset_y = - offset_y  # Y direction of g-code is the inverse of Y direction of Cura's scene space.
```
with 
```
offset_x = - offset_x  # X direction of g-code is the inverse of X direction of Cura's scene space.
```

and 
```
other_offset_y = -other_offset_y
left_unreachable_border = min(left_unreachable_border, other_offset_x - offset_x)
right_unreachable_border = max(right_unreachable_border, other_offset_x - offset_x)
```
with
```
other_offset_x = -other_offset_x
left_unreachable_border = max(left_unreachable_border, other_offset_x - offset_x)
right_unreachable_border = min(right_unreachable_border,  other_offset_x - offset_x)
```

Changing the Min/Max statements for Left/Right unreachable border brings them in line with the Top/Bottom unreachable border. Top = min, while Bottom = Max. If bottom = 0 and left = 0, Left should also use Max().

Then, removing the offset_y inversion, and replacing it with an offset_x inversion should resolve the issue where the disallowed areas seem inverted for both Y and X. 

Normally, I would test this by compiling and verifying my proposed code. But I followed everything in the wiki regarding setup for compiling on Windows, and it I cannot for the life of me get it to compile. VS2019 keeps complaining about nearly half of the python imports being unavailable, CMAKE issues, etc, etc. 
That, and I wanted to propose the changes with a lengthy argument for them to discuss, instead of just submitting a pull request for it.

RFBomb, 2021-08-29:
Alternatively, the issue may lie in the logic statements that build the polygons to display. 
Flipping the logic may be the solution. 

Go from:
```
if border_size - left_unreachable_border > 0:
...
if border_size + right_unreachable_border > 0:
...
if border_size + bottom_unreachable_border > 0:
...
if border_size - top_unreachable_border > 0:
```

To:
```
if border_size + left_unreachable_border > 0:
...
if border_size - right_unreachable_border > 0:
...
if border_size - bottom_unreachable_border > 0:
...
if border_size + top_unreachable_border > 0:
```

By flipping that logic (using my example), the match becomes:
```
//left_unreachable_border 
if 5 + ( -24) > 0:
...
//right_unreachable_border 
if (5 - (24)) = 29 > 0:
```
This means that the logic statements the control which polygons are drawn will correct it, instead of changing the match that calculates the *_unreachable_border variables. This said, I'm unsure how these points work, so the numpy.array[] may have to be adjusted here to accomodate the updated IF statement.

RFBomb, 2021-08-31:
I didn't even think about changing the 'Build Plate Adhesion Extruder' setting. 
My test printer shows the same functionality as your reply demonstrates.

Based on this function (I haven't looked at code to verify), It  _seems_ that the initially displayed grid is where it thinks the 'build plate adhesion' extruder can reach.

Once a part is placed onto the bed, it overlaps the 'bed adhesion' extruder with the parts selected extruders.

[image]

The above photo shows the overlap I refer to. I selected different materials to showcase the parts are set up for different extruders. Based on these numbers, this is a severely limited build volume, because it will only allow slicing where the two can overlap. 

----------------------------------------

[image]

Moving one of the two parts outside this center zone prevents slicing, even if that part's extruder can reach the all points. I toggled the 'Build Adhesion' extruder, which had no affect. This I think is fine, see note below the next picture.

[image]

Changing both parts to the same extruder though allowed it to print both parts, as long as the 'Build Plate Adhesion' extruder was set up as that same extruder. If you select the opposite extruder (parts set up for E2, build plate is E1), then the part outside of E1's zone will not be able to slice.  I think this functionality is perfectly fine, because of how the build plate adhesion works. 

----------------------------------------

Selection 'Build Plate Adhesion = NONE' is where a problem may lie though. 

[image]

As seen here, it will not slice due to one of the parts being outside of the shared zone, even though build plate adhesion is set to NONE. 

[image]

Changing the further part from E1 over to E2, so that both parts are on same extruder, the project can slice, even though neither part has shifted. 

##Based on these new findings, here are my thoughts:
- The zone where E2 was allowed to print was calculated properly, as evidenced by the grid when selecting E2 as the build plate adhesion extruder. 
- The zone where E1 was allowed to print calculated as if it had an offset, despite no offset being set up. (It actually calculated with an offset inverse of E2). 

##Therefore:
- This code block I reference in my original post is mostly correct, at least the idea of it is. Pasted below for reference. (I have omitted the If statements that follow this, since I now think they are functioning properly. I also rescind my thought about inverting X/Y offsets.)
```
other_offset_y = -other_offset_y
left_unreachable_border = min(left_unreachable_border, other_offset_x - offset_x)
right_unreachable_border = max(right_unreachable_border, other_offset_x - offset_x)
top_unreachable_border = min(top_unreachable_border, other_offset_y - offset_y)
bottom_unreachable_border = max(bottom_unreachable_border, other_offset_y - offset_y)
```
- The inversion seems to come from looping through and comparing the extruder in question with all other extruders to find a shared area where all can reach, not really an 'unreachable' area. Since this conflates 'Shared' with 'Unreachable', I think this should be implemented in a different method, if still needed. (I'm actually unsure how the shared area is compared to part placement, I'm assuming that the part is simply compared against all the 'unreachable (currently actually shared)' meshes for each extruder. )

## Possible Solve:
- There is no need to loop through all other extruders to determine their shared area, atleast not in this function. Elimnate the loop, and modify the Min\Max statements as follows. 
```
left_unreachable_border = min(left_unreachable_border, - offset_x)
right_unreachable_border = max(right_unreachable_border, - offset_x)
top_unreachable_border = min(top_unreachable_border, - offset_y)
bottom_unreachable_border = max(bottom_unreachable_border, - offset_y)
```
- This will cause the mesh drawn for each extruder to match up with the actual unreachable area. (All I did was remove the `Other_Offset_XY` values, since for E1 those would typically be zero anyway, resulting in a simple inversion of the current extruder's offset. )

RFBomb, 2021-09-05:
As I said in my post, I had some issues compiling the Cura build itself, so I'm unable to test this in cura. 

That said, I ported the code over into C#, made some tweaks and build a winforms app about a few hours to showcase the issue and the proposed change.

If you have VIsual Studio, you should be able to clone this repo, compile it, and test yourself. You will have to input your own offsets, but this tool showcases the old and new methods side by side. I added several ways to view the areas too. The rectangles generated by this testing tool's 'Old method" work identical to how cura is functioning currently.

https://github.com/RFBomb/CuraNozzleFix

Now, the only change I've done to the actual math to generate the polygon is the change I note in the 'Possible Solve' section above. The new method eliminates comparison to the other nozzles when calculating this nozzle's disallowed areas.

---

[image]

this first photo shows the 'accessible area' overlap, in the same style view that cura display it. 

---
[image]

Here is the inaccessible (retricted) areas. 

---

[image]

As you can see, there is a large area where printing is not allowed to go to, despite being accessible by both nozzles. 
In the 'Inaccessible' photo, you can also see that the two restricted areas mirror each other, even though only 1 of the two nozzles has an offset built into it. 

---

Not only does the new method provide a better / more accurate build volume, it also shows the real restricted area if one of the extruder's cannot reach some location due to the offset. This means that it will properly prevent slicing in an inaccessible area that was previously undetected. 

[image]

Ghostkeeper, 2021-09-08:
> Based on this function (I haven't looked at code to verify), It _seems_ that the initially displayed grid is where it thinks the 'build plate adhesion' extruder can reach.

The disallowed areas - at any moment in time - show where you can't print using the extruders currently in use. With an empty build plate, the only "used extruder" is the one used for the build plate adhesion. Setting the Adhesion Type to "None" results in no extruder being used, so you get a completely free build plate.

> The above photo shows the overlap I refer to. I selected different materials to showcase the parts are set up for different extruders. Based on these numbers, this is a severely limited build volume, because it will only allow slicing where the two can overlap.

This is correct, since Cura will also outline your brim/skirt/raft with a skirt line for every used extruder, which you will see if you inspect layer view in this case. This skirt line primes the extruder. Similarly, the active nozzle will also avoid previously printed objects (even if those are printed with the other extruder), and so must be able to travel around the other object with clearance. So it can only print your magenta Benchy if both extruders are able to reach the area covered by your brim around that Benchy, regardless of what extruder the Benchy itself is being printed with. These requirements collide with the suggestions you are making.

> The zone where E1 was allowed to print calculated as if it had an offset, despite no offset being set up. (It actually calculated with an offset inverse of E2).

If `nozzle_offsetting_for_disallowed_areas` is set to `True` (the default), then Cura will assume that the build volume is the **bounding box around the union of where both nozzles can reach**. If E2 has an offset of 24,24 and E1, there will be a zone in the right-back corner where E2 can reach but E1 can't, but at the same time there will be a zone in the front-left corner where E1 can reach but E2 can't. After all, if the print head is moved all the way into the front-left corner as far as it can go, the E1 nozzle will be further into the front-left corner than the E2 nozzle. This is symmetric even if only one of the nozzles has an offset from 0,0. If the offsets were -24,-24 and 0,0, the disallowed areas would be the same; only the outputted g-code coordinates would be different.

You found my explanation for this already in a commit back in 2016 too, quoting from your PR:

> What this assumes (I believe incorrectly) is that the left nozzle is thus also unable to reach the last 18mm of the build volume on the right-hand side. While certainly this is a possible scenario, I don't think this would be the standard.

Indeed, that is what it assumes. It assumes that the range of locations is limited by the motion range of the print head. If so, the left nozzle will be able to reach farther left than the right nozzle, and the right nozzle is able to reach farther right than the left nozzle. Even if only one of them has an "offset" defined in the settings from the position of the print head.

This assumption is common for many printers, but doesn't hold for all printers, of course. There are also printers where the *build plate* is the limitation of where things can be printed, rather than the movement range of the print head. Those printers can set `nozzle_offseting_for_disallowed_areas` to `False`. It will then no longer move the disallowed areas around the border with this offset. Note that this is separate from the `machine_use_extruder_offset_to_offset_coords` setting (which only affects how the g-code is output). If is assumption causes the front-left corner of the build volume to become a different coordinate than 0,0 according to your firmware, you can fix that with a `G92` command in the start g-code by moving the nozzle to a known valid position and then telling the firmware what position that is according to Cura through the `G92`.

For printers where the print head's range of motion is partially the limiting factor, but the build plate is limiting in other cases, custom disallowed areas can also be defined, per-extruder or for the print head. This way Cura can catch pretty much all cases for all FFF printers in the field.

RFBomb, 2021-09-08:
OK, so the purpose of the code in question was to assume symmetry in the print head. I I discovered it because my printer's E1 can reach entire plate, and only E2 has the right-hand side restriction. 

> This is correct, since Cura will also outline your brim/skirt/raft with a skirt line for every used extruder, which you will see if you inspect layer view in this case. This skirt line primes the [...]

The edits I made shouldn't have affected anything brim or border related though. The `border_size` calculation was set to 0 for my test app, sure. But that was more to showcase the 'reachable area' calculation more than anything. The border calc will still adjust the limiting area. My concern was basically that E1 (0,0) has an unnecessary bounding bounding box on the left hand side preventing it from reaching less than X=24 if E2 (-24,0) is enabled at all.

Out of curiosity, I loaded in an Ultimaker S5, which has the `nozzle_offsetting_for_disallowed_areas` set to false, and tested various offsets. What I found was that because the flag was set to false, no limiting box for the offset was ever created, even when I had set the X offset to a ridiculous value of +/-200mm.  Adjusting the brim size still affects the borders, but the offsets simply aren't taken into account. (I'm assuming mechanically the X can have both extruders reach the entire plate when not user-modified, which is why `nozzle_offsetting_for_disallowed_areas` can be set to false). 

My edits assumed that the 'unreachable symmetry assumption' was incorrect, and too restrictive due to the mirroring effect the code produces. I still think this is the case, but atleast I understand the reasoning behind it now. Safer to assume symmetry than not. I could see this applying to an IDEX style printer for example (though I'm a bit unclear how cura would handle such a printer, with the zero's being opposite sides and all.)

So the solution in my scenario would be to add in a restricted area for nozzle 2 and set`nozzle_offsetting_for_disallowed_areas` to false in the printer's file. I'm unaware of how to set up a restricted zone for a single extruder, which is why I targeted this code in the first place.

Ghostkeeper, 2021-09-08:
> The edits I made shouldn't have affected anything brim or border related though. The `border_size` calculation was set to 0 for my test app, sure. But that was more to showcase the 'reachable area' [...]

My explanation would've been better with a screenshot:
[image]
What I wanted to explain is that, even though this object is printed in magenta and the adhesion is also in magenta, there is still a yellow line around it (printed with the other extruder). This yellow line must be printed in a place where the yellow extruder can reach, and so the object can only be placed in a location where both nozzles can reach. So the brim calculations are irrelevant, but we shouldn't allow placing magenta objects in a region where only the magenta nozzle can reach. This is what went wrong with your example of the two Benchies, where the magenta Benchy could get printed only if Magenta was the only used colour.

This indeed means that the right nozzle (E1) can't reach all the way to 0,0. The left nozzle (E2) can, though.

> What I found was that because the flag was set to false, no limiting box for the offset was ever created, even when I had set the X offset to a ridiculous value of +/-200mm.

Yes, so if that flag is `False`, then it assumes that the build plate is the limiting factor. Ergo, the print head is free to move as far as needed to allow both nozzles to reach the entire build plate. Even 200mm.

> I just think its illogical to go with the assumption that the nozzle at 0,0 cannot reach 0,0 on the build plate, which is what the current code produces and my suggestion fixes.

Assuming the motion of the print head is the limiting factor, if the nozzle at 0,0 can reach the 0,0 coordinate on the build plate, that would mean that the nozzle at -24,0 can reach -24,0 on the build plate. So you would be unnecessarily limiting the range of motion of the printer. I don't think Cura's default assumption is meant to be on the safe side. It's meant to be on the common side. In the field, from what I've seen, the most common dual extrusion printers are shared-nozzle printers for which all of this is no issue at all (both nozzles at 0,0), and the second-most common type are printers whose motion is limited by the range of the gantry and endstops, so where the print head can move, rather than by the size of the build plate.

> OK, so the purpose of the code in question was to assume symmetry in the print head. I I discovered it because my printer's E1 can reach entire plate, and only E2 has the right-hand side restriction.

So in your case, the range of motion for E1 is limited by the build plate, while the range of motion for E2 is limited by the build plate on one side and the range of the print head on the other side? I would say that is quite an uncommon situation. You would have to implement that through custom disallowed areas in the printer definition file.

RFBomb, 2021-09-09:
I don't think my setup is all that unusual. Its basically just two extruders sitting side-by-side. It just happens that E1 is at 0,0 and E2 is at -24,0.
[image]

> ...This indeed means that the right nozzle (E1) can't reach all the way to 0,0. The left nozzle (E2) can, though.

No, I understood what you were saying regarding both nozzles reaching the location regarding bed adhesion. Here is my thought process:

Description within `_computeDisallowedAreasStatic`
>:param border_size: The size with which to offset the disallowed areas due to skirt, brim, travel avoid distance
         , etc.

Upper level method that actually tells the calculation to be performed:
>  def _updateDisallowedAreas(self) -> None:
...
disallowed_border_size = self.getEdgeDisallowedSize()
result_areas = self._computeDisallowedAreasStatic(disallowed_border_size, used_extruders)  # Normal machine disallowed areas can always be added.
...

Getting the border size:
>def getEdgeDisallowedSize(self):
...
self._edge_disallowed_size = max(move_from_wall_radius, support_expansion + farthest_shield_distance, support_expansion + bed_adhesion_size)
return self._edge_disallowed_size

and finally the snippet where the unreachable polygon is created within `_computeDisallowedAreasStatic`:
> if border_size - left_unreachable_border > 0:
                    result[extruder_id].append(Polygon(numpy.array([
                        [-half_machine_width, -half_machine_depth],
                        [-half_machine_width, half_machine_depth],
                        [-half_machine_width + border_size - left_unreachable_border, half_machine_depth - border_size - bottom_unreachable_border],
                        [-half_machine_width + border_size - left_unreachable_border, -half_machine_depth + border_size - top_unreachable_border]
                    ], numpy.float32)))

What I'm getting at is: Based on the result of  `getEdgeDisallowedSize(self)` that is entered into the `_computeDisallowedAreasStatic` call, the brim, support, etc have already been accounted for within each nozzle's unreachable areas. The argument that the unreachable area for E1 needs to be modified due to E2's offset due to bed adhesion  doesn't make sense since the border_size is already accounted for when generating the point tuples.

___

On the other side of this is that if ONLY E2 is enabled, the old method thinks that it can reach the entire build area, while this is not the case. This is because there are no other extruders in the used_extruder array to compare against, which allows slicing into an area that is unreachable (but this is ignored by the current calc).  Which is another reason I was pushing for this to be reviewed.

RFBomb, 2021-09-09:
By the way, I've updated my proposed solution with on that pull request's branch, even though the PR has been closed already. 
Basically, I reverted the change to current method then:

- Initialized 2 new variables to represent the other_offset_max for X and Y (used in the modified loop mentioned next)
- Modified the loop to return the maximum offset value found for extruders that aren't the currently selected extruder
- Move the min/max statement outside of the loop, and use the other_offset_max values in the calculation like the current method does. 

This setup will I think still do what you expect it to, while fixing what I think is the larger issue of slicing in an unreachable area while only 1 nozzle is selected. It should:

- Allow the creation of the unreachable area even when only 1 nozzle is selected.
- I think this will also prevent the mirroring effect from occurring. (Nozzle with X negative offset will not prevent nozzle with 0,0 as the offset from reaching zero.)
- It will still impose stricter limits though in cases where its required. (For example, if you have a nozzle with positive offset and a nozzle with offset of 0,0 the nozzle with 0,0 offset will be restricted by the one with positive offset.)

here are the edits:
[image]

RFBomb, 2021-09-09:
Ok, so I slept on it and I better understand the example you provided, and understand the logic behind. A nozzle with an offset of zero should still be restricted on the minimium side of the build plate if a nozzle with a positive offset needs to due a prime skirt . The same goes for the nozzle at 0 being restricted on the max-side due to another nozzle with a negative offset requiring to lay down the prime skirt. 
But, if both nozzles can reach some position (in my printer's example, both can reach from 0 -> 211 out of a max 235), then the available print size should be 211 - border size *2. The current method imposes a restriction on E1 due to the mirroring effect, meaning my actual build volume is from (24 + border) to (235 - 24 - border), meaning there is 24mm of usable space on the 0-side deemed unreachable.
I believe my the edits in my comment directly above this one should resolve this though. Once I get home, I'll load it into my tester to verify.

Ghostkeeper, 2021-09-09:
> The argument that the unreachable area for E1 needs to be modified due to E2's offset due to bed adhesion doesn't make sense since the border_size is already accounted for when generating the point tuples.

Maybe it helps for your thought process to simply take bed adhesion out of the equation. Let `border_size` be zero. Just as long as you're clear that when both extruders are used anywhere in the print, an object that's printed with one extruder still can only be printed in areas where both extruders can reach. In reality it'll need to be able to offset the disallowed areas for both extruders equally due to the bed adhesion and collision avoidance radius, but for now just assume that is 0. It makes these calculations much easier to understand.

> On the other side of this is that if ONLY E2 is enabled, the old method thinks that it can reach the entire build area, while this is not the case.

No, the current implementation should then show the disallowed areas for E2, which would mean that the far right side of the build volume can't be reached in your case. Even if the extruders are not used, they still participate in the calculation of the disallowed sides due to nozzle offsets. If you have a project file that shows otherwise, I'd love to see it. You can just load in a simple cube and disable the build plate adhesion. This is the bug that your modified branch should fix, but I don't see that bug happening by reproducing your situation: [disallowed_area_e2.zip](https://github.com/Ultimaker/Cura/files/7136358/disallowed_area_e2.zip)

> But, if both nozzles can reach some position (in my printer's example, both can reach from 0 -> 211 out of a max 235), then the available print size should be 211 - border size *2. The current [...]
I believe my the edits in my comment directly above this one should resolve this though. Once I get home, I'll load it into my tester to verify.

Again, if the right nozzle can reach to 0,0 and your left nozzle is fixed to the same print head and 24mm further left, that means that your left nozzle can reach to -24,0. The total build volume is the union of where both your nozzles can reach, so the build volume should start from X=-24, with a 24mm border on the left side where only the left nozzle can reach but not the right nozzle. Cura will assume that the coordinate system has 0,0 on the left front corner, but this would then have to be changed in the start g-code if your printer firmware assumes differently.

> I don't think my setup is all that unusual. Its basically just two extruders sitting side-by-side. It just happens that E1 is at 0,0 and E2 is at -24,0. The only difference between this and a chimera printer is really that its a second fully standalone extruder body.

I guess this is where we have to disagree then. The unusual part about your setup is that apparently the movement of the print head is the limiting factor on the right side, but the size of the build plate is the limiting factor on the left hand side. Two extruders side by side and having separate extruder trains is not unusual indeed, and neither is nozzle offsets going into the negative really, but that's all quite irrelevant to the issue you're having. The issue is that your printer is limited on one side by something else than on the other side, which is unusual but can be resolved with custom disallowed polygons in the printer definition.

RFBomb, 2021-09-10:
Firstly, I appreciate the in-depth discussion you are willing to continue on this subject. I know there are a lot of other 'open issue' tickets here, and dual extrusion is definitely secondary to single-extrusion setups priority-wise as I'm sure its much less common.

Ok, so to work around the 'mirror' restriction I had made 2 printer profiles, one of them had only 1 extruder set up. Without a second nozzle in the profile, I was able to use the full build plate with nozzle 1. Also, in my test app, I was looping through the 'enabled' extruders instead of 'all extruders', which produced the screenshots. After fixing that bit, and retesting my printer profiles, the disallowed areas in my test app match what cura actually does, and calculates properly. So my setting up multiple profiles is where my confusion came from for this aspect, I conflated that with the calculation. Continuing on....

> The unusual part about your setup is that apparently the movement of the print head is the limiting factor on the right side, but the size of the build plate is the limiting factor on the left hand side.

You are correct that the right-hand side is limited by the carriage, as the carriage will hit X-Max prior to E2 reaching Bed-Max. Where we seem to be disagreeing being interpreted that the build plate is the limiting factor on the left hand side, when I've stated that both nozzles can reach 0 on the build plate. 

> the build volume should start from X=-24, with a 24mm border on the left side where only the left nozzle can reach but not the right nozzle. 

This seems to be the confusion point between us regarding the restriction, because I agree with your next statement:

>Cura will assume that the coordinate system has 0,0 on the left front corner, but this would then have to be changed in the start g-code if your printer firmware assumes differently.

We agree that 0,0 should be the front left of the print bed. It would seem that you are thinking that because a print head has a negative offset, that that should be the start of the build volume (in this example -24). I disagree with that statement and do agree that the build volume starts at 0,0. As I'm sure most users would assume as well. 

A negative offset should mean that when the printer is at 0,0 then the nozzle with the negative offset is sitting in some area _outside_ of the build volume. In my case:

When E1 is at 0, E2 is sitting to the left of the build plate (hanging mid air, not above the plate). If my left-hand nozzle were sitting on the edge of the build plate at the minimum stroke position, then the left extruder should be supplied with an offset of 0, while the right extruder has an offset of 24. But, while at X-Min, the left extruder is physically hanging 24mm to the left of the plate. 

I'm running under the assumptions that, based off all mechanical designs of printer's I could find:

- A nozzle with a negative offset will be floating outside the build plate (to the left of the build plate) when at (0,0)
- A nozzle with a positive offset will never reach 0, unless the printer has a stroke limit negative the corresponding value.
            - Offset of +24 can only reach a min of 24, unless the printer references to less than 0 when it touches the homing switch (which is atypical, and thus shouldn't be accounted for unless cura wants to include stroke limits into the calculations, which would be a lot of effort for limited use printers.)
            - Since having a negative stroke limit can be considered a non-standard situation, it is safe to assume that this positive offset nozzle can not reach any value below the offset number.
            
- Any nozzle with offset of 0,0 should be able to reach the entire build plate, unless something obstructs it. 
            -  A nozzle2 with a negative offset should not count as an obstruction, as anything that extends the carriage in this direction will cause the carriage to hit the 0-switch sooner, thus moving the 0-position to the right. As far as the printer is concerned, the nozzle1 is still at (0,0), the firmware has no idea that the carriage was made larger, only that the 0-switch was triggered.
                 - If this nozzle2 with a negative offset is sitting on the build plate while touching the 0-switch, then I would argue that the printer is configured incorrectly, and this nozzle2 should actually be set to the offset of 0,0 while nozzle 1 has a positive offset. The Nozzle sitting closest to the (0,0) position should be configured as such, any nozzles to the left have a negative value (floating off the plate at (0,0), and nozzles to the right have a positive offset.
            - A Nozzle2 with a positive offset SHOULD count as an obstruction, because adding onto the  positive side of the carriage will mean that it potentially reaches a mechanical stop on the positive side sooner. 
            
- Thus, it makes sense that a positive offset nozzle2 restricts the movement of a nozzle1 with offset of (0,0), while a Nozzle2 with a negative offset should not restrict nozzle 1. (as the nozzle with a negative offset was off of the bed to begin with)

The above assumptions seem generally accurate as far as X is concerned. Y is a little trickier, since some printers sling the bed around, while others move the carriage extrusion around. 
 - For printers that sling the bed around, having a nozzle2 with a Y offset should have no effect on Nozzle 1's (with no offset) bed accessibility. 
[... cut ...]
[image]
Positive offset on X -- Nozzle with 0 offset has restriction due to the addon from a nozzle on positive side. Positive offset nozzle cannot reach the 0-side. 

[image]
Negative offset axis cannot reach positive side of bed. Both can reach negative side of bed.

[image]
Positive offset axis cannot reach negative side of bed. Both can reach positive side of bed.

[image]
Nozzle 2 has X and Y as negative. Both nozzles enabled.

[image]
Nozzle 2 has X and Y as negative. Nozzle 2 is disabled.

Ghostkeeper, 2021-09-14:
> We agree that 0,0 should be the front left of the print bed.

No, that's just what Cura will output. It's possible that the firmware would allow movement down to -24, but to get there you'd need a `G92` in the start g-code because Cura assumes that 0,0 is the start of the build volume.

> It would seem that you are thinking that because a print head has a negative offset, that that should be the start of the build volume (in this example -24). I disagree with that statement and do agree that the build volume starts at 0,0. As I'm sure most users would assume as well.

With that you are restricting the build volume artificially then. The left nozzle is able to reach -24. The build plate apparently continues as far down there. But you want to disallow printing to -24 with the left nozzle. Of course, if there's a good reason to restrict the build volume that's fine, but otherwise you're just needlessly reducing the build volume on the left side. In this case you appear to have a good reason to limit movement on the left side, because:

> When E1 is at 0, E2 is sitting to the left of the build plate (hanging mid air, not above the plate).

So indeed, on the right side movement is limited by the carriage. On the left side it is limited by the range of the build plate. That's fine and can be configured in Cura, but you have to use disallowed areas.
For me this confirms your situation as needing a disallowed area on the left side of the build volume. I will close the issue for this reason, as it is not a bug.

For the range of motion of the nozzles, it still doesn't matter whether one nozzle has X-24 and the other X0, or whether the one has X0 and the other X24. These coordinates are all virtual in the firmware of the printer. The physical end result is the same. Only the g-code coordinate range that the firmware expects is different. And this can be changed with a `G92` or with the nozzle offsets. So the size of the build volume in Cura shouldn't be affected by this, nor should the obstructed areas. That makes all of your reasoning about the nozzle offsets moot. I could set the nozzles to X-99999 and X-99975 and still print fine as long as I add `G92 X-99975` to the start g-code, with the same range of motion for both nozzles.

Similarly, it doesn't matter if a printer moves the bed or the print head, nor whether that bed is moved in the X direction, the Y direction or both. All that matters is the range of motion of the nozzles relative to the build volume. It's all just a different frame of reference for the same thing. If the printer moves the build plate rather than the print head, it is expected to invert the direction of motion automatically in firmware. There is no method in Cura to mirror the g-code other than to actually mirror the objects you're printing.

closed 2021-09-14 by Ghostkeeper

RFBomb, 2021-09-15:
I’ll accept your answer, as you’ve held your ground, which is fine. It’s your software. 

But:

> No, that's just what Cura will output. It's possible that the firmware would allow movement down to -24, but to get there you'd need a G92 in the start g-code because Cura assumes that 0,0 is the start of the build volume.

My whole argument was based on this here statement. Since Cura does not currently account for Min/Max stroke limits, then 0,0 should be assumed as the limit. Sure, G92 in the start code can ‘fix’ the issue, but then you are relying on the user to be aware of then implement a workaround to a problem that could be easily caught in the tool, and I’m arguing should be caught. 

The default setting is to have the nozzle offsets accounted for in the GCODE output. Which means that for Nozzle 2 (-24,0) to get to 0, the carriage must move nozzle 1 (0,0) to the 0 position. It doesn’t matter at all that it hangs off the bed when nozzle 1 is at 0, it’s not in use. Which still only supports my side of the argument, since this is still working within the reason that 0,0 is reasonable start of build plate. 

If the gcode output always used the left-most nozzle as the frame of reference, then this ar unmentioned of your I would agree with your argument (nozzle 2 would be treated as 0,0 and nozzle 1 treated as 24,0 ) but this is not the case. 

To have the ‘fix’ be G92 at the start and build custom areas is absurd. Custom areas can be required in some setups, but this is a relatively standard setup for dual extrusion. Say you did G92 X-24 in the start code. The printer will move to the new 0 for the purge, but it’s actually 24mm into the bed thanks to the nozzle offsets applied to the gcode. Oh crap, that’s where the part was supposed to go in the slicer! Now the part is also offset by 24mm, phew. But wait, is it still on the bed? 

Or how about this: disable nozzle 2 to have full access to the build plate with nozzle 1. Cura doesn’t support logic statement in user generated code, so now you have to manually edit either the printer profile or the output gcode to comment out that G92 whenever you want to use the nozzle that can hit the full build plate. Or, you do what I did and just have 2 profiles for the same printer to overcome this absurdity. 

Simply put, the gcode output doesn’t agree with your argument, Since any negative offset nozzle will just be told to move positive until 0,0 is hit. Whether that move is done in firmware or in the gcode itself doesn’t matter, the move to compensate will occur. Whereas a positive offset nozzle will determine that 0,0 is unreachable in both the firmware (unless negative stroke limit) and in Cura.

Ghostkeeper, 2021-09-22:
There is something to be said for that there should be a configuration option for when the 0,0 coordinate origin is not the front-left of the build volume, nor the centre (which does have an option since it's common for Delta-style printers), but some arbitrary position in Cura's build volume like (24,0). But there's two counter-arguments for adding this:
* That printer configuration is pretty uncommon as far as I've seen. This seems to be one point where you were disagreeing. Look through the printer profiles in Cura right now though. The MakeIt Pro is the only printer that needs this at the moment among the hundreds of printers that have profiles in Cura.
* The user doesn't need to configure this in their start g-code if it's configured in the printer definition (.def.json file). It's only necessary for custom-built printers or printers that otherwise don't have a profile yet in Cura.

It is not necessary to add or remove the `G92` command depending on which extruders are in use, or which are disabled in the front-end. The output coordinates of Cura don't depend on this either, and the printer firmware can't know without scanning through the g-code ahead of time for future tool switches.

RFBomb, 2021-09-22:
After this was initially closed, I did write a somewhat lengthy comment (that seems to have disappeared) further clarifying my position. 

> There is something to be said for that there should be a configuration option for when the 0,0 coordinate origin is not the front-left of the build volume, nor the centre

I don’t necessary disagree with this sentiment, and agree with your counter-arguments. But this statement appears to disagree with your previous statements (and bed bed size calculations) that say because Nozzle 2 has a negative offset, the build volume starts at -24 instead of 0. (as shown by your comment here:

> The left nozzle is able to reach -24. The build plate apparently continues as far down there. But you want to disallow printing to -24 with the left nozzle.

Which is the root contention point of the discussion. Cura gcode output assumes a bed point start of 0,0 (agreed). 

Attached are several projects to demonstrate what I'm referring to. These are 3 project files using the Custom printer with 1 or 2 extruders. If 2 nozzles, the second nozzle has my -24 offsets applied. (I applied it to Y as well to make XY identical though) I am using a 20x20mm cube to showcase.

[Project Examples.zip](https://github.com/Ultimaker/Cura/files/7210199/Project.Examples.zip)

Note: `Apply Extruder Offsets to GCODE` is set to OFF, just like it is on my printer. (Not that this should have affected E1 whose offsets are 0,0 )

-----
DualExtruderProject 1:
[image]

As can be seen in the photo, the cube refuses to slice when set for nozzle 1 (0,0) when the cube is in the corner of the plate.

When Selecting to slice with Nozzle 2, this is the output:

```
;Generated with Cura_SteamEngine 4.10.0
T1
M104 S215
M105
M109 S215
M82 ;absolute extrusion mode
;; START GCODE DISABLED FOR EXAMPLE
;; The following GCODE is start of generated code.
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:198
;LAYER:0
M107
;MESH:0.8mm Wall Flow Cube.stl
G0 F3600 X2.6 Y21.4 Z0.3
;TYPE:WALL-INNER
G1 F1500 E0
G1 F1800 X2.6 Y2.6 E0.35364
G1 X21.4 Y2.6 E0.70728
G1 X21.4 Y21.4 E1.06092
G1 X2.6 Y21.4 E1.41455
G0 F3600 X2.2 Y21.8
;TYPE:WALL-OUTER

```

As can be seen, T1 is moving to X=2.6 on the build plate for starting point of the part. (If offsets were applied this would be 26.6 due to the -24 offset). Meaning that the build volume does not start from -24, but from 0 according to Cura's output. 

Project 2

This shows where Nozzle 1 can print in the dual extruder setup.

[image]

Here, you can see the cube will slice at the edge of what Cura says Nozzle 1 can reach. Assuming that the build plate reaches to -24 (which has been stated multiple times in defense of denying the change), This cube's corner should start approximately at (0,0).

Output gocde:

```
;; START GCODE DISABLED FOR EXAMPLE
;; The following GCODE is start of generated code.
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:198
;LAYER:0
M107
;MESH:0.8mm Wall Flow Cube.stl
G0 F3600 X25.6 Y44.4 Z0.3
;TYPE:WALL-INNER
G1 F1500 E0
G1 F1800 X25.6 Y25.6 E0.35364
G1 X44.4 Y25.6 E0.70728
G1 X44.4 Y44.4 E1.06092
G1 X25.6 Y44.4 E1.41455
G0 F3600 X25.2 Y44.8
;TYPE:WALL-OUTER
```

Note how it starts at X=25.6  (rear left corner). 

----

Lastly, the same cube in a single nozzle setup.
[image]

Output GCODE:
```
;; START GCODE DISABLED FOR EXAMPLE
;; The following GCODE is start of generated code.
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:198
;LAYER:0
M107
;MESH:0.8mm Wall Flow Cube.stl
G0 F3600 X2.6 Y21.4 Z0.3
;TYPE:WALL-INNER
G1 F1500 E0
G1 F1800 X2.6 Y2.6 E0.35364
G1 X21.4 Y2.6 E0.70728
G1 X21.4 Y21.4 E1.06092
G1 X2.6 Y21.4 E1.41455
G0 F3600 X2.2 Y21.8
;TYPE:WALL-OUTER

```
X starts at 2.6, which is about what I would expect given the position of the part on the plate. This matches the output of when printing with Nozzle 2 in the first example. 

---------

Currently, the issue I'm arguing against is this statement:
> The left nozzle is able to reach -24. The build plate apparently continues as far down there. But you want to disallow printing to -24 with the left nozzle.

I _agree_ that technically the reachable volume is starting from -24 for nozzle 2. That is not in question. But `Reachable Volume != Build Volume`. As demonstrated, the GCODE output is consistent that the front left corner of the displayed build plate is deemed 0,0. 

This means that Cura has a discrepancy between the gcode output and where it allows the user to place the parts. This is what I'm trying to resolve.

If it matched, then I would have expected example 1's gcode to start around -21.4 (~2mm inward of the suggested build plate min of -24mm). But instead, it started at X=2.6mm. Likewise, example 2 should have started at X=1.6, but instead started at 25.6, since is the `build plate extended to -24` then that would mean that 0,0 should have remained untouched.

RFBomb, 2021-09-22:
In summary:

>>   (this is me talking) We agree that 0,0 should be the front left of the print bed.

>No, that's just what Cura will output. It's possible that the firmware would allow movement down to -24, but to get there you'd need a G92 in the start g-code because Cura assumes that 0,0 is the start of the build volume.

So we agree that Cura outputs gcode based off 0,0 being the front left of the print bed (which is where my quote above was trying to say).  When the user is placing parts onto the build plate, this is also how it is displayed to the user. BUT your statement of:

> With that you are restricting the build volume artificially then. The left nozzle is able to reach -24. The build plate apparently continues as far down there

conflicts with this. The disallowed mesh is drawn adjusts the display as if -24 is the location of the build plate, while the gcode generator is still treating that same point as 0,0 on the build plate. This is the conflict area I'm questioning. 

> There is something to be said for that there should be a configuration option for when the 0,0 coordinate origin is not the front-left of the build volume,

I dont think there needs to be an option. Either the disallowed area drawing needs to match GCODE generation (my proposal), OR the gcode generation needs to account for the extension of the front left corner in the case of negative offsets.

Ghostkeeper, 2021-09-29:
> But this statement appears to disagree with your previous statements (and bed bed size calculations) that say because Nozzle 2 has a negative offset, the build volume starts at -24 instead of 0.

Sorry for being unclear. According to Cura the build volume (its term for the total volume in which either of the nozzles can move) starts at 0,0. According to your firmware it starts at -24,0.

With "the left nozzle is able to reach -24" I was mentioning the coordinates in your firmware which was the context at the time. Cura still assumes the front-left corner of the build volume is 0,0 in its g-code coordinate system, both in where the disallowed areas end up and in the g-code generation. As far as I can see, the disallowed area drawing matches where the nozzles could possibly be depositing material, except that in your case your build plate doesn't match your build volume (which is what I mentioned before, the special case of your printer where the build plate limits deposition on the left side but not the right side).

To be precise about it, this disagrees with this statement in your summary:

> So we agree that Cura outputs gcode based off 0,0 being the front left of the print bed (which is where my quote above was trying to say, that you replied to). When the user is placing parts onto the build plate, this is also how it is displayed to the user. BUT your statement of:

Cura outputs g-code based off 0,0 being the front left of the *reachable volume of all nozzles*, which is in your case not the front left of the build plate.

RFBomb, 2021-09-29:
> Cura outputs g-code based off 0,0 being the front left of the reachable volume of all nozzles, which is in your case not the front left of the build plate. 

OK, this is where the confusion was based off of then. I didn't see this clearly documented in the code I poured through, nor anywhere else for that matter, and as such assumed that 0,0 was the start of the build volume, not that 0,0 would shift according to nozzle offsets like you say. (In my case, it shifts 0,0 by -24). 

Maybe its just me, but this is very non-intuitive. I had assumed that since cura starts the build plate at 0,0, you should configure your printer's firmware as such. (The nozzle closest to bed 0,0 should be 0,0 -> as long as that nozzle can reach 0,0). Based on this, I had made my printer's nozzle closest to 0,0 have an offset of 0,0. The secondary nozzle then had the -24 X offset. And the minimum limits were adjusted so that 0,0 was 0,0 on the build plate. (in my case I think its an offset of (-5,-1) will put the 0,0 nozzle at 0,0 on the plate. (I don't think I'm the only one with this assumption though, because a quick google search 'Cura Bed origin' pulls up several others that have had similarly confusing issues).

 So, now that the calculated origin is effectively at -24,0, the options a user has to resolve this are:

- Do a G92 at the start as you suggest. In my case (G92 X24)  won't resolve it because then nozzle 2 would be "At 0" just floating above left of the build plate. But it would now match Cura's  gcode coordinates. 
- Set my printer firmware so the left nozzle is at 0,0 and adjust the minimum stroke to make it sit at 0,0, then have nozzle 1 (right nozzle) at 24,0.  This will eliminate the need for G92 at start. Max stroke and bed size values. will have to be adjusted to ensure that nozzle 1 can reach entire build plate.
- As you said, custom disallowed areas.

I still don't agree with the shifting where 0,0 is with no clarification to the end user, but it appears this won't be changing any time soon. Calling it front left or the 'reachable area', is correct, but shifting where zero is due to a negative offset just seems completely wacky to me. If a sports ball hits the benches on the sideline, no-one says that ball is still in play.

RFBomb, 2021-11-08:
Reviving this thread, as I have recently finally been able to use my printer again after moving to a new place. 

As far as I can tell, disallowed areas affect all print heads. I don't see a way to set a disallowed area per nozzle. Also, the disallowed areas are calculated with their 0 being the bed center?

RFBomb, 2021-11-09:
Following up:
So I tried configuring Marlin to what you adamantly believe to be the 0 position, with Nozzle 2 being '0'. Turns out, you cant!
Marlin refuses to allow compiling with E0 not set to 0,0. 

Which just further supports my argument that this is a bug that should be fixed. If Nozzle 1 (E0) has to be 0 at a firmware level, it can be easily assumed that this nozzle can reach 0 on the build plate. Which goes against the argument that "the build volume starts at -24, so -24 is actually 0 in cura", which is your premise of rejecting this fix. 
Also, I haven't found anything that would descirbe how to implement your proposed solution of custom disallowed areas, because as far as I can tell, these areas affect all extruders. Meaning, at best, you wind up in the same situation.

The root of the problem here is that the 0 point gets shifted to the most-negative value. Which isn't how math works. My solution would essentially ignore the -24 that really doesn't exist in the build volume, which admittedly isn't the best solution, but its better than current. Ideally, it would expose that area as printable for Nozzle 2 like it currently does. but the positional value would be correct (result of -24 instead of incorrect value of 0)

GregValiant, 2021-11-12:
I started over on Reddit, and then went through the dissertation here.  I have some thoughts.

- Any single extruder printer that is converted to a "true" dual extruder will have the original nozzle that never leaves the build area, and a 2nd nozzle that does leave the build area.  If the customization causes an interference with the X motor housing or some other printer part that may be on the ends of the X beam then the X width of the build area will be altered since the X travel will be limited.  I will stay away from that situation for now and assume that there is no interference.
- The 0,0 point never changes.  Cura only controls the location of nozzle 0 which should always be the original nozzle as it's the one that can move everywhere within the build area.  When a tool change is made then Cura subtracts the offsets and so nozzle 0 is over air at the new location while nozzle 1 went along for the ride and is now over the print and extruding plastic.  The gcode bears that out.  If you bring a dual extruder gcode into Cura while a single extruder printer is active you will see a shift of the piece being printed by extruder 1.
- If a second hot end is added to the _left_ of the original hot end it will have an area on the far left that it can access but is unprintable as it is off the build plate (and extruder 0 can't get there).  It will also have an unreachable area on the right side that it just can't get to.
- If a second hot end is added to the _right_ of the original then it will also have those two areas with the unprintable area off the right edge and the unreachable area on the left side.

@RFBomb added a hot end to the left of the stock hot end.  His unprintable area would be from X=-24 to X=0 and the unreachable area would be from X=211 to X=235.
@Ghostkeeper, it seems to me that for his machine there should be a single gray area on the right side of the build plate.  The one on the left could move left equal to the X offset and moves it off the build plate.
If a user was to decide to make the _new_ extruder E0 then that would change things again.

It does occur to me that accounting for every 3rd party printer customized with 4th party parts and assembled by a 5th party might be beyond the scope of Ultimaker Cura to support.

That's my two cents anyway.  Good luck to both of you with this.

Ghostkeeper, 2021-11-15:
> I don't see a way to set a disallowed area per nozzle.

Indeed, but the [Nozzle Disallowed Area](https://github.com/Ultimaker/Cura/blob/c9feace0fbe02beb2089ec0af7be35127d7420f7/resources/definitions/fdmprinter.def.json#L436-L447) is the one that gets offset by the nozzle offset.

> Also, the disallowed areas are calculated with their 0 being the bed center?

Yes, it's in Cura's coordinate system (same as the objects you place with the move tool) rather than g-code coordinates.

> So I tried configuring Marlin to what you adamantly believe to be the 0 position, with Nozzle 2 being '0'. Turns out, you cant!
Marlin refuses to allow compiling with E0 not set to 0,0.

Okay, so the firmware doesn't support having an offset for the first extruder but Cura does. So basically for your firmware you need to have the coordinate origin of the g-code output be 24mm off the left side of the build plate. Now while that's not supported with an easy drop-down selection (where Cura only allows front-left or centre), it _is still supported_ by modifying the start g-code. I've mentioned this before, but it's really just as simple as this:

```
G0 X0
G92 X24
```

So yes, the g-code output is indeed limited to the most common 2 cases, but the setting change to make it work is still possible. That's what the start g-code field is meant for: All of the uncommon configurations that Cura doesn't have a setting for.

RFBomb, 2021-11-15:
> > I don't see a way to set a disallowed area per nozzle.
> 
> The [Nozzle Disallowed Area](https://github.com/Ultimaker/Cura/blob/c9feace0fbe02beb2089ec0af7be35127d7420f7/resources/definitions/fdmprinter.def.json#L436-L447) is the one that gets offset by the nozzle offset.
> 

How  would you go about setting this up custom for each nozzle then?

> > Also, the disallowed areas are calculated with their 0 being the bed center?
> 
> Yes, it's in Cura's coordinate system (same as the objects you place with the move tool) rather than g-code coordinates.
> 

So 0 is the 'bed center' as far as  disallowed  areas are calculated. Which is  why this problem comes up. It shifts  the left  corner relative to 0. 

Say a printer with a  200mm bed had 2 nozzle, one at 0 and 1 at -10. 
With respect to cura, nozzle 1 (0,0) would be 100 mm from bed center. while nozzle 2 should  be at 110  from center. Due to  current calculation,  it is actually 90 and 100 respectively. Thats the problem.

> Okay, so the firmware doesn't support having an offset for the first extruder but Cura does. So basically for your firmware you need to have the coordinate origin of the g-code output be 24mm off [...]
This still seems like it is producing a very reduced build volume is my point. If cura is calculating GCODE 0,0 from the left-most nozzle (which  has a negative offset), this is a bandaid only, and the problem still exists. The GCODE will still go by the bed-max  value, in my case im using 235. So,  this happens:

- Cura Says 0,0 is  left-most nozzle. 
- You have +235 with respect to 0.  
- 0,0 in reality is a actually  +24 according  to  cura. Which means that X235 is actually X259 in cura. 
- But wait, X259 isn't accessible because the bed max is X235!  (Unless bed is stretched to259, which I saw no evidence of in the code or my testing)

Meaning you lost 48mm of printable space on Nozzle 2 (24 each side) and 24mm of space on Nozzle 1 (which can reach the full  bed) due to the effective 0-shift. 

If you are going to recommend that gcode workaround, atleast offer the full workaround. You'd have to increase your bed size  by the same distance to account  for the shift in where the front-left corner (0,0) is located. In my case, I'd have to set the profile up with a bed size  of 259  to  account for the  0-shift.

> So yes, the g-code output is indeed limited to the most common 2 cases, but the setting change to make it work is still possible. That's what the start g-code field is meant for: All of the uncommon configurations that Cura doesn't have a setting for.
> 

I'm still confused as to which use case shifting the front-left corner works for. 

> You say that this makes the nozzle float above the build plate, which is correct according to the g-code coordinates but you can't print there. This can be prevented by adding disallowed areas then. That's what they are intended for.

@GregValiant  is spot on and understood my side of the discussion precisely. **0,0 should not be shifting**,  since  all nozzle offsets are assumed with respect to 0,0. 

_You shouldn't  need a disallowed are for an area that isn't even on the build plate._

Ghostkeeper, 2021-11-16:
> How would you go about setting this up custom for each nozzle then?

It automatically shifts by the nozzle offsets, causing them to disappear when they shift out of the build volume. It's not possible to have completely separate shapes of disallowed areas for each nozzle separately.

> So 0 is the 'bed center' as far as disallowed areas are calculated. Which is why this problem comes up. It shifts the left corner relative to 0.

The [Disallowed Areas](https://github.com/Ultimaker/Cura/blob/c9feace0fbe02beb2089ec0af7be35127d7420f7/resources/definitions/fdmprinter.def.json#L424-L435) setting doesn't shift per extruder. You'll indeed have to write the Disallowed Areas in a different coordinate system. It's different from the g-code coordinates in that the Y axis is flipped and the coordinate origin is different. This is only a hurdle for writing the profiles though, not for printing. The disallowed areas are not output to g-code.

> With respect to cura, nozzle 1 (0,0) would be 100 mm from bed center. while nozzle 2 should be at 110 from center. Due to current calculation, it is actually 90 and 100 respectively. Thats the problem.

Then your printer width needs to be 10mm higher, right? Cura doesn't calculate the width of your printer. It uses the setting given in the machine definition.

I'm confused as to why you think the 0,0 corner is shifting. When does it shift? When you change the build volume it doesn't shift because the g-code coordinates remain the same. Whether it shifts with the nozzle offset in the g-code coordinates depends on the [Offset with Extruder](https://github.com/Ultimaker/Cura/blob/f0cb9dd39ef996225ebd6d7b8d5ffc986c90e2f0/resources/definitions/fdmprinter.def.json#L508-L517) setting.

If I increase the size of the build volume, it grows towards the back-right. The 0,0 coordinate stays in the same spot on the build plate by definition. It doesn't shift. Cura will allow me to print bigger things, and will generate g-code coordinates then that go beyond the right and far side of the build plate. Maybe you are confused by how it looks in Cura as if it's growing from the centre of the build volume? That's just where the camera stays, really.

> In my case, I'd have to set the profile up with a bed size of 259 to account for the 0-shift.

That's correct then, right? It's reducing the build volume for you to make the reachable area of each nozzle correct as far as it knows. So you need to increase the size of the build volume in Cura to make it encapsulate the combined area of all of the nozzles.

I don't think disallowed areas need to be inherently linked to whether they are on the build plate or not. They can be disallowed for a myriad of reasons, and not being on the build plate is a perfectly fine reason.

RFBomb, 2021-11-16:
> Then your printer width needs to be 10mm higher, right? Cura doesn't calculate the width of your printer. It uses the setting given in the machine definition.

Why would the width need to be higher if the calculation were correct? Increasing the width only accounts for the fact that 0 gets shifted away from 0.

>I'm confused as to why you think the 0,0 corner is shifting. When does it shift? When you change the build volume it doesn't shift because the g-code coordinates remain the same. Whether it shifts with the nozzle offset in the g-code coordinates depends on the Offset with Extruder setting.

The 0-Shift I refer to is because the front-left corner is being calculated as the left-most + front-Most positions, even if those positions have negative values. I don't understand how you can't see this, when every workaround you suggest accounts for this very thing. Negative is less than zero. You can't print less than zero when the front left IS 0. Therefore, the negatives should be dropped.

The GCODE translation is moot here, because I am simply referring to the display area that allows for slicing. GCODE will be adjusted per that setting to enable/disable including the offsets in gcode output, yes. 

As in, since 0 is being shifted to the front left corner, nozzle 1 sitting at 0,0 is referred to as 24,0. With the GCODE modifier, its still 24,0. BUT, with nozzle 2's gcode modifier, it becomes 0,0 `for nozzle 2 only`. without the gcode modifier, its still printing 24,0 for the gcode. when that 24 is actually bed 0. IE zero has shifted 24mm.

> That's correct then, right? It's reducing the build volume for you to make the reachable area of each nozzle correct as far as it knows. So you need to increase the size of the build volume in Cura [...]

Yes, that is a workaround I discovered that needs to be done in addition to your workaround of using G92. a workaround because the problem of 0 is shifting to the negative offset.

Ghostkeeper, 2021-11-16:
> Why would the width need to be higher if the calculation were correct? Increasing the width only accounts for the fact that 0 gets shifted away from 0.

The width of the build volume is defined as the union of all areas that the nozzles can cover. If the nozzles have a big offset, they can cover a broader area. So yes, the width of the build volume needs to be higher because of the nozzle offsets.

I see now that what you meant with "is shifting" is simply that there is a difference between the coordinate system of Cura's g-code output and the coordinate system of your firmware. I got confused by how you said you didn't understand why Cura is making this difference. But it seems that all stems from the root cause that 0,0 is on your printer not the left-most coordinate that can be reached by a nozzle.

I hope this workaround with G92 works for you, and that you can configure the build volume correctly to be the union of all reachable positions of all nozzles. I get the feeling that you've found something that works for your custom printer now. The remaining issue is then whether Cura will make it easier to support coordinate origins other than the centre or the front-left of the build volume, and perhaps the documentation of that the size of the build volume in Cura needs to encapsulate all reachable positions by all nozzles. The documentation is covered by the Settings Guide, but not in Cura itself nor the wiki. I could add it to the wiki instructions to clear up the confusion for people who read that.

RFBomb, 2021-11-16:
This workaround should work, but its super clunky until you realize what its doing, and extremely non-intuitive.

As I've been saying, the major confusion point is that 0 is effectively shifted to the left-most nozzle. I'm not necessarily disagreeing with your logic of why it gets put where it is, in fact I 100% understand why it is the way it is, even if I think that way is incorrect.

But, my argument is that I would imagine 99% of users would assume that 0 doesn't move at all, whereas the way areas are currently drawn is effectively shifting where 0 is for the nozzles with no offsets.

> But it seems that all stems from the root cause that 0,0 is on your printer not the left-most coordinate that can be reached by a nozzle.

Yes, that is the very root of my argument. As I've said previously, I believe the nozzle that can reach the closest to 0,0 _on the bed_ when the printer homes should be the 0,0 nozzle. NOT the left-most nozzle itself. If the left-most nozzle were to be assumed 0,0, then it would make a whole lot more sense to simply force positive offsets in the extruder setup. 

My reasoning is that by allowing negative offsets in the extruder setup, and not dropping disallowed areas that occur less than 0, you effectively shift 0 to the negative nozzle. So either the calculation needs to be fixed to drop the areas less than zero, OR only positive offsets should be allowed, since that is how the math is being performed. (For obvious reasons, the latter would break any printer with negative offset profiles, so the fix to the calculation is what I'm arguing for)