Ultimaker/Cura #10355: Fail to slice large model in 4.11.0 Beta

jskylevy, 2021-08-30:
### Application Version

4.11.0

### Platform

Win 10

### Printer

Voxelab Aquila (+'Alex' Marlin Firmware)

### Reproduction steps

Open this model in Cura, slice using a saved profile, get the error that the slicing failed. I didn't get a chance to copy the exact error message.

Model:

### Actual results

The model fails to slioe.

### Expected results

The model is sliced and ready to save/send as Gcode.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

PC specs:

Ryzen 1700x CPU, AMD 380x GPU, 16 GB DDR3 ram. Multiple other programs were running. Since my CPU is very old, I suspect the problem was a lack of computing power.
[dragon-unsupported.zip](https://github.com/Ultimaker/Cura/files/7073562/dragon-unsupported.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7073564/cura.log)

fvrmr, 2021-08-30:
Hi @jskylevy thank you for your bug report.
I can reproduce your issue. I will bring this up for discussion with the team. Keep you posted.

johnkray, 2021-08-30:
I'm experiencing similar failed slicing issues. It has happened even on small parts. Slicing with 4.11 beta on Windows 10, I'm seeing ram usage spike upwards of 10GB right before slicing fails.

Ghostkeeper, 2021-09-08:
I can slice the model fine now. Peak memory usage of CuraEngine was 580MB. Slicing completed in 94 seconds. Memory usage of Cura in total ended up at 2GB since layer view was using much more memory than CuraEngine ever did.

During the beta period we made a fix that could also have fixed this issue: https://github.com/Ultimaker/CuraEngine/commit/71b7b6a43c95380e628063dfdbebce57b8f5d23a The symptoms for that bug match with what both of you describe: It got in an infinite loop, which also kept allocating more and more memory, leading eventually to a crash because it can't allocate more memory.

Perhaps it's worth another try in Cura 4.11 stable?

fvrmr, 2021-09-09:
I can slice it as well in 4.11 stable, please let us know if it works for you now.

johnkray, 2021-09-09:
I have had no slicing issues/runaway ram usage with the release version of 4.11. I was having this issue quite a lot with the beta with multiple different parts, but haven't experienced it once with the release version.

closed 2021-09-17 by Ghostkeeper