Ultimaker/Cura #10362: Layer slider up/down buttons

BeePerNet, 2021-08-30:
### Is your feature request related to a problem?

The way to go up and down by the slider or by the layer textbox number. When having a lot of layer, first way bypass some layers because of the slider bar pixel height is lower than the number of layers. It is slow with the textbox one. Let's find a more user friendly way to navigate between layers?

### Describe the solution you'd like

Add a button above and bottom the layer slider bar to go to the above or below layer of the selected slider round button. I saw that the highlight of the selected one disappear on focus lost. Keep the selected button highlighted.

### Describe alternatives you've considered

Keyboard buttons + and - could do the job, maybe with shift or ctrl because these ones are used for zoom. Another way could be to use all the space to have more pixels on the slider, but this will create the same problem on higher ones.

### Affected users and/or printers

Easy way to rapidly switch to the next layer when inspecting them before print. I think this could help a lot of user. I don't have the skills to evaluate how much.

fieldOfView, 2021-08-30:
The keyboard Up and Down arrows should work, provided you first click in the layer number field.

BeePerNet, 2021-08-30:
Yes it works, I had to ask to know, not intuitive and not mentionned in the doc. Or I did not see it, my apologies. Tooltips could be another option. Thanks if you decide to add some more options.

fvrmr, 2021-09-02:
Hi @BeePerNet thank you for your request. 
I have discussed your option about tooltips with our UX designer and since there are a lot of tickets on our backlog this is not a priority for us to add, so I will defer this issue
I hope you understand.

closed 2023-07-13 by github-actions[bot]